mobizt/Firebase-ESP8266 #75: Problem when get big data (nested) JSON

komputerboy, 2019-12-26:
Hi,

Now im facing problem when read data. But my problem is only occur when the data is quite large / nested.

My json data is like this:
```
"automation" : {
                "list" : {
                  "-Lx-LXFnaUfvUouthP0_" : {
                    "enabled" : true,
                    "if" : {
                      "-Lx-LXFpweKWAxPMe28V" : {
                        "schedule" : {
                          "schedule" : "Minggu, Senin, Selasa, Rabu, Kamis, Jumat, Sabtu",
                          "time" : "10:00"
                        }
                      }
                    },
                    "name" : "Nyalakan lampu pukul 10:00",
                    "then" : {
                      "-Lx-LXFrgt_md37dYsXK" : {
                        "status" : {
                          "path" : "-Lvy6cd-WipQ4H7raeoO/ruang/-Leys3zqG-jSit2FhMBY/alat/lampu/30aea4044b28/status",
                          "set" : true
                        }
                      }
                    }
                  },
                  "-Lx-YbBhLzKSyLsTngi9" : {
                    "enabled" : true,
                    "if" : {
                      "-Lx-YbBjGqed2rfMys0a" : {
                        "schedule" : {
                          "schedule" : "Minggu, Senin, Selasa, Rabu, Kamis, Jumat, Sabtu",
                          "time" : "23:0"
                        }
                      }
                    },
                    "name" : "Matikan lampu pukul 23:00",
                    "then" : {
                      "-Lx-YbBkNZWlmiBWuC8f" : {
                        "status" : {
                          "path" : "-Lvy6cd-WipQ4H7raeoO/ruang/-Leys3zqG-jSit2FhMBY/alat/lampu/30aea4044b28/status",
                          "set" : false
                        }
                      }
                    }
                  }
                }
              },
```
When i try to get `automation/list` data i got no data, my code is:
```
if (Firebase.get(firebaseData,"automation/list/"))
  {
    FirebaseJson &json = firebaseData.jsonObject();
    //Print all object data
    Serial.println("Pretty printed JSON data:");
    String jsonStr;
    json.toString(jsonStr, true);
    Serial.println(jsonStr);
    Serial.println();
}
```

but if i try to get `automation/list/-Lx-LXFnaUfvUouthP0_` data, there is no problem, i got the data correctly:
```
Pretty printed JSON data:
{
    "enabled": true,
    "if": {
        "-Lx-LXFpweKWAxPMe28V": {
            "schedule": {
                "schedule": "Minggu, Senin, Selasa, Rabu, Kamis, Jumat, Sabtu",
                "time": "10:00"
            }
        }
    },
    "name": "Nyalakan lampu pukul 10:00",
    "then": {
        "-Lx-LXFrgt_md37dYsXK": {
            "status": {
                "path": "-Lvy6cd-WipQ4H7raeoO/ruang/-Leys3zqG-jSit2FhMBY/alat/lampu/30aea4044b28/status",
                "set": true
            }
        }
    }
}
```

Could you please help me?
Thank You

mobizt, 2019-12-26:
Can you get the completed data from firebaseData.jsonString()?

komputerboy, 2019-12-26:
Okay, i try it

komputerboy, 2019-12-26:
still, got nothing

mobizt, 2019-12-26:
If data was truncated, try to change the value of [FIREBASE_RESPONSE_SIZE](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266.h#L57) which is 400 bytes by default.

This consume more free heap.

komputerboy, 2019-12-26:
i try to change its value to 4000, then restart my Arduino IDE, but still got nothing

mobizt, 2019-12-26:
Please export JSON from console and use some online JSON editor service to cut only data you want and post here?

I want to test and analyze it.

komputerboy, 2019-12-26:
okay

mobizt, 2019-12-26:
Due to [rx buffer size reserved for BearSSL](https://github.com/mobizt/Firebase-ESP8266/blob/master/src/FirebaseESP8266HTTPClient.cpp#L47) is not enough, which the library sets it to 512 byte as default.

You can change it to **_client->setBufferSizes(2048, 512);** // 6384 bytes for rx, 512 bytes for tx bytes.

This library allows user to reserved more memory for BearSSL libray, by call **Firebase.lowMemBSSL(false);** (16384 for rx, 512 for tx) in case of lage amount of data.

This will cost your contantly memory per Firebase Data object for WiFi client buffer memory.

I may let user to set the amont of reserved memory for BearSSL in the future update.

komputerboy, 2019-12-27:
okay, thank you

closed 2019-12-27 by komputerboy