mobizt/Firebase-ESP8266 #76: Cloud Firestore

agusfaudin, 2020-01-11:
How to change program from realtime database to be cloud firestore?
give me clue problem solution
thanks

mobizt, 2020-01-11:
It's not possible right now due to the following reasons.

1. Cloud Firestore authentication is different than RTDB, it requires custom or OAuth2 tokens. This required more works which I successfully done it.

2. Stream was not support via REST API. It supported via gRPC framewok.  Which gRPC is not supported in esp8266 and esp32 now (there is closed source of gRPC  library for esp32 built for esp32 google voice assistance SDK but we can't customize it for Firestore).

SERIDJ, 2020-01-17:
hi, please we can use gRpc server with Nanopb or samth with ESP32 DEVICE ?

mobizt, 2020-01-17:
@aERIDJ 

Due to gRPC is based on protobuf then you need to compile all protobuf dependencies too.  I also try nanopb to compile proto files but some proto files can't compile.

Finally I use protobuf-c to compile and generate .c and .h instead.  

I can't see the open source of  gRPC framework that support ESP32 for now, only some close source that compile only for google voice assistance.

SERIDJ, 2020-01-17:
@mobizt  thanks for ur reply 

yes me to iam using **nanopb ** and i have probleme  when i compile this line : 
 
bool status = pb_encode(&stream, TestMessage_fields, &message); 

 from : https://medium.com/grpc/efficient-iot-with-the-esp8266-protocol-buffers-grafana-go-and-kubernetes-a2ae214dbd29

what u thing its possible to send data to visual studio server grpc with protobuf-c from esp32 ?

mobizt, 2020-01-17:
If that system (PC, device or server) can install gRPC and protobuf. But gRPC is not available for esp32 then nothing to do, we can't send/receive message to/from gRPC server.

SERIDJ, 2020-01-17:
yes yes , but what u thing with node.js and python in esp32 ?

mobizt, 2020-01-17:
node.js in esp32, no
python in esp32 is only micro python.

SERIDJ, 2020-01-20:
hi, @mobizt 

i found PJON-gRPC what u thinks i can send data to grpc visual studio server  ???

https://github.com/Halytskyi/PJON-gRPC

Patys, 2020-01-21:
@mobizt 
Is it possible to add support to firestore with https://www.microchip.com/wwwproducts/en/ATECC608A chips? They can generate and store keys required for auth to Google IoT and AWS IoT. Maybe we can extend lib to support it?

mobizt, 2020-01-22:
@SERIDJ 
If you just ask the possibilities to implement your own client-server comunication using gRPC, I can't help or advice on that.

I focus only on how to use gRPC for Google Firestore in esp8266 or esp32 device.

mobizt, 2020-01-22:
@Patys 
It may possible and need more study.