mobizt/Firebase-ESP8266 #77: HELP

amansinghaljpr, 2020-01-12:
I tried the basic example and i am getting : 

FAILED
REASON: read Timeout
------------------------------------

In every test , i dont understand why that's happening

mobizt, 2020-01-12:
Check that your esp8266 is actually connect to internet (not only WiFi connected).

Your router may not response any data to device, reset router or change router or use mobile/PC WiFi hotspot.

Test internet connectivity by making simple http request (see in Arduino example) or ping by using ping library.

closed 2020-01-12 by mobizt

amansinghaljpr, 2020-01-12:
The problem is that the data is getting updated on the firebase , the serial monitor shows the error so its a proof that the data is getting uploaded its just that something or somewhere else is the problem

mobizt, 2020-01-12:
Nothing update in your database,as no response from firebase server due to no internet connection issue.

Did you try my suggestion above?

mobizt, 2020-01-12:
Use your mobile phone to share WiFi hotspot to esp8266.

amansinghaljpr, 2020-01-12:
Yes i did tried your suggestion and the firebase is right in front of me , the data is getting updated but there is no response on the serial monitor .

mobizt, 2020-01-12:
Update esp8266 package (Arduino Core SDK) from board managers to latest version. 

Share your code here.

mobizt, 2020-01-13:
How about making http request and ping test?  Share your test code (ping or http request) and result.

amansinghaljpr, 2020-01-13:
Okay so i did some research and i found the problem. The problem is not with the code but the type of board which i was using  Although the on i have is very similar to nodemcu but its actually not . Maybe the firmware of that chip was making it incompatible.
Also i would recommend you to put this in your repository that this library wont work properly on this particular
[image]

 board . 

This is the image of the board:
[image]

mobizt, 2020-01-13:
The board you showed  from robotdyn uses Espressif ESP8266EX mcu as in Node MCU and Wemos D1 Mini. 

It has 4 MB (32 Mbit flash) as in other boards and nothing is special.

In Arduino IDE, you can choose board Wemos D1 mini or Generic ESP8266 and choose proper flash size and SPIFFS for this board.

If you still get the error, the board you used may defect or the supply voltage is not stable during WiFi operation. 

You can connect board with ypur PC via shorter USB cable or use large capacitor 220 uf or greater connect between 3.3V and GND close to board.

Even you can use ESP-12F WiFi module with this library.