mobizt/Firebase-ESP8266 #78: HELP

amansinghaljpr, 2020-01-13:
Hey so , i am actually a beginner with this library and i am getting overwhelmed by looking at soooo many functions and i need some quick help . Here's what i need : 

I want to send a JSON which will be containing 2 values lets say gps values :
Lat:xxxxxx
long:xxxxx

Now i want to send it every 3 seconds , now how exactly i use your library so that under a perticular node the value gets updated every 3 seconds . Please guide me which functions to use and which examples to put my main focus upon .

mobizt, 2020-01-14:
See the doc at the first page (readme).

mobizt, 2020-01-14:
You need to play with Firebase at console.

Test with Firebase.set or Firebase.setXXX. This will create or overwrite whole data  at node or path you specified.

Test with Firebase.get or Firebase.getXXX. This will read the data from node or path you specified.

Test with example code of JSON on the bottom of this library's first github page. This will show you how to create, edit or modified the JSON object.

Test with Firebase.updateNode. This required JSON object, the data at specified path o node will be updated (just patch to the existing data or create new data if not exist).

Each test, you need to look at Firebase RTDB console on your project website. Look for how data change and its behavior.

Finally, test with stream (monitoring the changes) at the specified path or node. See the steam examples.

closed 2020-01-15 by mobizt