mobizt/Firebase-ESP8266 #79: Stream times out and reconnects causing missed data

jamarshon, 2020-01-22:
**Describe the bug**
A server is periodically (every one minute) updating the database and the ESP8266 is waiting on updates via `setStreamCallback`. Occasionally, the `timeoutCallback` will be called instead of the `dataAvailablecallback` causing missed data while it is reconnecting the stream.

**To Reproduce**
Steps to reproduce the behavior:
1. Use this example [Stream_callback.ino](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Stream_callback/Stream_callback.ino), but clear the void loop function and add a print timestamp for the streamCallback function
2. Look at logs
```
Stream Data1 available...
Time: 2973329 VALUE: 47

Stream Data1 available...
Time: 3033323 VALUE: 48

Stream timeout, resume streaming...

Stream Data1 available...
Time: 3153259 VALUE: 50

Stream Data1 available...
Time: 3213353 VALUE: 51

Stream Data1 available...
Time: 3273247 VALUE: 52

```

**Expected behavior**
There should not be the statement `Stream timeout, resume streaming...` and there should be a call to the `streamCallback` function with the data not being missing:
```
Stream Data1 available...
Time: 3093323 VALUE: 49 (approximately 3033323  + 60*1000)
````

**IDE and its version:**
 - Arduino 1.8.10

**ESP8266 Arduino Core SDK version**
 - Version 2.6.3

prajwal056, 2020-01-22:
How do i send location data to Firebase using NODEMCU

mobizt, 2020-01-22:
There are some known stream callback bugs that have been fixed in this [branch-1](https://github.com/mobizt/Firebase-ESP8266/tree/branch-1).

You can test it.

mobizt, 2020-01-22:
If you still get the same issue, make sure every counter value was successfully set to database not only increase even if it can't set to database due to network problem.

You can use other Firebase clients (web or mobile app to set the couter value to Firebase periodically.

jamarshon, 2020-01-22:
Thanks for the quick response 😄! I don't think it is an issue with setting to the database as I have this event handler in the node admin SDK:
```javascript
admin.database().ref().on("child_changed", function(snapshot) {
  var changedPost = snapshot.val();
  var datetime = new Date();
  console.log('child_changed val:', changedPost, 'at', datetime);
});
```` 
and when I compare the logs:
```
// Serial monitor
Stream Data1 available...
Time: 153361 VALUE: 1
Stream timeout, resume streaming...
Stream Data1 available...
Time: 273554 VALUE: 3
Stream timeout, resume streaming...
Stream Data1 available...
Time: 333396 VALUE: 4
Stream timeout, resume streaming...
Stream Data1 available...
Time: 394974 VALUE: 5
Stream Data1 available...
Time: 453453 VALUE: 6
Stream Data1 available...
Time: 513447 VALUE: 7
Stream Data1 available...
Time: 573542 VALUE: 8
Stream Data1 available...
Time: 633536 VALUE: 9
Stream Data1 available...
Time: 693530 VALUE: 10
Stream Data1 available...
Time: 753524 VALUE: 11
Stream Data1 available...
Time: 813619 VALUE: 12
Stream Data1 available...
Time: 873612 VALUE: 13
Stream Data1 available...
Time: 933607 VALUE: 14
Stream timeout, resume streaming...
Stream timeout, resume streaming...
Stream Data1 available...
Time: 1053533 VALUE: 16

// admin SDK
child_changed val: 1 at 2020-01-22T17:34:55.555Z
success: wrote 1 at 2020-01-22T17:34:55.620Z
child_changed val: 2 at 2020-01-22T17:35:55.556Z
success: wrote 2 at 2020-01-22T17:35:55.617Z
child_changed val: 3 at 2020-01-22T17:36:55.568Z
success: wrote 3 at 2020-01-22T17:36:55.634Z
child_changed val: 4 at 2020-01-22T17:37:55.573Z
success: wrote 4 at 2020-01-22T17:37:55.635Z
child_changed val: 5 at 2020-01-22T17:38:55.574Z
success: wrote 5 at 2020-01-22T17:38:57.306Z
child_changed val: 6 at 2020-01-22T17:39:55.588Z
success: wrote 6 at 2020-01-22T17:39:55.666Z
child_changed val: 7 at 2020-01-22T17:40:55.599Z
success: wrote 7 at 2020-01-22T17:40:55.666Z
child_changed val: 8 at 2020-01-22T17:41:55.601Z
success: wrote 8 at 2020-01-22T17:41:55.669Z
child_changed val: 9 at 2020-01-22T17:42:55.610Z
success: wrote 9 at 2020-01-22T17:42:55.678Z
child_changed val: 10 at 2020-01-22T17:43:55.622Z
success: wrote 10 at 2020-01-22T17:43:55.692Z
child_changed val: 11 at 2020-01-22T17:44:55.629Z
success: wrote 11 at 2020-01-22T17:44:55.689Z
child_changed val: 12 at 2020-01-22T17:45:55.630Z
success: wrote 12 at 2020-01-22T17:45:55.691Z
child_changed val: 13 at 2020-01-22T17:46:55.633Z
success: wrote 13 at 2020-01-22T17:46:55.710Z
child_changed val: 14 at 2020-01-22T17:47:55.640Z
success: wrote 14 at 2020-01-22T17:47:55.724Z
child_changed val: 15 at 2020-01-22T17:48:55.654Z
success: wrote 15 at 2020-01-22T17:48:56.448Z
child_changed val: 16 at 2020-01-22T17:49:55.660Z
success: wrote 16 at 2020-01-22T17:49:55.776Z
```
I noticed that the values 2 and 15 show up in the admin SDK meaning they are in the database but the ESP8266 does not print it out in the `streamCallback` function. The same behavior occurred in branch-1 (missing the value 66).
```
// Serial monitor
Timeouts 0 Stream data
Time: 5552 VALUE: 63

Timeouts 0 Stream data
Time: 53822 VALUE: 64

Timeouts 0 Stream data
Time: 113827 VALUE: 65

Timeouts 290605 Stream data
Time: 233845 VALUE: 67

Timeouts 290605 Stream data
Time: 293843 VALUE: 68

Timeouts 290605 Stream data
Time: 353861 VALUE: 69

Timeouts 290605 Stream data
Time: 413870 VALUE: 70

// admin SDK
child_changed val: 63 at 2020-01-22T18:36:56.158Z
success: wrote 63 at 2020-01-22T18:36:56.221Z
child_changed val: 64 at 2020-01-22T18:37:56.164Z
success: wrote 64 at 2020-01-22T18:37:56.229Z
child_changed val: 65 at 2020-01-22T18:38:56.167Z
success: wrote 65 at 2020-01-22T18:38:56.233Z
child_changed val: 66 at 2020-01-22T18:39:56.175Z
success: wrote 66 at 2020-01-22T18:39:56.241Z
child_changed val: 67 at 2020-01-22T18:40:56.177Z
success: wrote 67 at 2020-01-22T18:40:56.254Z
child_changed val: 68 at 2020-01-22T18:41:56.182Z
success: wrote 68 at 2020-01-22T18:41:56.253Z
child_changed val: 69 at 2020-01-22T18:42:56.197Z
success: wrote 69 at 2020-01-22T18:42:56.261Z
child_changed val: 70 at 2020-01-22T18:43:56.211Z
success: wrote 70 at 2020-01-22T18:43:56.275Z
```
It seems the `timeoutCallback` is related to when data goes missing.  Maybe data is sent exactly when the stream timeouts and is reconnecting? I'll see if I can create a minimum replicable code snippet to share.

mobizt, 2020-01-22:
If data still updates while stream was timed out, the latest data (recent) will be pushed to client when the client reconnected. You need to understand RTDB behavior which are different from MQTT.

mobizt, 2020-01-22:
Time out was not provided by Google Firebase. This library uses keep-alive event message from server to determine the time out when keep-alive event message was not received  within  30 seconds.  The keep-alive event message interval was not actually fixed. It depends on Google.

jamarshon, 2020-01-22:
Here is a minimum code snippet on branch-1, as you can see value 109 shows up in the server.js log but not the example.ino log.
[log.txt](https://github.com/mobizt/Firebase-ESP8266/files/4099558/log.txt)
[example.ino.txt](https://github.com/mobizt/Firebase-ESP8266/files/4099560/example.ino.txt)
[server.js.txt](https://github.com/mobizt/Firebase-ESP8266/files/4099562/server.js.txt)

That is very interesting. I'll dig around and see if I can find anything.

mobizt, 2020-01-23:
Did you read my post above?

> If data still updates while stream was timed out, the latest data (recent) will be pushed to client when the client reconnected. You need to understand RTDB behavior which are different from MQTT .

closed 2020-01-23 by mobizt

jamarshon, 2020-01-23:
> Did you read my post above?
> 
> > If data still updates while stream was timed out, the latest data (recent) will be pushed to client when the client reconnected. You need to understand RTDB behavior which are different from MQTT .

Yes but that statement implies that there should be no missing data because the client will get it once it reconnects but there is missing data observed in the logs.

mobizt, 2020-01-23:
From your log

14:20:56.637 -> Timeouts 0 Stream data
14:20:56.637 -> Time: 566075 VALUE: 107
14:20:56.637 -> 
14:21:56.638 -> Timeouts 0 Stream data
14:21:56.638 -> Time: 626079 VALUE: 108

It shows that the stream event occurred after 120 seconds after previous event. No time out during that.

You need to verify your node client at that time which what's happen. 

Your nodejs code can't confirm the communication delay and internet availability which may take longer time for each send operation.

In addition, counter still to be increased no matter what the last send() function was completed or not.

To be clear you need to wait for async task (send) to finish its operation, then increase the counter value.

Then it should be

var x =0;
function send() {
  admin.database().ref('/test').set(x)
    .then(() => {
      var datetime = new Date();
      console.log('success: wrote', x, 'at', datetime);
        x += 1;
      return null;
    })
    .catch(() => {
      console.log('error');
    });
}

Log the time since it start to send and when it finished.
In esp8266 device you should log the time or timestamp instead of millisecond count.

The library relies on REST which sinc HTTP stream was established, every event happened was push or trig by server as long as no internet interruption.

mobizt, 2020-01-23:
In the case where you need to keep track on every single data, use MQTT instead or use push in firebase to crate new node for every new data.

You can use push in javascript (nodejs) to the child of parent node that being stream in esp8266 to keep every single value also the last or old value that ever pushed.

mobizt, 2020-01-23:
I can prove that there is no bug or issue in streaming with the basic test.

Use two esp8266 devices, one for set data and other one for stream.

Here is the code for set only esp8266 device

```C++
#include <Arduino.h>
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "xxxxxxx.firebaseio.com" 
#define FIREBASE_AUTH "xxxxxxxxxxxxxx"
#define WIFI_SSID "xxxxxxxx"
#define WIFI_PASSWORD "xxxxxxxxxxx"

FirebaseData firebaseData1;
unsigned long pm = 0;
int counter =0;

void setup() {
  Serial.begin(115200);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

  Serial.println();
  Serial.println();
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.print("  OK\nConnected with IP: ");
  Serial.println(WiFi.localIP());

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

}

void loop() {

  if (millis() - pm > 3000 || pm == 0) {
    pm = millis();
    if (Firebase.setInt(firebaseData1, "test123", counter)) {
      Serial.println(counter);
      counter++;

    }
  }

}

```
Its result for 1 hour.
[set_device_result.txt](https://github.com/mobizt/Firebase-ESP8266/files/4101343/set_device_result.txt)

Here is the code for stream only esp8266 device.

```C++
#include <Arduino.h>
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>

#define FIREBASE_HOST "xxxxxxxxx.firebaseio.com" //Without http:// or https://
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxx"
#define WIFI_SSID "xxxxxxxxxxx"
#define WIFI_PASSWORD "xxxxxxxxxxx"

FirebaseData firebaseData1;

int count = 0;

void setup()
{

  Serial.begin(115200);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

  Serial.println();
  Serial.println();
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.print("  OK\nConnected with IP: ");
  Serial.println(WiFi.localIP());

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  Serial.print("Setup stream... ");
  if (Firebase.beginStream(firebaseData1,  "test123"))
    Serial.println("OK");
  else
    Serial.println(firebaseData1.errorReason());

  Firebase.setStreamCallback(firebaseData1, streamCallback, streamTimeoutCallback);
}
void loop()
{

}

void streamCallback(StreamData data)
{

  Serial.println(data.intData());

  if (data.intData() > count + 1 && count > 0)
  {

    Serial.println("-----------------------------------------------");
    Serial.print("Get ");
    Serial.print(data.intData());
    Serial.print(" instead of ");
    Serial.println(count + 1);
    Serial.println("-----------------------------------------------");

  }

  count = data.intData();
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
  {
    Serial.println("Stream timed out!");
  }
}

```
Its result for 1 hour.
[stream_device_result.txt](https://github.com/mobizt/Firebase-ESP8266/files/4101345/stream_device_result.txt)

These esp8266 devices use different network and ISP. One connect via Fiber broadband internet and other via mobile hotspot WiFi from different ISPs. 

You can see from the results, there is no stream event missing.  This library was past the stress test long enough to set to stable state.

jamarshon, 2020-01-23:
Thanks for the response!
First, there is missing data in `set_device_result.txt` as it says
```
11:47:41.175 -> 875
11:47:44.238 -> 876
11:47:47.238 -> 877
11:47:50.191 -> 878
11:47:53.253 -> 879
11:48:08.125 -> 884
11:48:11.107 -> 885
11:48:14.079 -> 886
11:48:17.085 -> 887
```
which means it is missing 880, 881, 882, 883. However, those numbers show up in `stream_device_result.txt` which indicates they are written to DB by the "server" ESP8266 but it doesn't report that it did?
```
11:47:53.550 -> 879
11:47:56.605 -> 880
11:47:59.720 -> 881
11:48:02.420 -> 882
11:48:05.576 -> 883
11:48:08.590 -> 884
```

Second, I agree that my admin SDK send() function could be more careful about the updating of the `counter` variable but regardless, the event handler confirms that the data is correctly written to the database in incremental order as it looks directly at `snapshot.val()` and sees that it is in fact written to the database and that is increasing by 1. Since, it is shown to be correctly in DB then we have to figure out why its not being reported in the stream for the ESP8266.
```javascript
admin.database().ref().on("child_changed", function(snapshot) {
  var changedPost = snapshot.val();
  var datetime = new Date();
  console.log('child_changed val:', changedPost, 'at', datetime);
});
```
Third, you mention "long time delay in loop more than 100 ms" and it seems your code snippet sends data every 3 seconds and mine does one every minute. Could delay length be related to the issue about missing data?
This could also be related to Serial.print just not firing correctly. I'll see if storing the data in a intermediate buffer and then reading it all at once at the end makes a difference.

mobizt, 2020-01-23:
Please don't conclude everything, your assumption is not right.

**First**, that is a bug in Arduino Serial print, maybe serial communication was failed.  
Timestamp data was added by Arduino IDE then it may cause the problem to mix with the real serial data from MCU.

Please look at the code on the set side first. 

It was successfully set then increase the counter. Then the counter value still changes from `11:47:53.253 -> 879` to `11:48:08.125 -> 884` at the next rounds

```C++
   if (Firebase.setInt(firebaseData1, "test123", counter)) {
      Serial.println(counter);
      counter++;
    }
```

You can try it for days or weeks, you still get the same result.

**Second**, there is no debug available from using SDK and you don't know how it works under google implementation web socket. 
Also, Firebase SDK can work while offline then you need to check the real data at console not only from **database().ref().on** event.

The Firebase client with this library works differently. It uses HTTP request through REST APIs then the response was expected from each request and in HTTP stream mode too. It's a real-time update as soon as the server pushes the response back, no data cache as in other SDK clients.

**Third**, the bugs are in polling the stream response part and nothing affected to set, get, push calls. 
If no delay uses inside the loop (master branch), no stream termination from the stream polling routine.

**I think all are clear and no more conversation on this.**

Test my code with real devices, not the cacheable SDK.