mobizt/Firebase-ESP8266 #81: lm35 sensor data is not pushing the data to firebase can any 1 help me

Prajwalk56, 2020-01-23:
#include <ESP8266WiFi.h>                                                // esp8266 library
#include <FirebaseArduino.h>                                             // firebase library
#define FIREBASE_HOST "nodemcu-c371a.firebaseio.com"
#define FIREBASE_AUTH "HwiNzFVToiVM2aMlLJHtLomx6od9ZXCD3WmDHqhK"
#define WIFI_SSID "qwerty"
#define WIFI_PASSWORD "123456"                                  //password of wifi ssid
// initializes or defines the output pin of the LM35 temperature sensor
int outputpin= A0;
//this sets the ground pin to LOW and the input voltage pin to high
void setup() 
{
    Serial.begin(9600);
    delay(1000);                
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);                         //try to connect with wifi
    Serial.print("Connecting to ");
    Serial.print(WIFI_SSID);
    while (WiFi.status() != WL_CONNECTED) 
    {
      Serial.print(".");
      delay(500);
    }
    Serial.println();
    Serial.print("Connected to ");
    Serial.println(WIFI_SSID);
    Serial.print("IP Address is : ");
    Serial.println(WiFi.localIP());                                                      //print local IP address
    Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);                                       // connect to firebase
    Firebase.setString("LED_STATUS", "OFF");                                          //send initial string of led status
}

void loop()       //main loop
{
String path = "/Temp";
int analogValue = analogRead(outputpin);
float millivolts = (analogValue/1024.0) * 3300; //3300 is the voltage provided by NodeMCU
float celsius = millivolts/10;
Serial.print("in DegreeC=   ");
Serial.println(celsius);
//---------- Here is the calculation for Fahrenheit ----------//
float fahrenheit = ((celsius * 9)/5 + 32);
Serial.print(" in Farenheit=   ");
Serial.println(fahrenheit);
//pushing celsius and farenheit to firebase
Firebase.pushString("/Temp", celsius);                                  
delay(1000);
}

mobizt, 2020-01-23:
**First of all you are using other Firebase library.**

This repository is for **FirebaseESP8266** library only.

You should open the issue [here](https://github.com/FirebaseExtended/firebase-arduino/issues).

For your issue, you should use **Firebase.pushFloat("/Temp", celsius);** instead of **Firebase.pushString("/Temp", celsius);**

closed 2020-01-23 by mobizt

Prajwalk56, 2020-01-23:
Thank you .this works