mobizt/Firebase-ESP8266 #82: How to send the firebase data to the google sheets..??

Prajwalk56, 2020-01-23:
#include <ESP8266WiFi.h>                                                // esp8266 library
#include <FirebaseArduino.h>                                             // firebase library
#define FIREBASE_HOST "nodemcu-c371a.firebaseio.com"
#define FIREBASE_AUTH "HwiNzFVTViVM2aMlLJHtLomx6L39ZXCD3WmDHqhK"
#define WIFI_SSID "natu"
#define WIFI_PASSWORD "12345678"                                  //password of wifi ssid
// initializes or defines the output pin of the LM35 temperature sensor
int outputpin= A0;
//this sets the ground pin to LOW and the input voltage pin to high
void setup() 
{
   String path = "/Temp";
    Serial.begin(9600);
    delay(1000);                
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);                         //try to connect with wifi
    Serial.print("Connecting to ");
    Serial.print(WIFI_SSID);
    while (WiFi.status() != WL_CONNECTED) 
    {
      Serial.print(".");
      delay(500);
    }
    Serial.println();
    Serial.print("Connected to ");
    Serial.println(WIFI_SSID);
    Serial.print("IP Address is : ");
    Serial.println(WiFi.localIP());                                                      //print local IP address
    Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);                                       // connect to firebase
    Firebase.setString("LED_STATUS", "OFF");                                          //send initial string of led status
int temp=analogRead(A0);
Serial.println(temp);
temp=temp*0.3222656;//((3.3/1024)*1000)/10
for (uint8_t i = 0; i < 20; i++)
{
Serial.print("in DegreeC=   ");
Serial.println(temp);
//---------- Here is the calculation for Fahrenheit ----------//
float fahrenheit = ((temp * 9)/5 + 32);
Serial.print(" in Farenheit=   ");
Serial.println(fahrenheit);
//pushing celsius and farenheit to firebase
Firebase.pushFloat("/Temp/fahre",fahrenheit); 
 Firebase.pushFloat("/Temp",temp);                               
delay(1000);
}
}
void loop()       //main loop
{
}

mobizt, 2020-01-23:
As I said that is not my library.

mobizt, 2020-01-23:
You have to open the issue here https://github.com/FirebaseExtended/firebase-arduino

closed 2020-01-23 by mobizt