mobizt/Firebase-ESP8266 #85: Reads wrong json array path

sarkrui, 2020-02-10:
**Describe the bug**
firebaseJson.get(localJsonData, "/path/") reads a shifted /path/ when no **real** value under the destinated path

**To Reproduce**
```c++
void readLocalColorQueue() {
  //Fetch JSON from Firebase
  Firebase.get(firebaseData, "/fireflies");
  FirebaseJson &localJson = firebaseData.jsonObject();
  FirebaseJsonData localJsonData;
  localJsonData = firebaseData.jsonData();
  localJson.get(localJsonData, device_id + "/colorQueue");

  FirebaseJsonArray queueArr2;
  localJsonData.getArray(queueArr2);

  //store localQueue array in blinkColorQueue[]
  actualQueueSize = queueArr2.size();

  for (size_t i = 0; i < (sizeof(blinkColorQueue) / sizeof(blinkColorQueue[0])); i++) {
    //Serialize with a 404 index;
    blinkColorQueue[i] = 404;
  }

  for (size_t i = 0; i < actualQueueSize; i++) {
    queueArr2.get(localJsonData, i);
    blinkColorQueue[i] = localJsonData.intValue;
  }
  //print out blinkColorQueue
  for (size_t i = 0; i < actualQueueSize; i++) {
    Serial.print(blinkColorQueue[i]);
    Serial.print(",");
  }

  //change local press state
  //needs to be uncommented if running with displayShow();
  localJson.set(device_id + "/tapped", true);

  String jsonStr;
  localJson.toString(jsonStr, true);
  Serial.println(jsonStr);

  //Push updated colorQueue
  Firebase.updateNode(firebaseData, "/fireflies", localJson);
  Serial.println("Push JSON");
}
```

**Expected behavior**
It should read the destinated path despite it might be empty.

**Screenshots**
[image]

**IDE and its version:**
 - Arduino, PlatformIO
 - Version 1.8.10

**ESP8266 Arduino Core SDK version**
 - Version 2.6.2

**_ Last but not least, thank you for praying for China, Wuhan, and the people. _**

sarkrui, 2020-02-10:
Seemed to be the same bug as #68

mobizt, 2020-02-10:
Sorry, I don't understand.

> reads a shifted /path/ when no real value under the destinated path

If the issue related to FirebaseJson, please provide the parsing example like this

1. Serialized JSON string and the path to parse that causes an error when parsing or modifying.

2. Serialized JSON string and the path to parse that is ok when parsing or modifying.

*The serialized JSON string needs to be set to FirebaseJSON via **setJsonData** to test instead of real data from RTDB.

sarkrui, 2020-02-10:
**JSON Body** 
```javascript
{
  "device_1": {
   (no colorQueue values here)
    "irrelevantVariable": true
  },
  "device_2": {
    "colorQueue": {
      "1": 2,
      "2": 3,
      "0": 1
    }
  }
}
```
**Serialized by**
```c++
json.set("/device_1/irrelevantVariable", true);
json.set("/device_2/colorQueue/[0]", 1);
json.set("/device_2/colorQueue/[1]", 2);
json.set("/device_2/colorQueue/[2]", 3);
```
but if I try accessing the array under `/device_1/colorQueue` by
```c++
json.get(JsonData, "/device_1/colorQueue");
JsonData.getArray(arr1);
```
it returns `1, 2, 3` values from `/device_2/colorQueue`.
However, when 
```c++
json.set("/device_1/irrelevantVariable", true);
```
is removed, the issue disappears.

_Arduino code attached below_
[getJsonArray_local.ino.zip](https://github.com/mobizt/Firebase-ESP8266/files/4180521/getJsonArray_local.ino.zip)

sarkrui, 2020-02-10:
_(I should have opened the issue at `FirebaseJson` repository. But here I wanna append some code. )_

```c++
#include <Arduino.h>
#include "FirebaseJson.h"
void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println("Get JSON");
  FirebaseJson json;

  json.set("/device_1/colorQueue/[0]", 1);
  json.set("/device_1/colorQueue/[1]", 2);
  json.set("/device_1/colorQueue/[2]", 3);

  //  json.set("/device_2/colorQueue/[0]", 4);
  //  json.set("/device_2/colorQueue/[1]", 5);
  //  json.set("/device_2/colorQueue/[2]", 6);
  json.set("/device_2/irrelevantVariable", true);

  json.set("/device_3/colorQueue/[0]", 7);
  json.set("/device_3/colorQueue/[1]", 8);
  json.set("/device_3/colorQueue/[2]", 9);

  FirebaseJsonData JsonData;
  JsonData.getJSON(json);
  }
}
```
**Serial Outputs**
```
Get JSON
array 1 value: 1
array 1 value: 2
array 1 value: 3
array 2 value: 7
array 2 value: 8
array 2 value: 9
array 3 value: 7
array 3 value: 8
array 3 value: 9
```
the values of array 2 should be: null.

mobizt, 2020-02-10:
Ok, I found the bug in multiple root elements parsing when the user tries to parse the invalid path and needs to be fixed.

mobizt, 2020-02-10:
Now the issue was fixed in the new update on GitHub.
Thanks for reporting the bugs.

sarkrui, 2020-02-10:
Thank you! It now works like a charm.

closed 2020-02-10 by sarkrui