mobizt/Firebase-ESP8266 #86: Incompatibility with FirebaseESP32

dg321, 2020-02-13:
**When adding FirebaseESP8266 and FirebaseESP32 in the same project, an error is generated.
The source code I am using is as follows:**

```
#ifdef ESP32
#include <WiFi.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <FirebaseESP32.h>
#elif defined(ESP8266)
#include <TaskSchedulerDeclarations.h>      // Processamento
#include <ESP8266WiFi.h>                    // https://github.com/esp8266/Arduino
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>                    // Include the mDNS library
#include <FirebaseESP8266.h>
#endif
```

**The error generated is below:**

```
Processing nodemcu-32s (platform: espressif32; board: nodemcu-32s; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 1.11.2 > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, 
olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-openocd-esp32 1.1000.20190708 (10.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.14.1
|-- <BibIoT>
|   |-- <ArduinoJson> 6.14.1
|   |-- <ArduinoOTA> 1.0
|   |   |-- <Update> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Firebase ESP32 Client> 3.6.4
|   |   |-- <FS> 1.0
|   |   |-- <SD(esp32)> 1.0.5
|   |   |   |-- <FS> 1.0
|   |   |   |-- <SPI> 1.0
|   |   |-- <SPIFFS> 1.0
|   |   |   |-- <FS> 1.0
|   |   |-- <HTTPClient> 1.2
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <WiFiClientSecure> 1.0
|   |   |   |   |-- <WiFi> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <TaskScheduler> 3.1.2
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <Firebase ESP8266 Client> 2.8.0
|   |-- <FS> 1.0
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <SPI> 1.0
Building in release mode
Compiling .pio\build\nodemcu-32s\src\main.cpp.o
Compiling .pio\build\nodemcu-32s\lib26d\Firebase ESP8266 Client_ID6247\FirebaseESP8266.cpp.o
Compiling .pio\build\nodemcu-32s\lib26d\Firebase ESP8266 Client_ID6247\FirebaseESP8266HTTPClient.cpp.o
In file included from C:\Users\Diego Grosmann\.platformio\lib\Firebase ESP8266 Client_ID6247\src\FirebaseESP8266HTTPClient.cpp:29:0:
C:\Users\Diego Grosmann\.platformio\lib\Firebase ESP8266 Client_ID6247\src\FirebaseESP8266HTTPClient.h:55:2: error: #error Your ESP8266 Arduino 
Core SDK is outdated, please update. From Arduino IDE go to Boards Manager and search 'esp8266' then select the latest version.
 #error Your ESP8266 Arduino Core SDK is outdated, please update. From Arduino IDE go to Boards Manager and search 'esp8266' then select the latest version.
```

mobizt, 2020-02-13:
Why this is an issue?

If you're new to esp32 and esp8266, please learn how to program it, and what is the difference between architecture and their core libraries.

closed 2020-02-13 by mobizt

mobizt, 2020-02-13:
Your macro definition is wrong here
`#elif defined(ESP8266)`

It should be
```C++
#ifdef ESP8266

#endif

#ifdef ESP32

#endif
```

Please keep in mind that the FirebaseESP32 and ESP8266 have some different functions due to different WiFi client and SSL/TTLS provider functionalities that's why the library should not conclude into one library to support all MCUs.

dg321, 2020-02-14:
When I execute the following code everything works normally
```
#undef ESP32

#ifdef ESP32
//#include <FirebaseESP32.h>
#endif

#ifdef ESP8266
#include <FirebaseESP8266.h>
#endif
```

But when the code below is executed the error occurs
```
#undef ESP32

#ifdef ESP32
#include <FirebaseESP32.h>
#endif

#ifdef ESP8266
#include <FirebaseESP8266.h>
#endif
```

For some reason the compiler is running #include <FirebaseESP32.h>. Other codes within the #ifdef ESP32 #endif directive are not executed.

If I uninstall the FirebaseESP32 library, both codes work normally.

This behavior is unexpected. And I would like to see a possibility to work around the problem.

_I used #undef ESP32 just to test and ensure that this directive had not been defined by mistake._

mobizt, 2020-02-14:
I don't know what you're trying to do.
The macro ESP32 defined in ESP32 core SDK while ESP8266 defined in ESP8266 core SDK.

And you make unclear and conflict in the first and the last statements.
> For some reason the compiler is running #include <FirebaseESP32.h>. Other codes within the #ifdef ESP32 #endif directive are not executed.

No need to use undef, in PIO just change the platform.ini between platform = espressif8266 and platform = espressif32 and board when compile.

mobizt, 2020-02-14:
You need to check your program flow of preprocessor directives, which code you're trying to include or compile and which one you want to skip from compile under the different platforms.

Then you can add the platform-specific library you want.

dg321, 2020-02-14:
Tentei verificar o fluxo das diretivas de pré-processador, mas não consegui identificar o problema.

Estou utilizado Windows 10, Visual Studio Code 1.42.1 e PlatformIO Home 3.1.0 Core 4.2.0.

Aqui está o codigo que está apresentando problemas. 

[TestFirebase.zip](https://github.com/mobizt/Firebase-ESP8266/files/4204395/TestFirebase.zip)

mobizt, 2020-02-14:
English?

dg321, 2020-02-14:
I tried to check the flow of the preprocessor policies, but I couldn't identify the problem.

I am using Windows 10, Visual Studio Code 1.42.1 and PlatformIO Home 3.1.0 Core 4.2.0.

Here is the code that is having problems.

[TestFirebase.zip](https://github.com/mobizt/Firebase-ESP8266/files/4204436/TestFirebase.zip)

mobizt, 2020-02-14:
Why you need to include both esp8266 and esp32 library into your project?

dg321, 2020-02-14:
251/5000
I am creating a library to use in various projects. I would like to create just one library to show duplicate codes.

I can create two libraries without any problems. But I would prefer to have only one that works on both platforms.

mobizt, 2020-02-14:
Post the result of compile here.

dg321, 2020-02-14:
```

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 2.3.3 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20603.200130 (2.6.3)     
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Firebase ESP32 Client> 3.6.4
|   |-- <SD(esp8266)> 2.0.0
|   |   |-- <SDFS> 0.1.0
|   |   |   |-- <ESP8266SdFat> 1.1.0
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <SPI> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Firebase ESP8266 Client> 2.8.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <SD(esp8266)> 2.0.0
|   |   |-- <SDFS> 0.1.0
|   |   |   |-- <ESP8266SdFat> 1.1.0
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
Building in release mode
Compiling .pio\build\nodemcuv2\src\main.cpp.o
Generating LD script .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\nodemcuv2\libbbb\SPI\SPI.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\FatLib\FatFile.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\FatLib\FatFileLFN.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\FatLib\FatFilePrint.cpp.o
Archiving .pio\build\nodemcuv2\libbbb\libSPI.a
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\FatLib\FatFileSFN.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\FatLib\FatVolume.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\FatLib\FmtNumber.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\FatLib\StdioStream.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\FatLib\fstream.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\FatLib\istream.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\FatLib\ostream.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\MinimumSerial.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\SdCard\SdSpiCard.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\SdCard\SdSpiCardEX.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\SdCard\SdioCardEX.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\SdCard\SdioTeensy.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\SpiDriver\SdSpiESP8266.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\SpiDriver\SdSpiParticle.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\SpiDriver\SdSpiSAM3X.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\SpiDriver\SdSpiSTM32.cpp.o
Compiling .pio\build\nodemcuv2\libfb6\ESP8266SdFat\SpiDriver\SdSpiTeensy3.cpp.o
Compiling .pio\build\nodemcuv2\libd19\SDFS\SDFS.cpp.o
Compiling .pio\build\nodemcuv2\libe39\SD\SD.cpp.o
Compiling .pio\build\nodemcuv2\libbc2\ESP8266WiFi\BearSSLHelpers.cpp.o
Archiving .pio\build\nodemcuv2\libfb6\libESP8266SdFat.a
Archiving .pio\build\nodemcuv2\libe39\libSD.a
Compiling .pio\build\nodemcuv2\libbc2\ESP8266WiFi\CertStoreBearSSL.cpp.o
Compiling .pio\build\nodemcuv2\libbc2\ESP8266WiFi\ESP8266WiFi.cpp.o
Archiving .pio\build\nodemcuv2\libd19\libSDFS.a
Compiling .pio\build\nodemcuv2\libbc2\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Compiling .pio\build\nodemcuv2\libbc2\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pio\build\nodemcuv2\libbc2\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
[... 23 lines cut ...]

 #include <WiFi.h>
                  ^
compilation terminated.
*** [.pio\build\nodemcuv2\lib5fd\Firebase ESP32 Client_ID6217\FirebaseESP32.cpp.o] Error 1
In file included from C:\Users\Diego Grosmann\.platformio\lib\Firebase ESP32 Client_ID6217\src\FirebaseESP32HTTPClient.cpp:36:0:
C:\Users\Diego Grosmann\.platformio\lib\Firebase ESP32 Client_ID6217\src\FirebaseESP32HTTPClient.h:37:24: fatal error: HTTPClient.h: No such file or directory

********************************************************************
* Looking for HTTPClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:HTTPClient.h"
* Web  > https://platformio.org/lib/search?query=header:HTTPClient.h
*
********************************************************************

 #include <HTTPClient.h>
                        ^
compilation terminated.
*** [.pio\build\nodemcuv2\lib5fd\Firebase ESP32 Client_ID6217\FirebaseESP32HTTPClient.cpp.o] Error 1
========================================================= [FAILED] Took 47.55 seconds =========================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

dg321, 2020-02-14:
Note that in the list of dependencies there is Firebase ESP32 Client.

However if I uninstall PlatformIO's FirebaseESP32 library the code compiles.

mobizt, 2020-02-14:
It's a PlatformIO problem. I will check and reply later

dg321, 2020-02-14:
I also believe that the problem is with PlatformIO.
Thank you very much, I will wait.

mobizt, 2020-02-15:
The LDF search bugs still exist in the current PlatformIO version.
You have two choices to prevent these bugs.

**1. Use the pio-ldf branch which I add the macro to skip the LDF search into header and CPP files.**

https://github.com/mobizt/Firebase-ESP8266/tree/pio-ldf
https://github.com/mobizt/Firebase-ESP32/tree/pio-ldf

**2. Edit platformio.ini**

To compile for ESP32, set the lib_ignore option to
lib_ignore = Firebase ESP8266 Client

To compile for ESP8266, set the lib_ignore option to
lib_ignore = Firebase ESP32 Client

dg321, 2020-02-15:
Thank you!
Both options work perfectly.