mobizt/Firebase-ESP8266 #88: Help Firebase Realtime Database periodic Fingerprint changes

danilomna, 2020-02-29:
Hi @mobizt

I´m planning move my IoT projects to use your library.

I have a doubt and maybe you can clarify to me the following:
I´m using Firebase realtime database on IOT device (ESP8266).
then two days ago device stopped communicating with firebase, after 
some time of research I found the firebase "fingerprint" has changed
and this was the reason my devices stopped working.

The library I´m using so far (not from yours) need to replace the fingerprint
in source code and update firmware on all devices to make them get back to work!!!

This is a terrible,  lets think about chaos scenary, thousand of IoT devices
deployied on field and all stop working at same time because firmware had 
not valid fingerprint anymore. This would be a complete chaos. :(

I would like to know how your library deals with this fingerprint issue??

How could we workaround this?? (to IOT devices never stop working becouse fingerprint changes).

cheers, 

@danilomna

mobizt, 2020-02-29:
This Firebase library uses WiFiClientSecure for network connection and no fingerprint or root certificate required. 

The library also provides a root certificate connection for the secure connection.

The API interfaces and usage of the library are different from other Firebase libraries.

For the usage please see the [examples](https://github.com/mobizt/Firebase-ESP8266/tree/master/examples) and readme doc.

mobizt, 2020-02-29:
The one reason that I create this library is the existing library from FirebaseExtended uses the HTTP client that required fingerprint to connect which is unacceptable for me.

danilomna, 2020-02-29:
> The one reason that I create this library is the existing library from FirebaseExtended uses the HTTP client that required fingerprint to connect which is unacceptable for me.

@mobizt   this is all that I want to heard:

"required fingerprint to connect which is unacceptable for me."

I´m going to move all projects to your library right now.

You job this is absolutelly awesome, very well done, congratulations. 

Allow me pay you a coffe? (paypal, btc)

mobizt, 2020-02-29:
Thanks and I appreciate your offer.
I'm happy if you like it. 
This is enough for me.

closed 2020-02-29 by danilomna

thetechguy2016, 2020-06-16:
> This Firebase library uses WiFiClientSecure for network connection and no fingerprint or root certificate required.
>
> The library also provides a root certificate connection for the secure [...]

Hi @mobizt 
is this library automatically get new certificate when rootCA certificate will expire. or we need to manually update?

or if it update rootCA automatically, can you send me the file reference to study.

Thanks

mobizt, 2020-06-16:
@thetechguy2016 

The root ca certificate (issued by the certificate authority company) is the top level of certificate chain which server (Firebase) SSL certificate was signed by ca and is in this certificate chain. 

The root certificate (rot ca certificate )  is trusted by the server and often used in verification and validation between the server and clients (apps) instead of the server SSL certificate which often expires and renews.

You need to check the expiry date of root certificate yourself with tool e.g. OpenSSL and update it to the device.

The library allows users to include the root CA certificate data or file which you can export from the web browser when you visit the Firebase server.

https://comodosslstore.com/resources/what-is-a-root-ca-certificate-and-how-do-i-download-it/

thetechguy2016, 2020-06-18:
Oo thanks, but i really don't understand that if the certificate got expire, how do we get update the certificate because we also use firebase Storage for OTA which further use the rootCa for the same.

mobizt, 2020-06-18:
Certificate keeps on server and belong to server, you can't do anything with it e.g. renew and replace because you are not server administrator and can't access the server to manage this.

The server admin may invoke, renew, replace that certificate as he want without notice, you can't do anything with certificate when it expired.

You need to visit to the server periodically (every months, two months or a year) to check the certificate of that server.
The server returns certificate information to client for validation and data exchanges. 

You can use web browser to download that server certificate or all certificate in the chain included root certificate and you can do automatically with OpenSSL tool which you can download and extract information which you can write the command script in Windows and Linux to control the OpenSSL tool.

You need need to manage to update new certificate data to your code yourself. 

You can search google and study more about SSL certificate.