mobizt/Firebase-ESP8266 #89: How to add ".sv" Firebase server timestamp to document

gyromech, 2020-03-01:
Hi,
I have the same behavior like issue #57 when I tried to add ".sv". I tried manually with Curl this same request and it works. I've tried push() and pushJSON() method.
When replacing ".sv" by anything else in my code, server response is ok.
There is some tricks for more verbose output from your library ?

My code :
```
//HEADER
    FirebaseData firebaseData;
    #define FIREBASE_HOST "myserver.firebaseio.com"
    #define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
//SETUP
    Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
    firebaseData.setBSSLBufferSize(1024, 1024);
    firebaseData.setResponseSize(1024);
    FirebaseJson jsonA;
    FirebaseJson jsonB;
    jsonA.clear();
    jsonB.clear();
    jsonA.add("temperature", t);
    jsonA.add("humidite", h);
    jsonA.add("vcc", v_cal);
    jsonB.add(".sv", "timestamp");
    jsonA.add("timestamp", jsonB);

    Firebase.push(firebaseData, "/sensors/TEST", jsonA)
```
`Firebase.push` return False.
`firebaseData.erroReason` give me : `internal server error, Unexpected server error`

data sent (`jsonA.toString(jsonStr, true)`):
```
{
    "temperature": 26.299999237,
    "humidite": 64,
    "vcc": 3.341234446,
    "timestamp": {
        ".sv": "timestamp"
    }
}
```

With Curl, this is the request : 

```curl -v -X POST --header "Authorization: key=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx" --header "Content-Type: application/json" https://myserver.firebaseio.com/sensors/TEST.json -d "{\"temperature\": 26.299999237,\"humidite\": 64,\"vcc\": 3.341234446,\"timestamp\": {\".sv\": \"timestamp\"}}"```

and the response : 
```
> POST /sensors/TEST.json HTTP/1.1
> Host: myserver.firebaseio.com
> User-Agent: curl/7.55.1
> Accept: */*
> Authorization: key=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
> Content-Type: application/json
> Content-Length: 97
>
* upload completely sent off: 97 out of 97 bytes
* schannel: client wants to read 102400 bytes
* schannel: encdata_buffer resized 103424
* schannel: encrypted data buffer: offset 0 length 103424
* schannel: encrypted data got 352
* schannel: encrypted data buffer: offset 352 length 103424
* schannel: decrypted data length: 323
* schannel: decrypted data added: 323
* schannel: decrypted data cached: offset 323 length 102400
* schannel: encrypted data buffer: offset 0 length 103424
* schannel: decrypted data buffer: offset 323 length 102400
* schannel: schannel_recv cleanup
* schannel: decrypted data returned 323
* schannel: decrypted data buffer: offset 0 length 102400
< HTTP/1.1 200 OK
< Server: nginx
< Date: Sun, 01 Mar 2020 10:54:11 GMT
< Content-Type: application/json; charset=utf-8
< Content-Length: 31
< Connection: keep-alive
< Access-Control-Allow-Origin: *
< Cache-Control: no-cache
< Strict-Transport-Security: max-age=31556926; includeSubDomains; preload
<
{"name":"-M1KiqQ7GhsO0iLoJLwC"}* Connection #0 to host myserver.firebaseio.com left intact
```
BR

mobizt, 2020-03-01:
Thank you for reporting this issue and clarify your test result.

This is the bug due to ETag header was set by default in the library while server timestamp was included in JSON data. This causes the internal server error response.

Now it fixed, please update the library (git version).

gyromech, 2020-03-01:
Thank you! It now works like a charm.
👍

closed 2020-03-01 by gyromech