mobizt/Firebase-ESP8266 #92: stream : connection refused on users path

phoenixxd, 2020-03-09:
**Describe the bug**
The beginStream(firebaseData, path) always says **connection refused**. This happens when the path begins with "/users/" like for example for a given user the path is "/users/oR58vG1jbTWJyDISScMAVHCFZf62"

The example works when the path is something like "/Test/Stream/" or "/Timer/oR58vG1jbTWJyDISScMAVHCFZf62"
The data is updated upon changes

I have gone through similar issues [5](https://github.com/mobizt/Firebase-ESP8266/issues/5), [7](https://github.com/mobizt/Firebase-ESP8266/issues/7) [56](https://github.com/mobizt/Firebase-ESP32/issues/56)
But nothing helped out there.

**To Reproduce**
Since it is happening for a specific path only, I cannot really determine what difference is present which results in such behaviour. I though maybe read/write rules are applied on the users/ path thats why. I removed the restrictions but still its the same.

**Expected behavior**
It should have worked for every path thats in the same RTDB

**Output logs - WIFI debug mode **
With users as root node
SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
Connecting to Wi-Fi.wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with **************, channel 2
dhcp client start...
wifi evt: 0
..............ip:192.168.1.2,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 3

Connected with IP: 192.168.1.2

[hostByName] request IP for: **************.firebaseio.com
[hostByName] Host: **************.firebaseio.com IP: 35.201.97.85
------------------------------------
Can't read stream data...
REASON: connection refused
------------------------------------

pm open,type:2 0
[hostByName] request IP for: **************.firebaseio.com
[hostByName] Host: **************.firebaseio.com IP: 35.201.97.85
Stream timeout, resume streaming...

------------------------------------
Can't read stream data...
REASON: connection refused
------------------------------------

[hostByName] request IP for: **************.firebaseio.com
[hostByName] Host: **************.firebaseio.com IP: 35.201.97.85
Stream timeout, resume streaming...

------------------------------------
Can't read stream data...
REASON: connection refused
------------------------------------

With Timer/Test as root node
`SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
Connecting to Wi-Fi.wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with LDrago, channel 2
dhcp client start...
wifi evt: 0
...........................ip:192.168.1.2,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 3

Connected with IP: 192.168.1.2

[hostByName] request IP for: **************.firebaseio.com
[hostByName] Host: **************.firebaseio.com IP: 35.201.97.85
pm open,type:2 0
------------------------------------
Stream Data available...
STREAM PATH: /Test/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: json
EVENT TYPE: put
VALUE: 
Pretty printed JSON data:
{
    "String": "this is a new string",
    "mydata": "hellokhgvjhv thete newod"
}

Iterate JSON data:

0, Type: object, Key: String, Value: this is a new string
1, Type: object, Key: mydata, Value: hellokhgvjhv thete newod
------------------------------------

Stream timeout, resume streaming...

Stream timeout, resume streaming...

Stream timeout, resume streaming...

Stream timeout, resume streaming...

Stream timeout, resume streaming...

Stream timeout, resume streaming...

Stream timeout, resume streaming...`

**IDE and its version:**
 - Arduino
 - Version [1.8.9]

**ESP8266 Arduino Core SDK version**
 - Version [2.5.2]

**Additional context**
Snippet I was using is modified from the example of stream itself.
`	#include "FirebaseESP8266.h"
	#include <ESP8266WiFi.h>

	#define FIREBASE_HOST "xxxxxxxxx.firebaseio.com"
	#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
	#define WIFI_SSID "xxxxxxx"
	#define WIFI_PASSWORD "xxxxxxxxx"

	//Define FirebaseESP8266 data object
	FirebaseData firebaseData;
	String path = "/users/oR58vG1jbTWJyDISScMAVHCFZf62";
	void printResult(FirebaseData &data);

	void setup()
	{

		Serial.begin(115200);

		WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
		Serial.print("Connecting to Wi-Fi");
		while (WiFi.status() != WL_CONNECTED)
		{
			Serial.print(".");
			delay(300);
		}
		Serial.println();
		Serial.print("Connected with IP: ");
		Serial.println(WiFi.localIP());
		Serial.println();

		Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
		Firebase.reconnectWiFi(true);

		if (!Firebase.beginStream(firebaseData, path))
		{
			Serial.println("------------------------------------");
			Serial.println("Can't begin stream connection...");
			Serial.println("REASON: " + firebaseData.errorReason());
			Serial.println("------------------------------------");
			Serial.println();
		}
	}

	void loop()
	{

		if (!Firebase.readStream(firebaseData))
		{
			Serial.println("------------------------------------");
			Serial.println("Can't read stream data...");
			Serial.println("REASON: " + firebaseData.errorReason());
			Serial.println("------------------------------------");
			Serial.println();
		}

		if (firebaseData.streamTimeout())
		{
			Serial.println("Stream timeout, resume streaming...");
			Serial.println();
		}

		if (firebaseData.streamAvailable())
		{
			Serial.println("------------------------------------");
			Serial.println("Stream Data available...");
			Serial.println("STREAM PATH: " + firebaseData.streamPath());
			Serial.println("EVENT PATH: " + firebaseData.dataPath());
			Serial.println("DATA TYPE: " + firebaseData.dataType());
			Serial.println("EVENT TYPE: " + firebaseData.eventType());
			Serial.print("VALUE: ");
			printResult(firebaseData);
			Serial.println("------------------------------------");
			Serial.println();
		}
	}

	void printResult(FirebaseData &data)
	{

		if (data.dataType() == "int")
			Serial.println(data.intData());
		else if (data.dataType() == "float")
			Serial.println(data.floatData(), 5);
		else if (data.dataType() == "double")
			printf("%.9lf\n", data.doubleData());
		else if (data.dataType() == "boolean")
			Serial.println(data.boolData() == 1 ? "true" : "false");
		else if (data.dataType() == "string")
			Serial.println(data.stringData());
		else if (data.dataType() == "json")
		{
			Serial.println();
			FirebaseJson &json = data.jsonObject();
			//Print all object data
			Serial.println("Pretty printed JSON data:");
			String jsonStr;
			json.toString(jsonStr, true);
			Serial.println(jsonStr);
			Serial.println();
			Serial.println("Iterate JSON data:");
			Serial.println();
			size_t len = json.iteratorBegin();
			String key, value = "";
			int type = 0;
			for (size_t i = 0; i < len; i++)
			{
				json.iteratorGet(i, type, key, value);
				Serial.print(i);
				Serial.print(", ");
				Serial.print("Type: ");
				Serial.print(type == JSON_OBJECT ? "object" : "array");
				if (type == JSON_OBJECT)
				{
					Serial.print(", Key: ");
					Serial.print(key);
				}
				Serial.print(", Value: ");
				Serial.println(value);
			}
			json.iteratorEnd();
		}
		else if (data.dataType() == "array")
		{
			Serial.println();
			//get array data from FirebaseData using FirebaseJsonArray object
			FirebaseJsonArray &arr = data.jsonArray();
			//Print all array values
			Serial.println("Pretty printed Array:");
			String arrStr;
			arr.toString(arrStr, true);
			Serial.println(arrStr);
			Serial.println();
			Serial.println("Iterate array values:");
			Serial.println();
			for (size_t i = 0; i < arr.size(); i++)
			{
				Serial.print(i);
				Serial.print(", Value: ");

				FirebaseJsonData &jsonData = data.jsonData();
				//Get the result data from FirebaseJsonArray object
				arr.get(jsonData, i);
				if (jsonData.typeNum == JSON_BOOL)
					Serial.println(jsonData.boolValue ? "true" : "false");
				else if (jsonData.typeNum == JSON_INT)
					Serial.println(jsonData.intValue);
				else if (jsonData.typeNum == JSON_DOUBLE)
					printf("%.9lf\n", jsonData.doubleValue);
				else if (jsonData.typeNum == JSON_STRING ||
						jsonData.typeNum == JSON_NULL ||
						jsonData.typeNum == JSON_OBJECT ||
						jsonData.typeNum == JSON_ARRAY)
					Serial.println(jsonData.stringValue);
			}
		}
	}
`

mobizt, 2020-03-10:
I test your code and did not find any bugs.

Please update the library.

Here the result when I set the value at the node **/users/oR58vG1jbTWJyDISScMAVHCFZf62**

```C++
Connecting to Wi-Fi.......
Connected with IP: 192.168.137.124

------------------------------------
Stream Data available...
STREAM PATH: /users/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: null
EVENT TYPE: put
VALUE: ------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /users/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: int
EVENT TYPE: put
VALUE: 888
------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /users/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: int
EVENT TYPE: put
VALUE: 999
------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /users/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: int
EVENT TYPE: put
VALUE: 1
------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /users/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: int
EVENT TYPE: put
VALUE: 2
------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /users/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: int
EVENT TYPE: put
VALUE: 3
------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /users/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: int
EVENT TYPE: put
VALUE: 4
------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /users/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: string
EVENT TYPE: put
VALUE: hello
------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /users/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: boolean
EVENT TYPE: put
VALUE: true
------------------------------------

------------------------------------
Stream Data available...
STREAM PATH: /users/oR58vG1jbTWJyDISScMAVHCFZf62
EVENT PATH: /
DATA TYPE: double
EVENT TYPE: put
VALUE: 123.456789000
------------------------------------

```

closed 2020-03-10 by mobizt

phoenixxd, 2020-03-10:
With the latest library update, it started working.
Thanks for you amazing library!! :) :)