mobizt/Firebase-ESP8266 #93: Compilation issue in 2.8.4

phoenixxd, 2020-03-10:
**Describe the bug**
There is a compilation issue, with String class not having **isEmpty()** method in the version 2.8.4

**To Reproduce**
Upon compilation
> C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp: In member function 'void FirebaseData::addNodeList(const String*)':
> 
> C:\Users\ADMIN\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp:5816:26: error: 'const class String' has no member named 'isEmpty'
> 
>          if(!childPath[i].isEmpty() && childPath[i] !="/")

**IDE and its version:**
 - Arduino
 - Version [1.8.9]

**ESP8266 Arduino Core SDK version**
 - Version [2.5.2]

**Solution**
I replaced line 5816 in FirebaseESP8266.cpp with the following and it compiled without any issue.
`if(childPath[i].length() != 0 && childPath[i] !="/")`

mobizt, 2020-03-10:
Thanks for inform. I will remove this isEmpty() function and update it soon.

phoenixxd, 2020-03-10:
I have raised a PR for this.

mobizt, 2020-03-10:
Sorry, I just update it already. And I really thank for PR.

phoenixxd, 2020-03-10:
no problem.

closed 2020-03-10 by phoenixxd