mobizt/Firebase-ESP8266 #95: HELP

Mukesh7631, 2020-03-11:
can I access locked data(Secured by ID and Password) from firestore by ESP8266.

mobizt, 2020-03-11:
Yes, using database secrets will by-pass all security rules.

mobizt, 2020-03-11:
The database secret used in the begin function as in this example https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Basic/Basic.ino#L21

Mukesh7631, 2020-03-11:
Sorry, I can't get a solution, can you give me a simple example.

mobizt, 2020-03-11:
Ok, I get it, you're talking about Cloud Firestore and the library is for Firebase RTDB which nothing related.

Mukesh7631, 2020-03-11:
[image]
 
I created a database in locked mode, Now I can't access my RTDB by FIREBASE_HOST & FIREBASE_AUTH.
If I connect the same code to test mode RTDB it works nicely.
so can access  lock mode RTDB by esp8266.

mobizt, 2020-03-11:
You're using Firebase RTDB, right?

As I said above, database secret is the way you can access the RTDB database with not related to security rules. This is the most privilege, no matter what your security rules are.

closed 2020-03-11 by Mukesh7631