mobizt/Firebase-ESP8266 #96: HELP_ send string from Serial to Firebase

xuanbka1, 2020-03-12:
Hi Mr.Mobizt !
 I have string that receive from UART, then I want to send it to FireBase. But it is not success.
here my code:
`if (Serial.available() > 0) {

count++;
const String  incomingByte = Serial.readString(); // read the incoming byte:

 if( Firebase.setString(firebaseData, path +"/UpdateMessage", incomingByte))
{
  Serial.println(incomingByte);
  }
  else
  {
    Serial.println(firebaseData.errorReason());
    }

 }`

and this result printed

> bad request, Invalid data; couldn't parse JSON object, array, or value.

Can You help me solve this problem !
 Thanks for your support !

mobizt, 2020-03-12:
Your string object,  incomingByte contains some special chars that not allow in Firebase.
You should check it before sending it.

xuanbka1, 2020-03-12:
oh! I add /CR/LF to end of UART message so i get troubles  ! Thank you so much !

closed 2020-03-12 by xuanbka1