mobizt/Firebase-ESP8266 #97: Firebase.beginStream(firebaseData, "/path/in/firebase") returns bad Request in version 2.2.0 of the Firebase-ESP8266 library.

samlidippos, 2020-04-04:
Hi there! I have been having issues with the begineStream(firebaseData, "/path/in/firebase") method of the version 2.2.0 of the library. I have installed the library throught the Arduino IDE (v1.8.12). What I am trying to do is to subscribe to a path, in my Firebase Database, and set the Stream callbacks (and the call back timeout function), as in the examples provided. Using the version of the library 2.8.0 or the latest (with the FirebaseJson objects) it works like a charm, however, I want to use an older version as the board that I am programming on, is the Esp8266 - ESP-01, and it has very limited memory. When I use the v2.8.0 the space used from the sketch is 99% of the board, and thus the things that I can add are very limited. My sketch is that loaded, because I first set up a webserver with esp, where the user can connect through browser, give the SSID, password, and a Serial Number of the device in an html form, and then the ESP connects to the provided WiFi, creates this branch inside the Firebase database (using the Serial Number) , and starts streaming at this branch for changes. Apart from this, I am using a serial connection between the ESP and a regular Arduino Board, in order to control a LED bulb with the data fetched from the branch in Firebase. The ESP however is not supposed to read only, but also to update the data on the same branch (because the user can manually change the state of the LED). 

**Line of Code that produces issue:**
`Firebase.beginStream(firebaseData, firebaseDevicePath)`

**Location:**
Inside `handleForm()` method

**Error returned:**
bad request

**Arduino IDE:**
1.8.12

**My code:**
```
#include "index.h"
#include "FirebaseESP8266.h"
#include "ESP8266WebServer.h"
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>

#define FIREBASE_HOST "xxxx"
#define FIREBASE_AUTH "xxxx"

String firebaseDevicePath = "";
String luminosityPath;
String jsonChildrenPath = "{\"Luminosity\":50}";
FirebaseData firebaseData;
ESP8266WebServer server(80);
bool connected_to_cloud = false;
char incomingStringBuffer[200];
char incomingBufferFromPC[200];
int length_of_string = 0;
int length_of_pc_string = 0;
int old_luminosity = 0;
int new_luminosity = 0;
const int capacity = JSON_OBJECT_SIZE(3);
StaticJsonDocument<capacity> response_from_arduino;

void setup()
{
  Serial.begin(9600);
  Serial.println();
  Serial.println("Setting soft-AP ... ");
  WiFi.mode(WIFI_AP);
  WiFi.softAP("FOS_DEVICE", "");
  
  server.on("/", handleRootPath);    //Associate the handler function to the path
  server.on("/action_page", handleForm); //Handle the form here
  server.begin();                    //Start the server
  Serial.println("Server listening");
}

void loop() {

  server.handleClient();         //Handling of incoming requests

  if (connected_to_cloud == true)
  {
    
    String IncomingMessageFromArduino = "";
    boolean messageReady = false;

    //THOSE LINES OF CODE ARE FOR THE SERIAL COMMUNICATION BETWEEN THE ESP AND THE ARDUINO
    while (Serial.available()) {
      if (Serial.available() > 0)
      {
        char c = Serial.read();
        incomingBufferFromPC[length_of_pc_string] = c;
        length_of_pc_string++;
        Serial.print(c);
      }
    }
    Serial.flush();
    if (length_of_pc_string > 0)
    {
      if (incomingBufferFromPC[length_of_pc_string - 1] == '\n') {
        incomingBufferFromPC[length_of_pc_string] = '\0';
        String myIncomingString(incomingBufferFromPC);
        length_of_pc_string = 0;
        DeserializationError err = deserializeJson(response_from_arduino, myIncomingString);
        if (err) {
                    Serial.print(F("deserializeJson() failed with code "));
                    Serial.println(err.c_str());
        }
        else {
          int lum_new = response_from_arduino["Luminosity"];
          Serial.println(lum_new);
          String myUpdateJson = "{\"Luminosity\":";
          myUpdateJson += lum_new;
          myUpdateJson += "}";
          if (Firebase.updateNode(firebaseData, firebaseDevicePath, myUpdateJson)) {
          } else {
            Serial.println(firebaseData.errorReason());
          }
        }

        //      }

      }
    }
    //**THIS CODE IS USED FOR THE SERIAL COMMUNICATION WITH THE ARDUINO AND THE ESP 
    //**NOT YET IMPLEMENTED
    //    while (Serial.available())
    //    {
    //      char c = Serial.read();
    //      //Serial.print(c);
    //      IncomingMessageFromArduino += c;
    //      messageReady = true;
    //    }
    //    if (messageReady)
    //    {
    //      int myNum = IncomingMessageFromArduino.toInt();
    //      String myUpdateJson = "{\"Luminosity\":";
    //      myUpdateJson += myNum;
    //      myUpdateJson += "}";
    //      if (Firebase.updateNode(firebaseData, firebaseDevicePath, myUpdateJson)) {
    //
    //      } else {
    //        Serial.println(firebaseData.errorReason());
    //      }
    //    }
    Serial.flush();
    delay(50);
  }

}

void handleRootPath() {            //Handler for the rooth path
  String s = MAIN_page;
  server.send(200, "text/html", s);

}

void handleForm()
{

  String ssid = server.arg("SSID");
  String password = server.arg("password");
  firebaseDevicePath = server.arg("serialNum");

  WiFi.begin(ssid, password);

  delay(1500);

  while (WiFi.status() != WL_CONNECTED) { // Wait for the Wi-Fi to connect
    delay(1000);
  }
  delay(1000);

  Serial.println("Connection established!");
  delay(500);

  delay(1000);
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);

  delay(1000);
  String finalPath = "/";
  finalPath.concat(firebaseDevicePath);
  finalPath.concat("/Luminosity");
  luminosityPath = finalPath;
  finalPath.concat(jsonChildrenPath);
  delay(500);

  firebaseDevicePath = "/" + firebaseDevicePath;

  if (Firebase.updateNode(firebaseData, firebaseDevicePath, jsonChildrenPath)) {
    delay(500);

    Serial.println("Slot for this device was created at the cloud");
    connected_to_cloud = true;
  } else {
    Serial.println(firebaseData.errorReason());
  }
  delay(500);
  if (!Firebase.beginStream(firebaseData, firebaseDevicePath))
  {
    //Could not begin stream connection, then print out the error detail
    Serial.println(firebaseData.errorReason());
  }
  Firebase.setStreamCallback(firebaseData, streamCallback, streamTimeoutCallback);
  delay(500);
  Serial.println("Started the stream");
  String s = "<a href='/'> Go Back </a>";
  server.send(200, "text/html", s); //Send web page

}

//Global function that handles stream data
void streamCallback(StreamData data)
{

  Serial.println("Stream Data...");
  Serial.println(data.streamPath());
  Serial.println(data.dataPath());
  Serial.println(data.dataType());
  if (data.dataType() == "int")
    Serial.println(data.intData());

}

////Global function that notifies when stream connection lost
////The library will resume the stream connection automatically
void streamTimeoutCallback(bool timeout)
{
  if (timeout) {
    //Stream timeout occurred
    Serial.println("Stream timeout, resume streaming...");
  }
}
```
The code of the html is just a simple form, so I do not provide it here. If anyone think that it is necessary I can post it.
I am a newbie and off course the code is redundant, but I will fix this soon. I have read the similar issues, but they all use a higher version of the library, which is something that I think I cannot do, because of the sketch space. If you have any suggestion for my project for saving up space, and using the latest version of the library, I would be grateful if you could share it! I have already tried to use the v2.8.0 of the library, without including the Serial Communication with the Arduino. The Streaming works great! But I cannot add any code after that unfortunately.
Best regards!

mobizt, 2020-04-04:
I recommend using the ESP-01 black module instead of the blue one.

The black module is the upgrade version of the blue module with 1 MB flash size, 2 times that of the blue module.

mobizt, 2020-04-04:
The bad request error caused by the operation rejected by the server and returns HTTP code 400.

Check your node path that used in the functions. It must be non-empty strings and can't contain ".", "#", "$", "[", or "]"

Your issue about flash memory, you should use the black version of ESP-01 that I  reply above.

closed 2020-04-04 by mobizt

mobizt, 2020-04-04:
Finally, I'm not recommend to use a years ago outdated version of library.

Your json object which you created by joining the string may not valid and when it use in updateNode (2.2.0), it may reject by server.

samlidippos, 2020-04-06:
I see! Many things that I could have done better! Thank you very much for the response. I have already got the black esp and I think that I will be able to use the latest version of the library! Thanks!