mobizt/Firebase-ESP8266 #98: This is the code that i want to dipalyon the firebase dataset.  The arduino ide shows the data perfectly .But the data is not stored in the firebase.

Prajwalk56, 2020-04-08:
#include <FirebaseArduino.h>
#include <ESP8266WiFi.h>
#define FIREBASE_HOST "arduinotest-f6d5e.firebaseio.com" 
#define FIREBASE_AUTH "u65MmIKJqyvHoUmG6hMMK0KoCuMYl7ZRF06Y4JFG" 
#define WIFI_SSID "kick" 
#define WIFI_PASSWORD "12345678"
float fire;
void setup() {
Serial.begin(9600);
pinMode(D5, OUTPUT);
pinMode(D6, OUTPUT);
pinMode(A0, INPUT);
// connect to wifi.
WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
Serial.print("connecting");
while (WiFi.status() != WL_CONNECTED) {
Serial.print(".");
delay(500);
}
Serial.println();
Serial.print("connected: ");
Serial.println(WiFi.localIP());
Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
Firebase.set("LED_STATUS", 0);
}
int n = 0;
void loop() {
int analogValue = analogRead(A0);
float millivolts = (analogValue/1024.0) * 3300; //3300 is the voltage provided by NodeMCU
float celsius = millivolts/10;

Serial.print("in DegreeC=   ");
Serial.println(celsius);
delay(1);
if(celsius>35){
digitalWrite(D6,HIGH);
Firebase.setInt("fire_sensor_status",1);
}
n = Firebase.getInt("LED_STATUS");
// handle error
if (n==1) {
Serial.println("LED ON");
digitalWrite(D5,HIGH);  
return;
delay(10);
}
else {
Serial.println("LED OFF");
digitalWrite(D5,LOW);  
digitalWrite(D6,LOW);  
Firebase.setInt("fire_sensor_status",0);
return;
}
}

mobizt, 2020-04-08:
The library you're using is not belong to this GitHub repo.

Please search the library FirebaseArduino.

mobizt, 2020-04-08:
You may need to download the zip file from FirebaseArduino and replace the old files that installed to update the fingerprint data in the source code.

closed 2020-04-08 by mobizt