mobizt/Firebase-ESP8266 #99: HELP: CloudMessaging, set, stream with the same loop

elsayed-mohammed, 2020-04-09:
I want to use three Firebase services
FCM
Stream
Set
but I am facing a problem when I used them in the same loop.
the problem: Exception(29) lack of memory because I make three FirebaseData because if i use same FirebaseData to stream and set, when set the stream capture data but the data in path not changed.
so, if there any way to stop FirebaseData of FCM after used to free heap and when I need it again reinitialize because I will use FCM after long time if certain condition occur.

mobizt, 2020-04-10:
Please update the ESP8266 Arduino Core SDK (esp8266 board package) to the latest version to gain more free heap.

Every FirebaseData object consumes RAM, mostly used by internal WiFiClient. Then I don't recommend using more than two objects.

You can use one FirebaseData object for streaming with callback and another one FirebaseData object for set/push/get calls included FCM.

In general one FirebaseData object for streaming and set/push/get calls (even FCM) are OK and you need to prepare the idle time (10 seconds) to allow the stream to establish i.e don't set/push/get too frequent.

elsayed-mohammed, 2020-04-10:
Thank you for your reply.
I follow your instruction and made two FirebaseData object, as you mentioned.
but cause crash exception 29.
My project require to check data changed on specific path on RTD and if changed receive data from other path by get, in addition to use set when push button pressed.
So I want to use Stream to avoid periodic check.on change data that's make an excessive bandwidth from 10GB of free plan, and I want the response is fairly good, when push button pressed set data very quick.
But I think when this operations done in short period this will cause lack memory exception.
So, Are you can Advice me what I can do in this case?
Thanks in advance

mobizt, 2020-04-10:
Please post your code.

mobizt, 2020-04-10:
Normally if you using 3rd party library, you need to check the free heap at any place of your code to debug the free heap.

mobizt, 2020-04-10:
From the examples https://github.com/mobizt/Firebase-ESP8266/tree/master/examples
You will see the intensive example that does a lot of work that supported by Firebase.

**I can't advise anything without a look at your implemented code because of some improper or misuse or misunderstanding, lack of memory management may cause the error.**

elsayed-mohammed, 2020-04-10:
> From the examples https://github.com/mobizt/Firebase-ESP8266/tree/master/examples
> You will see the intensive examples that do a lot of works that supported by Firebase.
> 
> **I can't advise anything without a look at your implemented code because of some improper or misuse or misunderstanding, lack of memory management may cause the error.**

I appreciate that.
I know very well the extent of your great effort in this library and examples, quick response to the library help request.
Thank you sir, I will follow your advice and if I solve my problem, I will tell you.
Thank you.

mobizt, 2020-04-10:
You should post the test code that I can verify and test for you.

Exception 29 does not tell much information.
The possibilities can be stack overflow as it prints out all stack trace to serial and wdt reset due to long-running or blocking code.

You also need to share the debug serial output too if you want someone to help.

mobizt, 2020-04-11:
If you send FCM by calling **Firebase.broadcastMessage**, and you add multiple device tokens.

You may get the exception error due to the reserved buffer size for the message was fixed to 400 bytes and not fit with the message in which the device tokens included.

Now it was fixed already, you can update the library.

The following is the code that sets the string, sends FCM message, and listens to the stream with two FirebaseData, and you can try it.

```C++

#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

#define FIREBASE_HOST "YOUR_FIREBASE_PROJECT.firebaseio.com" //Without http:// or https:// schemes
#define FIREBASE_AUTH "YOUR_FIREBASE_DATABASE_SECRET"
#define WIFI_SSID "YOUR_WIFI_AP"
#define WIFI_PASSWORD "YOUR_WIFI_PASSWORD"

#define FIREBASE_FCM_SERVER_KEY "YOUR_FIREBASE_PROJECT_CLOUD_MESSAGING_SERVER_KEY"
#define FIREBASE_FCM_DEVICE_TOKEN_1 "RECIPIENT_DEVICE_TOKEN"
#define FIREBASE_FCM_DEVICE_TOKEN_2 "ANOTHER_RECIPIENT_DEVICE_TOKEN"

//Define FirebaseESP8266 data object
FirebaseData firebaseData;
FirebaseData stream;

unsigned long sendDataPrevMillis = 0;
unsigned long sendFCMPrevMillis = 0;

String path = "/Test/Stream";

uint16_t count = 0;

void printResult(StreamData &data);
void sendMessage();
void streamTimeoutCallback(bool timeout);
void streamCallback(StreamData data);

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
  firebaseData.setBSSLBufferSize(1024, 1024);

  //Set the size of HTTP response buffers in the case where we want to work with large data.
  firebaseData.setResponseSize(1024);

  //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
  stream.setBSSLBufferSize(1024, 1024);

  //Set the size of HTTP response buffers in the case where we want to work with large data.
  stream.setResponseSize(1024);

[... 159 lines cut ...]
  }
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
  {
    Serial.println();
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
}

void streamCallback(StreamData data)
{
  Serial.println("Stream Data available...");
  Serial.println("STREAM PATH: " + data.streamPath());
  Serial.println("EVENT PATH: " + data.dataPath());
  Serial.println("DATA TYPE: " + data.dataType());
  Serial.println("EVENT TYPE: " + data.eventType());
  Serial.print("VALUE: ");
  printResult(data);
  Serial.println();
}
```

closed 2020-04-11 by mobizt