mobizt/Firebase-ESP8266 #101: Soft WDT reset, caused every time trying to execute Firebase.updateNode

samlidippos, 2020-04-21:
Hello! I have a question, on how to use the Firebase.updateNode with the ESP8266  ESP-01 (black module with 1mb flash memory). In my code I first set up a webserver with esp, where the user can connect through browser, give the SSID, password, and a Serial Number of the device in an html form, and then the ESP connects to the provided WiFi, creates this branch inside the Firebase database (using the Serial Number) , and starts streaming at this branch for changes. Apart from this, I am using a serial connection between the ESP and a regular Arduino Board, in order to control a LED bulb with the data fetched from the branch in Firebase. The ESP however is not supposed to read only, but also to update the data on the same branch (because the user can manually change the state of the LED). 

**Line of Code that produces issue:**
Firebase.updateNode(firebaseData, "/", updateData)

**Location:**
Inside handleForm() method

**Throws Exception:**
Soft WDT reset

**Details:**

- SDK Core: 2.6.3 
- FirebaseESP8266 library: 2.8.8
- CPU Freq: 80MHz
- Crystal Freq: 26MHz
- Flash Size: 1MB (FS:none OTA:~502Kb)
- Flash Mode: DOUT (compatible)
- Flash Frequency: 40MHz
- Reset Method: no dtr (aka ck)
- Debug Level: None
- IwIP Variant: v2 Lower Memory 
- V Tables: Flash
- Exceptions: Disabled (new can abort)
- Erase Flash: Only Sketch
- Espressif FW: nonos-sdk 2.2.1 + 119(191122)
- SSL Support: All SSL ciphers (most compatible)

**Powering:**
I used the instructions below to power the ESP8266
https://tttapa.github.io/ESP8266/Chap02%20-%20Hardware.html
The 5V supply come from a: FTDI FT232RL USB TO TTL SERIAL CONVERTER
which is powered by a phone charger that can send up to 2A. 
I don't know if it is appropriate. 

**Arduino IDE:**
1.8.12

**The code:**
```
#include "index.h"
#include "FirebaseESP8266.h"
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include "ESP8266WebServer.h"

#define FIREBASE_HOST "xxx"
#define FIREBASE_AUTH "xxx"

String luminosityPath;
FirebaseData firebaseData;
ESP8266WebServer server(80);
bool connected_to_cloud = false;
char incomingStringBuffer[200];
char incomingBufferFromPC[200];
int length_of_string = 0;
int length_of_pc_string = 0;
String DeviceSerialNumber;

void setup()
{
  Serial.begin(9600);
  Serial.println("Setting soft-AP ... ");
  WiFi.mode(WIFI_AP);
  WiFi.softAP("FOS_DEVICE", "");

  server.on("/", HTTP_GET, handleRoot);   //Associate the handler function to the path
  server.on("/action_page", handleForm); //Handle the form here
  server.begin();                    //Start the server
[... 123 lines cut ...]
0x40228f24: operator new(unsigned int) at /workdir/repo/gcc/libstdc++-v3/libsupc++/new_op.cc line 52
0x40209b5e: FirebaseESP8266::sendRequest(FirebaseData&, unsigned char, std::string const&, unsigned char, unsigned char, std::string const&, std::string const&, std::string const&) at C:\Users\Σαμ\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 2528
0x4021346c: ESP8266WiFiSTAClass::status() at C:\Users\Σαμ\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 634
0x4020a8a9: FirebaseESP8266::buildRequest(FirebaseData&, unsigned char, unsigned char, std::string const&, char const*, bool, std::string const&, std::string const&) at C:\Users\Σαμ\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 314
0x40228f24: operator new(unsigned int) at /workdir/repo/gcc/libstdc++-v3/libsupc++/new_op.cc line 52
0x40228fff: std::string::_S_copy_chars(char*, char const*, char const*) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 406
0x402299a3: std::string::_S_construct (char const*, char const*, std::allocator  const&, std::forward_iterator_tag) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc line 148
0x40228ea0: operator delete[](void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_opv.cc line 33
0x4020aa48: FirebaseESP8266::updateNode(FirebaseData&, String, FirebaseJson&) at C:\Users\Σαμ\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseESP8266.cpp line 1488
0x4022927a: std::string::_Rep::_M_dispose(std::allocator  const&) at /workdir/arena.x86_64/gcc/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 254
0x4020bdd8: FirebaseJson::_init() at C:\Users\Σαμ\Documents\Arduino\libraries\Firebase_ESP8266_Client\src\FirebaseJson.cpp line 91
0x40216b3c: String::copy(char const*, unsigned int) at C:\Users\Σαμ\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\WString.cpp line 214
0x40203c6e: handleForm() at C:\Users\Σαμ\Documents\Arduino\ESP_DECEMBER_2019/ESP_DECEMBER_2019.ino line 171
0x40100ccc: realloc(void*, size_t) at C:\Users\Σαμ\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 760
0x40216af3: String::reserve(unsigned int) at C:\Users\Σαμ\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\WString.cpp line 146
0x402034be: std::_Function_handler   >::_M_invoke(std::_Any_data const&, String&, String&, String const&, int, int, int, int) at c:\users\Σαμ\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40216a06: String::changeBuffer(unsigned int) at C:\Users\Σαμ\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\WString.cpp line 187
0x40201658: esp8266webserver::FunctionRequestHandler ::handle(esp8266webserver::ESP8266WebServerTemplate &, HTTPMethod, String) at C:\Users\Σαμ\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h line 40
0x40100a9f: free(void*) at C:\Users\Σαμ\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\umm_malloc\umm_malloc.cpp line 362
0x40201658: esp8266webserver::FunctionRequestHandler ::handle(esp8266webserver::ESP8266WebServerTemplate &, HTTPMethod, String) at C:\Users\Σαμ\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h line 40
0x4022040e: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\Σαμ\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x402014b8: esp8266webserver::FunctionRequestHandler ::canHandle(HTTPMethod, String) at C:\Users\Σαμ\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h line 23
0x401000e1: std::function ::operator()() const at c:\users\Σαμ\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2465
```
Inside the stack decoding there is the line 171 from HandleForm which is the line that the ESP8266 is trying to update a node (more specifically to add a new one). I am a newbie and I don't know exactly if I am setting the settings correctly, or whether I have to move to FreeRTOS (instead of the nonos-sdk) or if there is any other problem in the code (delays, string usage or other). Thanks in advance for your help!

mobizt, 2020-04-21:
You should show the wdt reset message from serial because it can tell the reset causes.

samlidippos, 2020-04-21:
**Those are the last lines before the crash, along with the crash lines:**
```
Connection established!
IP address:	
192.168.1.6
/Bro/Luminosity
('Bro' is the node name)

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3ffff4c0 end: 3fffffb0 offset: 01b0
3ffff670:  00000005 00000000 3fff0064 00000008  
3ffff680:  00003a98 00050416 3fff0064 40215202  
3ffff690:  000000e6 00000005 bfad28f3 00000000  
3ffff6a0:  00000001 00007659 3fff2cdc 402391a6  
3ffff6b0:  3fff31dc 3ffef7f8 3fff2cdc 00000001  
3ffff6c0:  00000001 00000001 3fff0064 4021541c  
3ffff6d0:  3ffef7f8 00000000 3fff0064 402155ff  
3ffff6e0:  00000026 3ffeefb4 00000001 402178d3  
3ffff6f0:  00000000 00000000 00000000 40217e8a  
3ffff700:  00000000 3fff2404 3fff0064 402141ac  
3ffff710:  000001bb 00000d50 3ffeedb8 00000000  
3ffff720:  000001bb 3fff0064 3ffef7f8 00000000  
3ffff730:  000001bb 3fff0064 3ffef7f8 4021571d  
3ffff740:  40221b68 5561c923 40221b68 5561c923  
3ffff750:  3ffe89c1 00000000 3ffee9d4 4020afbd  
3ffff760:  3fff22bc 3ffee968 3ffee9d4 4020affc  
3ffff770:  3fff2af8 4021345c 00000000 40228e90  
3ffff780:  3fff22bc 0000007d 3ffeedb8 00000000  
3ffff790:  3fff22bc 3ffee968 3ffeedb8 40207f77  
3ffff7a0:  3fff2240 3ffef918 00000019 00000000  
3ffff7b0:  00000000 00000592 00000592 40100814  
3ffff7c0:  3ffef7f8 3ffefab8 3ffe89c1 3fff2274  
3ffff7d0:  3ffffac0 00000001 00000020 40100a9f  
3ffff7e0:  40261d52 3ffee968 3fff2af8 00000001  
3ffff7f0:  00000006 00000007 3ffff8e4 3ffee9d4  
3ffff800:  3ffff8e0 0000007d 3ffffac8 3ffefad4  
3ffff810:  00000007 3ffeedb8 3ffee968 00000006  
3ffff820:  00000007 3ffeedb8 3ffee968 402081ca  
3ffff830:  3ffff944 4021345c 00000026 40228f14  
3ffff840:  3ffff8e4 3ffff8e0 3fff0064 3ffee968  
3ffff850:  00000000 00000006 3ffeedb8 3ffee968  
3ffff860:  00000000 00000006 3ffeedb8 40209b4e  
3ffff870:  3ffff944 4021345c 3ffefa84 00000000  
3ffff880:  00000007 3ffee9d4 3ffff8e4 00000000  
3ffff890:  3ffff8e8 3ffee968 3ffee968 00000000  
3ffff8a0:  00000000 3ffee968 00000001 4020a899  
3ffff8b0:  3ffff8e0 3ffff944 3ffff940 40228f14  
3ffff8c0:  3ffefa9c 3ffefa9c 00000001 40228fef  
3ffff8d0:  3ffff94c 3fff221c 3ffffbf1 40229993  
3ffff8e0:  3ffefa90 3fff2240 3ffffac0 40228e90  
3ffff8f0:  00000001 00000006 3ffeedb8 00000007  
3ffff900:  3ffff948 3fff2280 00000000 3ffff8e0  
3ffff910:  3ffff950 00000000 3ffff990 3ffeedb8  
3ffff920:  3ffee968 3ffff944 3ffff940 4020aa38  
3ffff930:  00000000 3ffff944 3ffff940 4022926a <
3ffff940:  3ffef4c0 3ffef4c0 3fff2228 3fff2280  
3ffff950:  00000001 3ffef4c0 3ffff990 4020bdc8  
3ffff960:  3fff2280 00000001 3ffffbf0 40216b2c  
3ffff970:  3ffeeda0 3ffffbf0 3ffe872a 3ffeedb8  
3ffff980:  3ffeeda0 3ffee968 3ffeeeb0 40203c62  
3ffff990:  00000001 ffffffff 00000000 ffffffff  
3ffff9a0:  00000000 ffffffff 00000001 ffffffff  
3ffff9b0:  ffffffff 00000001 00000100 00000100  
3ffff9c0:  3ffef97c 3ffef984 3ffef994 3ffef99c  
3ffff9d0:  3ffef9a4 3ffef9ac 3ffef9b4 3ffef9c4  
3ffff9e0:  3ffef9cc 3ffef9dc 3ffef9bc 3ffef9e4  
3ffff9f0:  3ffef9ec 3ffef9f4 3ffefa04 3ffefa14  
3ffffa00:  3ffefa24 3ffefa2c 3ffefa3c 3ffefa4c  
3ffffa10:  3ffefa5c 3ffefa6c 3ffefa7c 3fff2208  
3ffffa20:  3ffef4c0 3fff1f3c 3fff204c 40100ccc  
3ffffa30:  00000020 00000014 00000000 00000000  
3ffffa40:  00000000 00000000 00000000 00000000  
3ffffa50:  00000000 00000000 00000000 0000000f  
3ffffa60:  00000000 00000020 80fffb60 00000000  
3ffffa70:  00000000 00000000 3ffffb00 65646e75  
3ffffa80:  656e6966 89ff0064 3ffe0100 00000000  
3ffffa90:  00000000 00000000 00000000 00000000  
3ffffaa0:  00000000 00000000 3ffef4c0 40216ae3  
3ffffab0:  3ffef96c 00000000 00000000 00009823  
3ffffac0:  00000001 ffffffff 00000000 ffffffff  
3ffffad0:  00000000 ffffffff 00000001 ffffffff  
3ffffae0:  ffffffff 00000001 00000100 00000100  
3ffffaf0:  3ffef754 3ffefc84 3fff1ef4 3fff1efc  
3ffffb00:  3fff1fdc 3fff1fe4 3fff274c 3ffefcec  
3ffffb10:  3fff2044 3ffefcf4 3fff2754 3fff2054  
3ffffb20:  3fff205c 3ffef6f4 3ffef704 3ffef714  
3ffffb30:  3ffef724 3ffef78c 3ffef79c 3ffef7ac  
3ffffb40:  3ffef94c 3ffef95c 3ffef7bc 3fff2258  
3ffffb50:  3ffef4c0 00000030 ffffffff 402034be  
3ffffb60:  00000000 00000000 00000000 00000000  
3ffffb70:  00000000 00000000 00000000 00000000  
3ffffb80:  00000000 00000000 00000000 00000000  
3ffffb90:  3ffeef00 00000010 80000000 00000000  
3ffffba0:  00000000 00000000 00000000 65646e75  
3ffffbb0:  656e6966 89ff0064 00000100 00000000  
3ffffbc0:  00000000 00000000 00000000 00000000  
3ffffbd0:  00000000 00000000 3ffef4c0 402169f6  
3ffffbe0:  3fff1dac 00000000 00000000 00000000  
3ffffbf0:  696d002f 69736f6e 81007974 3ffef81c  
3ffffc00:  000f000f 04fee8a4 006f7242 40201658  
3ffffc10:  83002980 3fff2004 0013001f 00100814  
3ffffc20:  3fff1f64 0014001f 00fee880 40201658  
3ffffc30:  00000001 00000001 00000020 40100a9f  
3ffffc40:  3ffffc80 3fff1cac 00000000 40201658  
3ffffc50:  00000001 00000001 3fff1c84 402203fe  
3ffffc60:  00000001 402014b8 3fff1c84 401000e1  
3ffffc70:  3fff1c84 3ffee8c0 3fff1c84 4020168e  
3ffffc80:  3fff1f00 000c000f 8000057c 80100814  
3ffffc90:  3fff1c84 3ffee8c0 3ffee880 402037ca  
3ffffca0:  3fff1fc4 000c000f 00ff2500 0000008f  
```
How can I check which are the wdt reset causes? (I thought the only way was through the exception decoder)

mobizt, 2020-04-21:
The error is due to conflict when you disable the AP with **WiFi.softAPdisconnect** in the **handleForm** function while **server.handleClient()** still in processing.

You do not need to disable AP to connect in STA mode.

The following example shows how to run both AP and STA to send data to Firebase while you can serve as the local AP web server.

Please edit **FIREBASE_HOST**, **FIREBASE_AUTH**, **WIFI_SSID**, and **WIFI_PASSWORD** in the below code.

```C++

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

#include <FirebaseESP8266.h>

#define FIREBASE_HOST "xxxxxxxxxxxx"
#define FIREBASE_AUTH "xxxxxxxxxxxx"
#define WIFI_SSID "xxxxxx"
#define WIFI_PASSWORD "xxxxxxx"

unsigned long sendDataPrevMillis;
int count = 0;
FirebaseData firebaseData;

ESP8266WebServer server(80);

void handleRoot()
{
  server.send(200, "text/plain", "hello from esp8266!");
}

void handleNotFound()
{
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++)
  {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void setup(void)
{
  Serial.begin(115200);
  WiFi.mode(WIFI_AP_STA);

  if (WiFi.softAP("TEST_SSID", ""))
  {
   //The AP IP of web server that allow client to connect.
    IPAddress address = WiFi.softAPIP();
    Serial.print("Soft-AP IP address = ");
    Serial.println(address);
  }

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(WIFI_SSID);

  //STA IP for internet
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (MDNS.begin("esp8266"))
  {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.on("/gif", []() {
    static const uint8_t gif[] PROGMEM = {
        0x47, 0x49, 0x46, 0x38, 0x37, 0x61, 0x10, 0x00, 0x10, 0x00, 0x80, 0x01,
        0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0x2c, 0x00, 0x00, 0x00, 0x00,
        0x10, 0x00, 0x10, 0x00, 0x00, 0x02, 0x19, 0x8c, 0x8f, 0xa9, 0xcb, 0x9d,
        0x00, 0x5f, 0x74, 0xb4, 0x56, 0xb0, 0xb0, 0xd2, 0xf2, 0x35, 0x1e, 0x4c,
        0x0c, 0x24, 0x5a, 0xe6, 0x89, 0xa6, 0x4d, 0x01, 0x00, 0x3b};
    char gif_colored[sizeof(gif)];
    memcpy_P(gif_colored, gif, sizeof(gif));
    // Set the background to a random set of colors
    gif_colored[16] = millis() % 256;
    gif_colored[17] = millis() % 256;
    gif_colored[18] = millis() % 256;
    server.send(200, "image/gif", gif_colored, sizeof(gif_colored));
  });

  server.onNotFound(handleNotFound);
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void)
{
  server.handleClient();
  MDNS.update();

  if (millis() - sendDataPrevMillis > 15000 ||  sendDataPrevMillis ==0)
  {
    sendDataPrevMillis = millis();
    count++;

    Serial.println("------------------------------------");
    Serial.println("Set JSON...");

    FirebaseJson json;
    json.add("data", "hello").add("num", count);
    if (Firebase.setJSON(firebaseData,  "/Test/Json", json))
    {
      Serial.println("PASSED");
      Serial.println("PATH: " + firebaseData.dataPath());
      Serial.println("TYPE: " + firebaseData.dataType());
      Serial.print("VALUE: ");

      Serial.println(firebaseData.jsonString());

      Serial.println("------------------------------------");
      Serial.println();
    }
    else
    {
      Serial.println("FAILED");
      Serial.println("REASON: " + firebaseData.errorReason());
      Serial.println("------------------------------------");
      Serial.println();
    }
  }
}
```

samlidippos, 2020-04-21:
alright! I run your code and it runs smoothly, with no errors at all. 
However, in my code I removed the line WiFi.softAPdisconnect, but still no luck. Same Exception (WDT reset). The thing is that I want the user to provide the SSID and PASSWORD for the home wifi through the html form, which I serve through the webserver, and not to provide through the sketch in setup function. Another thing that I tried to change was the Serial rate to 115200 but, as in your code, but no luck still.

mobizt, 2020-04-21:
Your code does not work in any way, bad design flow, and misconception.

You're trying to switch the WiFi code back and forth between AP and STA and trying to do something that impossible.

Forget about your bad code and try to adapt from my above example code.

closed 2020-04-21 by mobizt

mobizt, 2020-04-22:
These are the guides for you to follow.

1. Set up the WiFi mode to WIFI_AP_STA **in the setup** which is essential.
2. Start the WiFi STA connection with your WiFi hotspot **in the setup**.
3. Start the WiFi AP with your preferred SSID and also PSK **in the setup**.
4. Set up Firebase and its stream callbacks as required **in the setup**.
5. Send the data to Firebase **in the loop** depends on the condition.
6. Send the data to Firebase depends on the webserver event **in the webserver event handler**.

mobizt, 2020-04-22:
I close this issue due to not related to the Firebase library.

The issue is due to conflicts in WiFi and Web server operations or long run code in the server.handleClient(); due to the WiFi mode change that makes software watchdog reset.

innovativejay, 2021-08-31:
That was insightful, thanks !!