mobizt/Firebase-ESP8266 #102: Is onDisconnect method available

thetechguy2016, 2020-05-13:
Hello @mobizt 
It’s like a blessing to have this library, I want to say thanks to you from all developers using this library. 
 we Don’t find onDisconnect method to create presence system like [This](https://firebase.googleblog.com/2013/06/how-to-build-presence-system.html?m=1) .I want to ask you if this functionality available ,please let us know , if it is not available, can you do this for all of us. 
A Great Thanks To You

mobizt, 2020-05-13:
The features you mention are not available in the REST APIs used by this library.

It can do by web socket in other platforms that require the SDK installation.

closed 2020-05-13 by mobizt

thetechguy2016, 2020-05-13:
Can we create our own logic to achieve this in any way.

mobizt, 2020-05-13:
Client (REST or HTTP) can't sync with RTDB as RTDB is not act as a cloud server to compute or run user script.

The stream (HTTP) is the mode that the server keeps the connection open and push data (one way) to the client.  The client can't send any data to the client in stream mode.

No way to do interchanges the data between client and server except for web socket which Google does not except the general web socket but has own protocol and needs to work with SDK.

You can use the timestamp which stores for specific interval e.g., every 1-2 minutes. And use other clients to compare these device timestamps with the current timestamp and determine the offline/online of the device due to the second count since the last timestamp update.

You can use the Google Cloud Functions to do check these timestamps periodically and notify the status of the devices to another online client.

This is an indirect approach and async that can work in this situation.

thetechguy2016, 2020-05-13:
Thanks for your support @mobizt , and for this library also, i want to buy a drink for you if you dont mind.