mobizt/Firebase-ESP8266 #103: QUESTION

SourabhDaphale, 2020-05-15:
Hi,
New to this library and wanted to know if we could stream the firebase data and also get/set data using getInt/setInt function in the same code??

mobizt, 2020-05-15:
Please see the stream example in examples folder.

SourabhDaphale, 2020-05-15:
Hi, 
Saw the stream example, in which "Hello World" is set as string after a set time. 
I removed the hello world if-loop from the code and observed that only when some data is changed in the database, 'firebaseData.streamAvailable()' loop function is executed.
So, I put my setInt() function inside the streamAvailable() loop but then the loop gets executed continuously. 

if (firebaseData.streamAvailable())
{
    Serial.println("------------------------------------");
    Serial.println("Stream Data available...");
    Serial.println("STREAM PATH: " + firebaseData.streamPath());
    Serial.println("EVENT PATH: " + firebaseData.dataPath());
    Serial.println("DATA TYPE: " + firebaseData.dataType());
    Serial.println("EVENT TYPE: " + firebaseData.eventType());
    Serial.print("VALUE: ");
    printResult(firebaseData);
    Firebase.setInt(firebaseData, "/data/count", 2);
    Serial.println();
}

This change is making the code run continuously even without a change in data. Did I do something wrong? Please help me out.
Note: the path for stream and to set the value for count is different.

mobizt, 2020-05-15:
If you are new to this library, I recommend the following example instead.

https://github.com/mobizt/Firebase-ESP8266/tree/master/examples/Different_objects_stream
https://github.com/mobizt/Firebase-ESP8266/tree/master/examples/Stream_callback

## Firebase Data Object usage

The object called **FirebaseData** holds the WiFi client instance and data for sending and receiving.

The connection between client and server will keep alive in stream mode. 

To make the stream run smoothly, don't use delay and some blocking code in the loop.

The stream will interrupt (connection close) when the user calls other functions that use the same Firebase Data object.

When the operation from other functions finished, the client will work in the stream mode again.

Because the mode changes, when the user used the same Firebase Data object to do all, the user needs to prepare the idle time for the stream to stay connected at least 8-10 seconds before changing it to other modes.

That's why we should not call set function too often in "Stream example".

When the stream connected, the server will send the payload data to the client immediately for the first time.

When the connection mode changed (between the stream and other Firebase calls), you will always get the stream data even no data changes in your streaming path.

You shall not change the node data that belongs to the path that you're listening to immediately, upon the stream event triggered (by stream data available) because it will act as a positive feedback system.

The "Stream.ino example" should use in case you have the limit free memory and the data latency are acceptable.

closed 2020-05-15 by mobizt