mobizt/Firebase-ESP8266 #105: Store data asynchronously

am-canes, 2020-05-26:
I'm using ESP8266 with platformIO and I'm connecting it with firebase.
When I store some data using set function (setFloat, setInt,...) it takes a time between 200ms to 1,5s to execute the command. Is it possible to execute this command asynchronously? I have some critical sensors and my code couldn't be blocked. 

Thanks.

mobizt, 2020-05-26:
It depends on your network condition.

The connection in the get, set. push and update functions are keeping alive unless the time out.
The client (your device) can send/receive data as fast as possible.

Only, the first call after initialize and the Firebase Data object uses with these functions does not use for streaming, the connection time may take longer due to the SSL handshaking with the server.

mobizt, 2020-05-26:
Async operation is not supported.  The library works through the REST API not web socket client and needs to wait for the server response to be completed.

closed 2020-05-26 by mobizt