mobizt/Firebase-ESP8266 #113: Can't filter query that needs to be ordered by key

TMaxedCode, 2020-06-24:
I am trying to filter the following json data by its key value:
```
{
  "temp" : {
    "1" : 25,
    "2" : 24
  },
}
```
I want to be able to get only the highest value key in temp which is 2 in the json data. How would I be able to filter the results and get that filtered data? I also have to following rules defined to try and filter the data using your example.
```
{
  "rules": {
    ".read": true,
    ".write": true,
    ".indexOn":"temp"
  }
}
```
Here is the code I used (firebaseData2 is defined already):
```
 QueryFilter query;
  query.orderBy("temp");
  query.startAt(0);
  query.endAt(9999);
  query.limitToLast(1);
  if (Firebase.getJSON(firebaseData2, "/", query))
  {
    Serial.println(firebaseData2.jsonString());
  }
  else
  {
    Serial.println(firebaseData2.errorReason());
  }
```
Thanks for your help.

mobizt, 2020-06-24:
Please try this

```C++

  QueryFilter query;
  query.orderBy("$key");
  query.startAt("0");
  query.limitToLast(1);
  if (Firebase.getJSON(firebaseData2, "/Temp", query))
  {
    Serial.println(firebaseData2.jsonString());
  }
  else
  {
    Serial.println(firebaseData2.errorReason());
  }
```

TMaxedCode, 2020-06-24:
@mobizt It works perfectly. Thanks. The docs about filtering data is confusing. Sorry for asking this question. I can help you make the docs more understandable if you want.

mobizt, 2020-06-24:
From further test, it does not work when you add **query.endAt("999")**.

You can pull request if you want.

TMaxedCode, 2020-06-24:
Its ok I don't really need query.endAt("999"). Thank you for the information though. Have a nice day!

closed 2020-06-24 by TMaxedCode