mobizt/Firebase-ESP8266 #114: What best secure rules can work with this library? HELP

shoaib2527, 2020-06-28:
Everything is working fine, but Firebase Realtime Database rules are
{
  "rules": {
    ".read": true,
    ".write": true
  }
}
I need to make it secure. 
So i need your help to  figure out what best secure rules will work with this library.

mobizt, 2020-06-28:
When you use database secret with this client, your device can read and write access to your database without restriction by read and write rules because database secret  is always by-pass these rules.

I can't recommend about read and write options, because it can be any up on your implementation, especially when you also use it for other clients with the Firebase SDK e.g. web, android, iOS, java, php and unity.

closed 2020-06-28 by mobizt

shoaib2527, 2020-06-28:
Okay I'll try and update here.
Thanks

mobizt, 2020-06-28:
You 're not need to update. This is info what you should know.