mobizt/Firebase-ESP8266 #116: HELP Firebase Storage Fingerprint

danilomna, 2020-07-03:
Hi Mobizt,

As you knkw I'm a big fan of yours and your job with firebase and Esp8266, your Esp8266 firebase client is awesome and deal with fingerprint in a very professional way.

I would like to talk to you about another issue that a very skilled programmer like you could help solve.

I need the Esp8266 get file from Firebase Storage, the file is firmware binary, I'm doing a OTA firmware update and storing firmware file in firebase storage, but issue is to do this I also need to deal with fingerprint, that changes and many times cause esp8266 can not get file.

How could I do this properlly and solve the fingerprint change issue? 
Could you do something to solve that in firebase storage? like you did with your approach to firebase database.

If you accept the challenge I can pay you for your time and work.

If there is any other way we talk abou this, let me know, like skype(for text chat), email or other way.

Can you help with firebase storage fingerprint issue?

Best regards
Danilo

mobizt, 2020-07-03:
For BearSSL WiFi Client in  esp8266 Arduino sdk v2.5.x or later

```C++

#include <ESP8266HTTPClient.h>

  ...
  ...

  WiFiClientSecure client;
  
  //nonsecure mode
  client.setInsecure();

  //set rx and tx buffer size if needed
  //client.setBufferSizes(1024, 1024);

  HTTPClient https;

  //public access url
  //https://firebasestorage.googleapis.com/v0/b/[BUCKET_NAME]/o/[OBJECT_NAME]?alt=media

  //authorize access url using access token
  //https://storage.googleapis.com/download/storage/v1/b/[BUCKET_NAME]/o/[OBJECT_NAME]?alt=media

  if (https.begin(client, "URL"))
  {

  }
```

I don't have the esp8266 device with me during the weekend to test the above code but I'm sure it works.

mobizt, 2020-07-03:
Tank you for your offer but I still don't want to take the benefit from people using my library.

I don't use chat and you can contact me via Email which written in the examples comment.

arlucio, 2020-07-03:
If you don't care about TLS security to download the firmware you can just take the s out of the generated file https link, and download the file using a simple http get request, firebase storage accept this.

closed 2020-07-04 by mobizt