mobizt/Firebase-ESP8266 #117: HELP: readStream() stopped working for me

Sebbecking, 2020-07-12:
Hi there,

I have an application that uses your (great!) library. 
I own 12 embedded devices with that application, and they worked fine for months. 
Sadly, today they all stopped working. I investigated the problem and found out, that `readStream` fails with a "connection refused" error.

I use the library in two places: 
- Once on startup of the device, a timestamp and a few strings are written to the database with `pushTimestamp` and `pushString`. This works fine and still continues to work.
- I also register a stream with `beginStream` (which still works) and subsequently call `readStream` in the main loop. This call returns false, the errorReason is "connection refused"

I used curl to verify that the authKey is still correct, the firebase service works normally and that the firebase certificate has not changed. 
I also reflashed one device and updated the library to the current version, but that did not help.
Do you have any ideas what the problem could be? (Especially, as pushing works fine, but reading a stream doesn't)

Best,
Sebastian

mobizt, 2020-07-12:
Do you use delay or blacking code e.g. sensor reading or long time operation in loop?

Please try to remove those I mentioned first.

Check and print the free heap before and after each Firebase function call in loop.

Debug the WFi in Arduino IDE settings.

Recreate the sketch that has issue and post it here (without third party library) if all above can't help.

Sebbecking, 2020-07-12:
Hi, thanks for your fast reply.

I found the problem with my application: As I mentioned, I push a bit of debugging-information to the database on each startup. This data accumulated over time and grew quite big. 
As the initial readStream returns the full node, all that data was reflected to the client which apparently could not handle it.
The solution was to clear the debugging information in the database regularly.

Thanks for your help and sorry to waste your time :)

Sebastian

closed 2020-07-12 by Sebbecking