mobizt/Firebase-ESP8266 #118: Timestamp as Node Name

CaioFPeres, 2020-07-15:
Hello, i have read all the docs but couldn't figure out how can i get the timestamp, or set as Node/folder name, for organization purposes.

What i want to achieve is something like this:

[https://i.imgur.com/38A7GmT.png](url)

Doesn't necessarily needs to have another timestamp inside node.

Is it possible to do it? Thanks is advance!

mobizt, 2020-07-15:
Sorry I don't understand your question.

Please try the [examples](https://github.com/mobizt/Firebase-ESP8266/tree/master/examples) first for how to use.

CaioFPeres, 2020-07-15:
I would like to set the timestamp on the path.
Something like this:

Firebase.setInt(firebaseData, "Anything/" + Timestamp + "/Variable", variable);

mobizt, 2020-07-15:
What is the type of Timestamp? Number or string?
If it is number you should make it as  string.

**Firebase.setInt(firebaseData, "Anything/" + String(Timestamp) + "/Variable", variable);**

closed 2020-07-15 by mobizt

CaioFPeres, 2020-07-15:
Ok, but how can i get it from firebase servers? I need this timestamp but dont know how to get it. 

In javascript, there is a function called firebase.database.ServerValue.TIMESTAMP which i can assign to a variable, like this:

timestamp = firebase.database.ServerValue.TIMESTAMP;

and then i could set to the path name. How can i do the same using the library?

mobizt, 2020-07-15:
https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Timestamp/Timestamp.ino

CaioFPeres, 2020-07-15:
I understand now, thank you.