mobizt/Firebase-ESP8266 #119: HELP  expected unqualified-id before '-' token

poli44, 2020-07-16:
Hi, when compiling  ESP8266 core 2.72 and use mobizt/Firebase-ESP8266 lib

xxx/Documents/Arduino/libraries/MOBIZT/src/FirebaseESP8266HTTPClient.h:86:41: error: expected unqualified-id before '-' token
 #define HTTPC_ERROR_CONNECTION_REFUSED (-1)
                                         ^
xxx/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h:63:15: note: in expansion of macro 'HTTPC_ERROR_CONNECTION_REFUSED'
 constexpr int HTTPC_ERROR_CONNECTION_REFUSED __attribute__((deprecated)) = HTTPC_ERROR_CONNECTION_FAILED;
               ^
xxx/Documents/Arduino/libraries/MOBIZT/src/FirebaseESP8266HTTPClient.h:86:41: error: expected ')' before '-' token
 #define HTTPC_ERROR_CONNECTION_REFUSED (-1)
                                         ^
xxx/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h:63:15: note: in expansion of macro 'HTTPC_ERROR_CONNECTION_REFUSED'
 constexpr int HTTPC_ERROR_CONNECTION_REFUSED __attribute__((deprecated)) = HTTPC_ERROR_CONNECTION_FAILED;
               ^

Previus version core 2.5.2 works fine. 

Best regards

mobizt, 2020-07-16:
I test compile the library version 2.9.1 with ESP8266 Core SDK 2.72, Arduino IDE v1.8.12 on Windows 10 and doesn't found any error.

Please post your sketch that doesn't work and all the compilations error message.

closed 2020-07-16 by mobizt

mobizt, 2020-07-17:
The library updated which fix the macro name conflicts with ESP8266HTTPClient.h for your case.

Please update.

poli44, 2020-07-29:
thanx