mobizt/Firebase-ESP8266 #120: How to get keys only form JSON

gtrpope19, 2020-07-17:
This is my JSON , How I get time only (9:00,13:00,16:00) and get value in "End" if I don't know keys name of time (9:00,13:00,16:00)

{
    "Monday" : {
      "09:00" : {
          "Attendance" : {
            "1" : {
              "560111222333" : 1,
              "Date" : "11/05/20"
            },
            "dummy" : "dummy"
          },
          "End" : "12:00",
          "Late" : "9:15",
          "Student" : {
            "560111222333" : {
              "FingerprintID" : 1,
              "Name" : "Saranphat moolpho"
            },
            "560333222111" : {
              "FingerprintID" : 2,
              "Name" : "Pattanun punnamsai"
            }
          }
        
      },
      "13:00" : {
         "Attendance" : {
            "1" : {
              "560111222333" : 1,
              "Date" : "11/05/20"
            },
            "dummy" : "dummy"
          },
          "End" : "15:00",
          "Late" : "13:15",
          "Student" : {
            "560111222333" : {
              "FingerprintID" : 1,
              "Name" : "Saranphat moolpho"
            },
            "560333222111" : {
              "FingerprintID" : 2,
              "Name" : "Pattanun punnamsai"
            }
          }
      },
      "16:00" : {
         "Attendance" : {
            "1" : {
              "560111222333" : 1,
              "Date" : "11/05/20"
            },
            "dummy" : "dummy"
          },
          "End" : "18:00",
          "Late" : "16:15",
          "Student" : {
            "560111222333" : {
              "FingerprintID" : 1,
              "Name" : "Saranphat moolpho"
            },
            "560333222111" : {
              "FingerprintID" : 2,
              "Name" : "Pattanun punnamsai"
            }
          }
      }
    },
  "State" : "1 5603051624158"
}

mobizt, 2020-07-17:
Please check your post, reformat or post the image of JSON instead.

closed 2020-07-17 by mobizt

gtrpope19, 2020-07-17:
sorry for my code This is my firebase
[image]

mobizt, 2020-07-17:
There are two approaches.

### 1. Without filter

Get all JSON under the path **/Date/Data/Monday** and iterate to get all the keys and values.

```C++
  if (Firebase.getJSON(firebaseData, "/Date/Data/Monday"))
  {
    FirebaseJson &json = firebaseData.jsonObject();
    size_t len = json.iteratorBegin();
    String key, value;
    int type = 0;
    bool parent = false;
    for (size_t i = 0; i < len; i++)
    {

      json.iteratorGet(i, type, key, value);
      if (!parent)
      {
        if (key.indexOf(":") != -1)
        {
          parent = true;
          Serial.println("Time: " + key);
        }
      }
      else
      {
        if (key == "End")
        {
          Serial.println("End: " + value);
        }
      }
    }
    json.iteratorEnd();
  }
```
It's not recommend for large JSON data under the path **/Date/Data/Monday**.

### 2. Use filter.

Change the value (time string) of the key **End** to Unix timestamp (total seconds from midnight 1 Jan 1970) and filter for the value is in the specific range.

See [this](https://github.com/mobizt/Firebase-ESP8266/issues/84) for how to filter.

For your case, you can use

```C++
    QueryFilter query;

    query.orderBy("End").startAt(60 * 60 * 9).endAt(60 * 60 * 10).limitToLast(1); //9.00 to 10:00

    if (Firebase.getJSON(firebaseData, "/Date/Data/Monday", query))
    {
      if (firebaseData1.dataType() == "json")
      {
        //parse the query result
        FirebaseJson &json = firebaseData.jsonObject();
        size_t len = json.iteratorBegin();
        String key, value;
        int otype = 0;
        for (size_t i = 0; i < len; i++)
        {
          json.iteratorGet(i, otype, key, value);
          if (otype == FirebaseJson::JSON_OBJECT && key.length() > 1)
          {
            Serial.println("Time: " + key);
          }else if (key == "End")
          {
            Serial.println("End: " + value);
          }
        }
        json.iteratorEnd();
        json.clear();
      }
    }
```

And  you must add the indexOn with the key to filter in the database rules

```
{
    "rules": {
        ".read": "auth != null",
        ".write": "auth != null",
              ".indexOn":"temp",
        "Date": {
            "Data": {
                ".indexOn": "End"
            }
        }

    }
}
```

gtrpope19, 2020-07-17:
Thank you for your information . I will try this .

mobizt, 2020-07-17:
The second approach is to reduce to size of JSON per request by filter it before request (get). 
You may need to make multiple requests to meet your requirement but it good for low memory device.

mobizt, 2020-07-17:
I edit the code of above post to use the same JSON iteration.

mobizt, 2020-07-17:
Sorry I edit the above post multiple times. Now it is the final edit.

gtrpope19, 2020-07-17:
Your welcome . I use esp8266 and I'm come frome Thailand . Thank you very much

mobizt, 2020-07-17:
ยินดีครับ

gtrpope19, 2020-07-17:
ขอบคุณมากครับ ตอนแรกใช้ library อีกตัวอยู่ พอมาลองใช้ตัวนี้ดีมากเลยครับ

gtrpope19, 2020-07-17:
ลองใช้โค้ดตัวแรกแล้ว output ออกมาเป็น
Time : 09:00
End :12:00
End :15:00
End :18:00
ถ้าต้องการเวลาจากทั้งสาม keys เลยต้องทำแบบไหนหรอครับ

mobizt, 2020-07-17:
ลืมใส่ parent = false; ต่อจาก if (key == "End") ผมแก้ให้ใหม่แล้วในโพสข้างบน

gtrpope19, 2020-07-17:
ได้แล้วครับ ขอบคุณมากๆเลยครับ ผมลองทำมาหลายวันมาก ขอถามอีกนิดนะครับ คำสั่ง iterator ใช้สำหรับทำอะไรหรอครับ รบกวนด้วยนะครับ

mobizt, 2020-07-17:
เอาไว้อ่าน key/value ของ JSON ทั้งหมดออกมาทีละลำดับครับ ถ้าไม่ระบุหรือรู้พาท ที่จะ parse

gtrpope19, 2020-07-17:
ขอบคุณครับ ขออีกนิดนะครับ if (key.indexOf(":") != -1) อันนี้ทำหน้าที่อะไรครับ แล้วทำไมต้อง json.clear ด้วยครับ

mobizt, 2020-07-18:
indexOf ใช้เพื่อหาจุดเริ่มต้นตัวอักษรใน string ก็คือใช้หาว่ามีตัวอักษรนั้นในข้อความหรือเปล่า ถ้าไม่มีจะคืนค่า -1 ถ้ามีก็คืนค่าเป็นลำดับของตัวอักษรนั้น ถ้าเจออยู่ที่ตำแหน่งแรกของข้อความ ก็คืนค่า 0 ออกมา
ส่วน json.clear เอาไว้ล้างหรือคืนค่าหน่วยความจำที่ใช้ไปในตัวแปร json

gtrpope19, 2020-07-18:
ขอบคุณสำหรับข้อมูลนะครับ ไม่ทราบว่าพอจะมีช่องทางติดตามช่องทางอื่นอีกมั้ยครับ