mobizt/Firebase-ESP8266 #122: ENHANCEMENT

rajmohanjena, 2020-07-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mobizt, 2020-07-27:
@rajmohanjena What is your issue?

rajmohanjena, 2020-07-27:
Sir if we can have a feature in the library to create and call a function when the nodemcu/esp8266 doesnt connect to the internet. It would be great. I am not sure if this is available, kindly make me aware of it.
For EX- We can use it in projects like in terms of Home Automation Systems when the internet is not available we can check it and call a function within which the relays will be directly be controlled from the pins which are connected to other pins through switches..

mobizt, 2020-07-27:
The library provides the error checking which includes server connection error and all server responses errors.

You can check the example and test what's the **firebaseData.errorReason()** returns when the internet connection was disconnected.

mobizt, 2020-07-27:
You also get the http code (number) from server when call the functions with **firebaseData.httpCode()**.

When you call the functions e.g. get, set, push and delete, the http code should be 200 for success operation.

To add additional feature about the internet connection status requires http client to work and use more memory. 

If we use the Firebase Data object to check the server connection status, it will interrupt the stream connection if we use that Firebase Data object for stream too.

That is why that feature is not implemented.

closed 2020-07-27 by mobizt