mobizt/Firebase-ESP8266 #123: Design question - is Firebase and this library the right choice?

bbence84, 2020-07-29:
Hi!
I am planning to create an IoT device (based on ESP8266) which will have multiple users (around 50-100 devices at first). On very high level, the approach is the following:

1. The user registers an account on a mobile app and provisions (adds) the device in the mobile app

2. Once the device is provisioned, it would send periodic even data:
- temp and humidity every 30sec
- ad-hoc event data whenever an event occurs (I could reduce the frequency of this being sent to Firebase by checking if multiple events are very close to each other, then I would report only 1, e.g. if in a rapid succession there are 20 events in 1 minute, I would only send 1)

3. Whenever Firebase recieves an event, I would trigger a Cloud Function and check if a certain threshold is exceeded (e.g. temp too high) and I would send a push notification via FCM to the mobile device

4. There's one use case which would mean that the user triggers an action on the mobile device and the ESP8266 device (a specific device registered to the specific user) should act on it within a few seconds at most. Previusly, I have used MQTT for this and it worked great. But I am wondering if this library and Firebase in general is fit for this purpose. Is it efficient to periodically e.g. poll the database for such actions? Billing quota wise, I am not sure how that would add up, because initially I am shooting for the free plan to be sufficient. :) 

My main concern is point 4, but I am also wondering if the Realtime Database is an overkill for my use case (I understand that Firestore is not really feasible on Arduino / ESP8266 due to the fingerprint requirement).

Thanks a lot in advance!

mobizt, 2020-07-29:
The Firebase RTDB is just the JSON like no-sql database which can be notify the change event to the client.

To use the Firebase in IoT (serverless) project, you should redesign your work flow based on the features available and the limitations of the Firebase RTDB and the client.

Here are the abilities and limitations of Firebase RTDB and this library.

- Firebase RTDB server is not a cloud server i.e. it can not compute or making a decision on any data that being receive or response unless you use the cloud function.

- The security rules for authentication which allow read/write access for specific user and location can't apply when use this library due the database secret was used which will bypass the authentication in security rules.

- This library works with REST APIs which some features may not available as in other platforms e.g. client connected/disconnected event is not available.

- The data change notification event is the HTTP stream event which client will get the response immediately when server push the data to it.

- No guarantee that all simultaneously event data will be received by the client in the stream mode unless the latest (newest) data.

- You can't delay or ignore the data changes event as long as the client stay connected unless client ignored it.

- Use query filter for data retrieving to reduce response data that being process in low memory device.

- Use ETag when many clients push the data to the same path simultaneously to ignore the consequence data from other client that ETag does not match as ETag at the node will change when its data changed.

- Be careful, in the stream mode (realtime event), the data that pushes to client can be large and can't handle by the device.

- You should separate the kept small live (sensor value, status) data from the large static (log, config) data.

- Use device identifier in the database path to identify the devices/clients data.

For Cloud Firestore the fingerprint is not a problem, it's just for network authentiction which can be ignore or use the certificate data instead.

The Firestore supports REST APIs too but the realtime feature is not available due to some performance limit.

The realtime feature is available for client that use the Firebase SDK and RPC (gRPC) capable clients which currently not available for the embedded device.

closed 2020-07-29 by mobizt

mobizt, 2020-07-29:
You may need to test with multiple web client (JavaScript) and some esp8266 devices while you are prototyping the project.

Choose Firebase RTDB if realtime **database** and **serverless** are important, otherwise choose MQTT,