mobizt/Firebase-ESP8266 #125: Question : FirebaseJson with Firebase.setJSON causing reboot (ESP8266)

majo551, 2020-08-03:
Hi
are there any limitations/ rules when setting Firebase.setJSON(firebaseData, path,  **firebaseJson **) ? My code gets regularly rebooted in ESP8266 when I try to save the Json object to Firebase.  I have no issue with any other data type, only with FirebaseJson  object.  Sometimes it works, sometimes it causes reboot:

```
Exception (29):
epc1=0x40236779 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff730 end: 3fffffc0 offset: 0190

````

I tried setting 
firebaseData.setBSSLBufferSize(1024, 1024);  (even larger numbers did not help)
 firebaseData.setResponseSize(1024);
also played with Firebase.setMaxRetry(firebaseData,4)
none helped.

the object cotains 4 nodes (key/value pairs )with 8 (key,value) pairs for each node value

Thanks for advice.
Marian

mobizt, 2020-08-03:
Please post your JSON data and your test or working code.

majo551, 2020-08-03:
I can share the json itself, code contains also sensitive account ids etc,, but my observation is that when I set the individual nodes in cycle rather than the whole object at once , the crash does not seem to appear so far. So it looks the object complexity makes this happen.

here is the full json example:

> [ {
  "clouds" : "100",
  "description" : "overcast clouds",
  "observation_time" : "1596449375",
  "pop" : 0,
  "temp" : 20.18,
  "timezone_offset" : "7200",
  "type" : "current",
  "weather" : "Clouds",
  "weather_id" : 804,
  "wind" : 2.68
}, {
  "clouds" : "100",
  "description" : "overcast clouds",
  "observation_time" : "1596448800",
  "pop" : 0.76,
  "temp" : 20.18,
  "timezone_offset" : "7200",
  "type" : "hourly",
  "weather" : "Clouds",
  "weather_id" : 804,
  "wind" : 2.68
}, {
  "clouds" : "100",
  "description" : "light rain",
  "observation_time" : "1596452400",
  "pop" : 0.76,
  "temp" : 19.57,
  "timezone_offset" : "7200",
  "type" : "hourly",
  "weather" : "Rain",
  "weather_id" : 500,
  "wind" : 0.39
}, {
  "clouds" : "100",
  "description" : "overcast clouds",
  "observation_time" : "1596456000",
  "pop" : 0.73,
  "temp" : 19.27,
  "timezone_offset" : "7200",
  "type" : "hourly",
  "weather" : "Clouds",
  "weather_id" : 804,
  "wind" : 0.81
} ]

and the code that crashes sometimes:

```
 for (int i = 0; i < MAX_FCST_COUNT_ALLOWED; i++)
    {
	
      json.add ("type",weatherData[i].fcstType);  

      json.add("wind",weatherData[i].windSpeed);
      json.add("pop",weatherData[i].pop);
      
      json.add("clouds",String(weatherData[i].clouds));

		  json.add("weather",weatherData[i].main);
		  json.add("weather_id",weatherData[i].weatherId);
		  json.add("temp",weatherData[i].temp);
		  json.add("observation_time",String(weatherData[i].observationTime));
		  json.add("timezone_offset",String(weatherData[i].timezoneOffset));
		  json.add("description",weatherData[i].description);

		  fbData.add(String(i),json);

		 
    }//iterate data
	
	//SAVE TO FB whole weather object at once
		//crashes here....	
	 	Firebase.setJSON(firebaseData,PATH_WEATHER , fbData);

```
	
If I do the saving in the cycle, it dos not seem to crash at all. However I wanted to avoid quick sequential write to FB not to do time expensive network operations too often  and avoid potential issues from FB side when trying to send data in quick sequences one by one.

working code:

```
for (int i = 0; i < MAX_FCST_COUNT_ALLOWED; i++)
    {

      json.add ("type",weatherData[i].fcstType);  

      json.add("wind",weatherData[i].windSpeed);
      json.add("pop",weatherData[i].pop);
      
      json.add("clouds",String(weatherData[i].clouds));

		  json.add("weather",weatherData[i].main);
		  json.add("weather_id",weatherData[i].weatherId);
		  json.add("temp",weatherData[i].temp);
		  json.add("observation_time",String(weatherData[i].observationTime));
		  json.add("timezone_offset",String(weatherData[i].timezoneOffset));
		  json.add("description",weatherData[i].description);

 				Firebase.setJSON(firebaseData,PATH_WEATHER +String(i), json);

    }//iterate data
```

mobizt, 2020-08-03:
To clear memory used, for each round of iteration, please clear internal buffer of json object by add **json.clear()** at the end of the loop.

```C++
 fbData.add(String(i), json);
 json.clear();
```

mobizt, 2020-08-03:
If you don't clear the **json** object for each iteration, the redundant key with new value will be added to it and then added to  **fbData** object and your free heap will be reduced until the point where the device reset error.

majo551, 2020-08-03:
Thanks, I changed the code  and will monitor; 
This small piece did not come to my mind, since the uploaded object looked as expected. :-) Sounds like the solution.
So far no crash.

majo551, 2020-12-28:
I update the code to the latest version  after few months. 

The  reboot issue started again. I traced back and the reboot  started to appear (the code be unstable ) from  the code version 2.9.3   Looks like some change in the library cause the reboot when doing mainly Firebase.setJSON  function with the earlier mention data load .

I did not do any change to the originally posted code which worked until the version 2.9.2  

The latest version (3.0.4)  crashes always on the Firebase.setJSON 

> Exception (3):
> epc1=0x40100e68 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4006f331 depc=0x00000000

mobizt, 2020-12-28:
Can't be compared. The latest version is a very big step changes but I only increase the minor division number and it should be final release of library be for integrate with esp32. 

You should check the free heap.

mobizt, 2020-12-29:
I reopen the issue as this is the bug due to ssl client used after disposed by other processes in stream mode but nothing related to the JSON internal lib.

mobizt, 2020-12-29:
The library updated with the bugs fixed.

closed 2020-12-29 by mobizt

majo551, 2020-12-29:
I tried 3.0.5 ,crashes at the same point 
Exception (3): 
epc1=0x40100e68 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4002f661 depc=0x00000000

does the setJSON take longer than in 2.9.2 , so the longer JSON saving would cause watch dog reboot  ?
Never mind. I might need to stay on 2.9.2 for now. Thanks for checking

mobizt, 2020-12-29:
I recommend you to check the free heap. 
There are no bugs on this topic.

mobizt, 2020-12-29:
Your code has the memory leaks which caused this behavior when the free heap to slow for SSL client and system tasks to work.