mobizt/Firebase-ESP8266 #128: Q?: Recommended rules for Firebase

worthapenny, 2020-08-13:


mobizt, 2020-08-13:
You can set read and write rules to be any i.e. true or false or any condition because the library uses database secret for authentication and the read and write rules will be ignored.

worthapenny, 2020-08-13:
@mobizt Thanks for the speedy response! In that case, (a general question while I got you here), I feel little uncomfortable putting the FIREBASE_HOST and the FIREBASE_AUTH in plain text as I plan to share my project with my non-tech buddies. So, I will have the ESP8266 preprogrammed but still wondering if there is a way to encrypt the host and auth information.

mobizt, 2020-08-13:
Firebase authentication required these information  i.e project url and access token or database secret (used in this library).

No matter what you're program in any platform e.g. android, iOS, web (javascript), Java, theses information always required in source code and you can't ignore or hide them.

mobizt, 2020-08-13:
You can use base64 encode and decode.

You give them the encoded string of host and auth and you need to implement the code to decode on your buddy sketch. 

https://techtutorialsx.com/2017/12/09/esp32-arduino-base64-enconding/

mobizt, 2020-08-13:
The **auth** in database rules i.e. **".read": "auth != null"** is the access token which ESP8266 does not has the memory and tool (SSL/TLS library) that can make these token unless ESP32 can do.

closed 2020-08-13 by mobizt