mobizt/Firebase-ESP8266 #129: HELP

enigma0503, 2020-08-13:
I am running the example sketch named "Beginner_start_here"; but I'm getting the following error on my serial monitor:

Error in setInt, unauthorized, Could not parse auth token.
Error in getInt, unauthorized, Could not parse auth token.
Error in get, unauthorized, Could not parse auth token.

[image]

mobizt, 2020-08-13:
Did you change [this](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Beginner_start_here/Beginner_start_here.ino#L21) with your database secret?

enigma0503, 2020-08-13:
Yes, I've replaced that with my url and the corresponding key. Also, read and write rules in firebase are set to "true"

mobizt, 2020-08-13:
The error returned from Firebase server which means the **FIREBASE_AUTH** you provided is not valid.

You should copy it from  the console as in the image below.

[image]

closed 2020-08-13 by mobizt