mobizt/Firebase-ESP8266 #130: please help

milinddhanke, 2020-08-25:
please help me to set integer in path, a simple line, i am new about your library, i just wanted to send data of lm35 temperature sensor.

mobizt, 2020-08-25:
Please start with [this example](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Beginner_start_here/Beginner_start_here.ino).

And others examples [here](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples).

closed 2020-08-25 by mobizt

milinddhanke, 2020-08-29:
SORRY TO DISTURB YOU, i need one more help in stream program , i'm using Firebase.setInt(firebaseData, "DM3/DM3OUT/R1",incomingdata); this is i did because whatever i,m get from stream i want it set in other path of firebase. now problem is as setInt executed again the stream data is coming. that i don't want. Please help.

mobizt, 2020-08-29:
You should prevent this recursion by re-design the database structure.

Exclude the the data that you want to manually set from the stream path (included its children nodes).

The data for configuration, settings and large history (log) data should separate from kept small stream data.

milinddhanke, 2020-08-29:
#include <ESP8266WebServer.h>
#include <WiFiManager.h> 

#include <ESP8266WiFi.h>
#include "FirebaseESP8266.h"

#define FIREBASE_HOST "xxxxxxxxxxxxxxxxxxxxxxxx" 
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxx"

//Define FirebaseESP8266 data object
FirebaseData firebaseData;

String path = "DM3/DM3IN/PUT";

uint16_t count = 0;

void printResult(FirebaseData &data);
int incomingdata = 0;

void setup()
{
 
  Serial.begin(115200);
  WiFiManager wifiManager; 
  //wifiManager.resetSettings();     
  wifiManager.autoConnect("Connect T0 AP");   
  Serial.println("connected...yeey :)");

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);  
  firebaseData.setBSSLBufferSize(1024, 1024);
  firebaseData.setResponseSize(1024);

  if (!Firebase.beginStream(firebaseData, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();    
  }
}

void loop()
{
   
  if (!Firebase.readStream(firebaseData))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't read stream data...");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  if (firebaseData.streamTimeout())
  {
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }

  if (firebaseData.streamAvailable())
  {   
    printResult(firebaseData);  
    if (incomingdata == 12345) 
    {Serial.println("ha ha ha");      
    Firebase.setInt(firebaseData, "DM2OUT",incomingdata);        
    }    
   }     
}

void printResult(FirebaseData &data)
{

  if (data.dataType() == "int")
   { Serial.println(data.intData());
     incomingdata = data.intData(); 
   }  
}

[image]

milinddhanke, 2020-08-29:
please help me to sort my problem.

mobizt, 2020-08-29:
Problem?

milinddhanke, 2020-08-29:
it is continuously repeating!

milinddhanke, 2020-08-29:
Actually whatever i get from steam i want it again send to firebase for confirmation whether data is received by esp or not.

milinddhanke, 2020-08-29:
there may some network or internet issue, so need this confirmation.

milinddhanke, 2020-08-29:
i'm integrating esp+firebase+android app(made in MIT app inventor)

mobizt, 2020-08-29:
Did you know you get the repeat data?

When you want to print something to serial for debugging, you need to include some extra information where this data comes from.

milinddhanke, 2020-08-29:
Did you know you get the repeat data?>>>>>>>>>>>>yes, i could see from serial monitor.

When you want to print something to serial for debugging, you need to include some extra information where this data comes from.>>>>>>>>>>>>>couldn't understand!

mobizt, 2020-08-29:
Do not repeat the question when you answer, its not polite OK?

How you prove that the data is repeated when you print only the same value i.e. ha ha ha and 12345.

milinddhanke, 2020-08-29:
Sorry!!!!!!!!!

Because, i'v written in program when the value is 12345, if (incomingdata == 12345) 
    {Serial.println("ha ha ha");  and i wanted that value (whatever the value) again send to firebase via 
Firebase.setInt(firebaseData, "DM3/DM3OUT/R1",incomingdata); 

i'm not much expert in programming!!!!

mobizt, 2020-08-29:
Your stream monitoring path is **DM3/DM3IN/PUT** and the storing value path that is not specified in your above code is **DM3/DM3OUT/R1**, both are difference and you **will not** get the repeated data from stream when you set data. 

You should print some extra text to let you know about where the data is from like this.

Stream data: 123456
Stored data (set): 123456 

FYI, When the device first established the stream connection,  server will send the data at the streaming path to the device once.

milinddhanke, 2020-08-29:
thank you, i'm trying to set

milinddhanke, 2020-08-29:
Thanks for your support, my problem is solved. In future, if i come to your country i'll contact you, we'll meet. Great work you did, i was waiting for your library !!!!!!!!