mobizt/Firebase-ESP8266 #132: HELP

milinddhanke, 2020-09-03:
Please help, something has changed in google firebase, my realtime data is disappear and database secrete is also have problem. 
This is massage  "Unable to load the database secrets"

mobizt, 2020-09-03:
Firebase service is online normally.

Check your internet connection and you may delete your web browser cache or use other web browsers to sign in or use your other google/gmail  accounts to sign in or create new project.

closed 2020-09-03 by mobizt

milinddhanke, 2020-09-03:
Thank you, I test it using other operator internet like Airtel, Vodaphone, BSNL (India), But it is not working with JIO (India) network.
Any Suggestion.

mobizt, 2020-09-03:
Ask Google representative on your country or Firebase Facebook page.

milinddhanke, 2020-09-03:
ok, Thanks

milinddhanke, 2020-09-04:
As per your suggestion i mail to google firebase, and replied as follows!!!!!!!!!!!!

My name is Jonathan, thanks for reaching out.

 
We've seen multiple reports about similar issues, and from what we have found so far is that this is very likely a network issue, related to the internet provider.

 
Now, one user reported a DNS error, so one thing to try is using public DNS servers to verify if this is the issue. For example, from Google, you can use these: 8.8.8.8 and 8.8.4.4 (click here for more details).

 
Finally, if that confirms it, what I suggest is to open the browser's developers tools and to gather the logs from the "Network" tab when you go to the Database Dashboard and get in touch with your provider to ask about this. I'll check the team to see what we can do, but getting in touch with the provider looks like the best option at this moment.

 
So, let me know how it goes, and feel free to reach out with any further questions.

Kind regards,

Jonathan

Any comment from your side??????????

mobizt, 2020-09-04:
Did you try it? I mean set up your PC network adapter to use DNS.

Can you go to Firebase console (website)? If yes, and you still get the problem in getting the information from console, you may browsing through the proxy server/gateway which you need to contact your ISP.