mobizt/Firebase-ESP8266 #134: unsupported Unsigned Long into firebasejson

ahmaddiqi, 2020-09-09:
I want to add unsigned long variable to json. but, I can't. because, if I convert it to String, I can't process it

I found error like this 
`undefined reference to bool Firebase Json::set<unsigned long>(String const&, unsigned long)`

I hope can be solved 👍

mobizt, 2020-09-09:
Unsigned long is not supported by the FirebaseJson library.
The library supported int, float and double data types for the number.

closed 2020-09-09 by mobizt

eligt, 2020-12-07:
Note that while it is true that Firebase advertises support for `int` only, Firebase's `int`s are actually 64bit which means that their API supports both `long` as well as `long long` integers.

I've tested this by successfully pushing some 32bit `long unsigned int`s (for instance the return value from `millis()`) by adding the following method to `FirebaseJson`:

```cpp
void FirebaseJson::set(const String &path, unsigned long int value)
{
    char BUF[36];
    sprintf(BUF, "%lu", value);
    _set(path.c_str(), BUF);
    std::string().swap(_jsonData._dbuf);
}
```

mobizt, 2020-12-07:
This is library for embedded device and will not support many types of data due to cpu/mcu architectures and platform dependent.