mobizt/Firebase-ESP8266 #137: Terminate firebase connection

lukagrbec, 2020-09-14:
Hello,

I was wondering if there is a way to end the connection with Firebase. For example if there is a command firebase.begin(), is there Firebase.end()? I noticed in /src/FirebaseESP8266HTTPClient.cpp that there is:

 FirebaseHTTPClient::~FirebaseHTTPClient()
{
  _client->stop();
}

But I'm not sure how to use it. Do I need to add any libraries or are there other issues? Can anyone help me with a workaround?

Thanks in advance!

mobizt, 2020-09-14:
No end function is available.

To just stop (pause) the Firebase Data object for all Firebase call, please use
**firebaseData.pauseFirebase(true);**

To end the stream that assign to Firebase Data object that works for stream, please use
 **Firebase.endStream(firebaseData);**

The above examples assumed that firebaseData is the declared Firebase Data object and all resource (memory) still be reserved.

If you want to destroy the Firebase Data object to gain memory, you can declare Firebase Data object for store/restore data in local scope and it's destructor function (~) will call automatically when exit that scope.  The Firebase Data object that works for stream must be declared out of any scope (global).

You can call the destructor function of Firebase Data object with  function **delete** to deallocate the object when you declare or allocate the Firebase Data object dynamically with the **new** operator.

Here the example to allocate/deallocate the Firebase Data object dynamically.

```C++

#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

#define FIREBASE_HOST "YOUR_FIREBASE_PROJECT.firebaseio.com" //Without http:// or https:// schemes
#define FIREBASE_AUTH "YOUR_FIREBASE_DATABASE_SECRET"
#define WIFI_SSID "YOUR_WIFI_AP"
#define WIFI_PASSWORD "YOUR_WIFI_PASSWORD"

//Define FirebaseESP8266 data object
FirebaseData *firebaseData1 = new FirebaseData();
FirebaseData *firebaseData2 = new FirebaseData();

unsigned long sendDataPrevMillis = 0;

String path = "/Test/Stream";

int count = 0;

bool deallocated = false;

void printResult(FirebaseData &data);
void printResult(StreamData &data);

void streamCallback(StreamData data)
{

  Serial.println("Stream Data1 available...");
  Serial.println("STREAM PATH: " + data.streamPath());
  Serial.println("EVENT PATH: " + data.dataPath());
  Serial.println("DATA TYPE: " + data.dataType());
  Serial.println("EVENT TYPE: " + data.eventType());
  Serial.print("VALUE: ");
  printResult(data);
  Serial.println();
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
  {
    Serial.println();
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
}

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
[... 249 lines cut ...]
    {
      Serial.print(i);
      Serial.print(", Value: ");

      FirebaseJsonData *jsonData = data.jsonDataPtr();
      //Get the result data from FirebaseJsonArray object
      arr->get(*jsonData, i);
      if (jsonData->typeNum == FirebaseJson::JSON_BOOL)
        Serial.println(jsonData->boolValue ? "true" : "false");
      else if (jsonData->typeNum == FirebaseJson::JSON_INT)
        Serial.println(jsonData->intValue);
      else if (jsonData->typeNum == FirebaseJson::JSON_FLOAT)
        Serial.println(jsonData->floatValue);
      else if (jsonData->typeNum == FirebaseJson::JSON_DOUBLE)
        printf("%.9lf\n", jsonData->doubleValue);
      else if (jsonData->typeNum == FirebaseJson::JSON_STRING ||
               jsonData->typeNum == FirebaseJson::JSON_NULL ||
               jsonData->typeNum == FirebaseJson::JSON_OBJECT ||
               jsonData->typeNum == FirebaseJson::JSON_ARRAY)
        Serial.println(jsonData->stringValue);
    }
  }
}

```

closed 2020-09-14 by mobizt

mobizt, 2020-09-14:
If you mean the http connection, the connection was closed after server sent the response. 

The close method in the destructor function of FirebaseHTTPClient is to ensure the http connection closed already before destructed the object.

The class FirebaseHTTPClient designed for internal use, to remove unused and unnecessary functions for the header and payload handlers as in the HTTPClient library.

mobizt, 2020-09-14:
If you want to gain the free memory and stop using the Firebase Data (that holds the WiFiClientSecure instance) during process your another network tasks by using the WiFiClientSecure or HTTPClient objects, you can use the allocation/deallocation code example in my first reply.

lukagrbec, 2020-09-14:
Hi,

thanks for your help. I really appreciate your effort. I will definitely try implementing the code you sent. The reason for as to why I posted this question was that I wrote a program which first establishes a connection with Firebase and queries some data, then if a certain condition is satisfied it sends a SMS using Twilio. However, the two parts (querying from Firebase and sending SMS) work fine separately, but not together. I think that the reason for this might be that both use http clients and they might interfere with each other. So, I wanted to close the Firebase http connection before connecting to Twilio. 
But if I understand you properly the connection closes itself automatically after it queries some data? Do you think the two libraries could be interfering with each other because of web-clients?

Best regards,
Luka

mobizt, 2020-09-14:
No interfearing as it is single core mcu.  But it will be good to destroy some Firebasse Data object to release the memory.

The Firebase Data object holds the WiFiClientSecure instance which declare as static (global) as in all examples.  If you use two Firebase Data objects for data store/restore and stream, it consumed memory about 20k or 10k for each object which mostly used by WiFiClientSecure and bearSSL that create internally.

If you declare another WiFiClientSecure too and consumed the large memory too , the memory may not enough that's why I recommend to destroy the Firebase Data object instead. You may let the Firebase Data object that works for stream continue running and destroy only the Firebase Data object that used for store/restoring data to gain about 10k memory.

lukagrbec, 2020-09-16:
Hi,

thank you for your reply. I implemented the code and it works fine, however, is there a way to not permanentely delete the firebaseData? The problem is that I am measuring temperatures and sending them to Firebase. Besides that there is a value stored in Firebase that needs to be queried to see, whether the temperatures have exceeded that value and if they have, a SMS should be sent. So, I need to be able to continue pushing data and also clear the memory every once in a while. Do you ahve any suggestions for that? Thanks again!

Kind regards,
Luka

mobizt, 2020-09-16:
You just construct it again any time.

```C++
firebaseData2 = new FirebaseData();
firebaseData2->setBSSLBufferSize(1024, 1024);
firebaseData2->setResponseSize(1024);

```

and again can delete when not use.
```C++
delete  firebaseData2;

```
Or declare Firebase Data object as local object in the blocks or functions when you want to use as

```C++
FirebaseData firebaseData;;
firebaseData.setBSSLBufferSize(1024, 1024);
firebaseData.setResponseSize(1024);
```

and it will destroy (destructed) when exit the function

lukagrbec, 2020-09-19:
Hello,

I managed to make it work. Thanks for all your help. I just have one more question. I'm trying to push the temperature and timestamp at the same time (to the same node). This is my code:
`temp1 = sensors.getTempCByIndex(0);
 StaticJsonBuffer<256> jsonBuffer;
JsonObject& timeStampObject = jsonBuffer.createObject();
timeStampObject[".sv"] = "timestamp";

JsonObject& root = jsonBuffer.createObject();
JsonArray& timestamp = root.createNestedArray("timestamp");
timestamp.add(timeStampObject);
JsonArray& temperature = root.createNestedArray("temperature");
temperature.add(temp1);
`
However, when I try to push it using Firebase.push(*firebaseData2, path + "/Int", root) I get an error. I guess that Firebase.push can't push a json array. Do you have any advice on how I could push this data? (If you would like to see a screenshot of how I want Firebase to be structured follow this link: https://imgur.com/a/IVWYwh7 ).

Thank you for taking the trouble to help me. I do appreciate it.

Cheers,
Luka

mobizt, 2020-09-19:
My library has built-in FirebaseJson, the JSON parser and builder.

Other third party JSON libraries are not supported.

Please check the examples.

mobizt, 2020-09-19:
First, you need to learn how FirebaseJson works.
https://github.com/mobizt/Firebase-ESP8266#parse-create-and-edit-json-objects

then
https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/jsonObject/jsonObject.ino
https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/jsonArray/jsonArray.ino

lukagrbec, 2020-09-20:
Hello,

thanks again. I greatly apprecitate your help. Code now works and I am able to push arrays. The only problem is that I can't add a timestamp to the array. I saw in your library (/examples/Timestamp/Timestamp.ino) that there is a command to push timestamps to Firebase (Firebase.pushTimestamp). However, I would like to add a timestamp to an array and then push that array. Is there a way to do that? 
Besides that I was wondering if there is any way to change the value that I read from Firebase from string to int? I noticed that firebaseData.toInt() doesn't work. 
Thanks for your time.

Best regards,
Luka

mobizt, 2020-09-20:
The function for timestamp is actually set the special JSON payload that represents the server value key/value as 
 **{".sv": "timestamp"}** 

The timestamp of server will automatically set to that node when calling the setTimestamp function.
Then the miiliseconds (timestamp) can get from the getDouble function.

If you want to store the timestamp (number) as normal data, you can get the timestamp from RTC chip or get the time_t (or uint32_t) variable from the **time** function as in the NTP examples that available on the internet e.g.
**time_t timestamp = time(nullptr);**

mobizt, 2020-09-20:
Firebase Data object is the class which nothing related to Arduino data types conversion function.

The method **toInt** is from Arduino's String class not Firebase library classes

lukagrbec, 2020-09-23:
Hi,

thanks for your reply. I got the timestamp from time_t timestamp = time(nullptr); however, I get an error when I try to push it in an array with another integer. Do you know how I could solve this error?
I looked through available functions and methods and noticed that there isn't a function to convert the Firebase Data object from string to int. Is there any way to bypass that? Could you store the string of the Firebase Data object as an Arduino data type to be able to use the functions? Or is there another way? 

Kind regards,
Luka

mobizt, 2020-09-23:
You can use integer like this
`int timestamp = (int)(time(nullptr));`

lukagrbec, 2020-09-26:
Hi,

thanks for your relpy. After implementing your solution the code worked well for one set and one get. However, if I add one more get and one more set (altogether one push, two sets and two gets) Twilio message isn't sent. I believe because there isn't enough heap. Do you have any suggestions how to solve this? I also posted this question with code included on Arduino Stack Exchange (please follow this link: https://arduino.stackexchange.com/questions/78213/esp8266-nodemcu-running-out-of-heap?noredirect=1#comment173013_78213).
I greatly appreciate your help and would like to make a contribution to your project. Therefore, I would like to know if you still use the Paypal account posted on Github (paypal.me/mobizt). 

Best regards,
Luka

mobizt, 2020-09-27:
You should debug the free heap yourself before and after every Firebase function calls.

You need to choose and manage the memory usage yourself.

I can't correct of fix the code for you unless only guide.

mobizt, 2020-09-27:
You also can use the internal WiFiClientSecure object inside the FirebaseData object.
This way is safe for you without to reallocate the FirebaseData object.

1. Declare FirebaseData object as global variable.

     **FirebaseData firebaseData;**

2. When you want to pause the Firebase operation which close the current connection to Firebase too.

     **firebaseData.pauseFirebase(true);**
     **firebaseData.clear();**

 3. Create your local WiFiClientSecure object which copy the reference from internal WiFiClientSecure

     **WiFiClientSecure client = firebaseData.getWiFiClient();**

4. Use the WiFiClientSecure object which you created.

5. When you want to call the Firebase functions, resume the FirebaseData object operation.

     **firebaseData.pauseFirebase(false);**

lukagrbec, 2020-10-02:
Hi, for the past few days I was modifying the code. I used all the commands you sent in the previous post except the WiFiClientSecure client = firebaseData.getWiFiClient();. When I have one get and one push the code works fine and the message get sent (free heap right before sending the message is 24k). However, when I add one more get and one more set, the message doesn't get sent, even though the heap right before sending the message is also 24k. Do you have any suggestions about what could be wrong since the free heap is the same?

Thanks,
Luka

mobizt, 2020-10-02:
You should use the same Firebase Data object (only one) for all operation except for stream which should be another Firebase Data object.

Each Firebase Data object hold the WiFiClientSecure which use large memory (> 50k each) and most of them reserved in the mbedTLS  resources then Firebase Data object is not a normal variable and declare it one as global for all operations is enough.

Please update the library to latest (v 2.7.9) due to the slow connection bug in v 2.7.6 to 2.7.8.

mobizt, 2020-10-02:
In te new version library, you can use **firebaseData.stopWiFiClient()** to close the Firebase connection which will release the memory used by the internal WiFiClient.

mobizt, 2020-10-02:
I edit the above reply due to mistake.

lukagrbec, 2020-10-02:
Thanks for your prompt reply. I tried using WiFiClientSecure client = firebaseData.getWiFiClient();, however, wherever I place it in the code I get an error: "firebaseData was not declared in this scope". Could you please provide a bit more information on where to set the variable?

mobizt, 2020-10-03:
That's a example, firebaseData is the Firebase Data object which may be any name you chosen.