mobizt/Firebase-ESP8266 #138: Actually not an Issue or Feature, Google requires  security rules!

floatAsNeeded, 2020-09-16:
Hi all!

I saw recently Google send me a warning saying that it could stop receiving client requests if I don't modify security rules on my realtime database. Because at the moment they are set to allow read and write from anyone.

```
{
  /* Visit https://firebase.google.com/docs/database/security to learn more about security rules. */
  "rules": {
    ".read": true,
    ".write": true
  }
}
```

I'm using a Wemos D1 mini pro and I was wondering what security rules I could use that would work also with this library.

Does someone know anything?

Thank you!

mobizt, 2020-09-16:
The database rules for read and write can set to false or anything you want.

The database secret that uses as the authentication by the library will always access to read and write data no matter the database rules settings.  

The read and write rules are for the Firebase SDK client (java, Android, iOS and web) only.

closed 2020-09-16 by mobizt

floatAsNeeded, 2020-09-16:
Thank you very much for the info!