mobizt/Firebase-ESP8266 #139: HELP: parse array

xralphack, 2020-09-16:
```
void setup()
{
  Serial.begin(9600);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
  directData.setBSSLBufferSize(1024, 1024);

  //Set the size of HTTP response buffers in the case where we want to work with large data.
  directData.setResponseSize(1024);

  if (!Firebase.beginStream(directData, "/0/c"))
  {
    Serial.println("Could not begin stream");
    Serial.println("REASON: " + directData.errorReason());
    Serial.println();
  }
}

void loop()
{
  if (!Firebase.readStream(directData))
  {
    Serial.println();
    Serial.println("Can't read stream data");
    Serial.println("REASON: " + directData.errorReason());
    Serial.println();
  }

  if (directData.streamTimeout())
  {
    Serial.println();
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }

  if (directData.streamAvailable())
  {
    printResult(directData);
  }
}

void printResult(FirebaseData &data)
{

  if (data.dataType() == "int")
    Serial.println(data.intData());
  else if (data.dataType() == "float")
    Serial.println(data.floatData(), 5);
  else if (data.dataType() == "double")
    printf("%.9lf\n", data.doubleData());
  else if (data.dataType() == "boolean")
    Serial.println(data.boolData() == 1 ? "true" : "false");
  else if (data.dataType() == "string")
    Serial.println(data.stringData());
  else if (data.dataType() == "json")
  {
    Serial.println();
    FirebaseJson &json = data.jsonObject();
    //Print all object data
    Serial.println("Pretty printed JSON data:");
    String jsonStr;
    json.toString(jsonStr, true);
    Serial.println(jsonStr);
    Serial.println();
    Serial.println("Iterate JSON data:");
    Serial.println();
    size_t len = json.iteratorBegin();
    String key, value = "";
    int type = 0;
    for (size_t i = 0; i < len; i++)
    {
      json.iteratorGet(i, type, key, value);
      Serial.print(i);
      Serial.print(", ");
      Serial.print("Type: ");
      Serial.print(type == FirebaseJson::JSON_OBJECT ? "object" : "array");
      if (type == FirebaseJson::JSON_OBJECT)
      {
        Serial.print(", Key: ");
        Serial.print(key);
      }
      Serial.print(", Value: ");
      Serial.println(value);
    }
    json.iteratorEnd();
  }
  else if (data.dataType() == "array")
  {
    Serial.println();
    //get array data from FirebaseData using FirebaseJsonArray object
    FirebaseJsonArray &arr = data.jsonArray();
    //Print all array values
    Serial.println("Pretty printed Array:");
    String arrStr;
    arr.toString(arrStr, true);
    Serial.println(arrStr);
    Serial.println();
    Serial.println("Iterate array values:");
    Serial.println();
    for (size_t i = 0; i < arr.size(); i++)
    {
      Serial.print(i);
      Serial.print(", Value: ");

      FirebaseJsonData &jsonData = data.jsonData();
      //Get the result data from FirebaseJsonArray object
      arr.get(jsonData, i);
      if (jsonData.typeNum == FirebaseJson::JSON_BOOL)
        Serial.println(jsonData.boolValue ? "true" : "false");
      else if (jsonData.typeNum == FirebaseJson::JSON_INT)
        Serial.println(jsonData.intValue);
      else if (jsonData.typeNum == FirebaseJson::JSON_FLOAT)
        Serial.println(jsonData.floatValue);
      else if (jsonData.typeNum == FirebaseJson::JSON_DOUBLE)
        printf("%.9lf\n", jsonData.doubleValue);
      else if (jsonData.typeNum == FirebaseJson::JSON_STRING ||
               jsonData.typeNum == FirebaseJson::JSON_NULL ||
               jsonData.typeNum == FirebaseJson::JSON_OBJECT ||
               jsonData.typeNum == FirebaseJson::JSON_ARRAY)
        Serial.println(jsonData.stringValue);
    }
  }
  else
  {
    Serial.println(data.payload());
  }
}

```

[image]

### result

```
Connecting to Wi-Fi.....................
Connected with IP: 192.168.66.18

Pretty printed Array:

    "root": [
    10,      
    20,      
    30
]

Iterate array values:
```

I can not get iterate array values?

mobizt, 2020-09-16:
Thy to avoid use the number as top level (root) key in Firebase which is lead to error in interpretation when this payload parses with the FirebasseJson.

mobizt, 2020-09-17:
That's a bug in the library and I will let you know when it fixed.

xralphack, 2020-09-17:
I saw this problem when I specify the path to array data,

mobizt, 2020-09-17:
The issue was fixed and please update the library to v 2.9.4 (download via GitHub).

closed 2020-09-17 by mobizt

xralphack, 2020-09-17:
Will you release to PlatformIO lib registry?

mobizt, 2020-09-17:
What do you mean?

mobizt, 2020-09-17:
Ok I get it.
The Firebase libraries for ESP8266 and ESP32 are available in the pio registry for years, you can search it.

xralphack, 2020-09-17:
https://platformio.org/lib/show/6247/Firebase%20ESP8266%20Client

current version is 2.9.3 (released 4 days ago)

mobizt, 2020-09-17:
You need to wait, the system will auto update.

In pio on vscode, you can click at the line #include <FirebaseESP8266.h> and press F12 to see the header file then press Shift + Alt + R to see the library file in the explorer then download the library zip file and extract and replace all.

xralphack, 2020-09-17:
got it, I will try later. Thanks for your quick response~

xralphack, 2020-09-17:
Iterate array values works now, another question is once I update value on firebase console. only changed value being printed, is it possible to get whole array instead of the changed item

```
Pretty printed Array:
[
    20,
    30,
    40
]

Iterate array values:

0, Value: 20
1, Value: 30
2, Value: 40
```

once I update 20 > 50

```
printed

50
```

```
however I expect

Iterate array values:

0, Value: 50
1, Value: 30
2, Value: 40

```

mobizt, 2020-09-17:
No the Firebase server only push the changes to client.

mobizt, 2020-09-17:
For stream

The absolute of parent node path
**firebaseData.streamPath()**

The relative child path that its value changes
**firebaseData.dataPath()**

Then the absolute child path is 
**firebaseData.streamPath() + firebaseData.dataPath()**

xralphack, 2020-09-17:
So I have to fetch data again via absolute child path

mobizt, 2020-09-17:
Why?

xralphack, 2020-09-17:
Ok I get it now. I can update local array based on path and changed value

mobizt, 2020-09-17:
Once the stream establish connection at the first time, it will return all data under the stream path and you can hold it in FirebaseJson object or FirebaseJsonArray for use later and you can change/add its data based on the data path that changes (relative child path) when the stream data is available.

xralphack, 2020-09-17:
cool~ I will try it

xralphack, 2020-09-17:
```

void streamCallback(MultiPathStreamData data)
{
  Serial.println();
  Serial.println("Stream Data1 available...");

  size_t numChild = sizeof(childPath) / sizeof(childPath[0]);

  for (size_t i = 0; i < numChild; i++)
  {

    if (data.get(childPath[i]))
    {
      Serial.println("path: " + data.dataPath + ", type: " + data.type + ", value: " + data.value);
    }
  }

  Serial.println();
}

Stream Data1 available...
path: /c, type: array, value: [[20,40]]
path: /d, type: array, value: [[50,30]]
```

I want to build FirebaseJsonArray for /c and /d separately for later

data.value is type of string

I can not find something like data.jsonArray();

xralphack, 2020-09-18:
ok, I would start from single path.

You mentioned:
you can change/add its data based on the data path that changes (relative child path) when the stream data is available.

This part takes a lot of time and seems to have a chance to be automated, what do you think?

mobizt, 2020-09-18:
What do you mean it takes a lot of time? 

You should have local data in case of the dataset (may be a FirebaseJson object) that being process more often as you want and also can be store as string to file to flash or SD card and then restore to save memory.

When everything done and ready to update to remote server, just send it a whole data set at once which faster than sending multiple times of small data. 

When remote data changes and stream event occurred, just update the local dataset based on data path e.g update the whole dataset when the data path is /,  and update the specific path if the data path is "/xxx.

xralphack, 2020-09-18:
The following logic can be handled automatically?
e.g update the whole dataset when the data path is /, and update the specific path if the data path is "/xxx.

mobizt, 2020-09-18:
This is a library not application.

mobizt, 2020-09-18:
I gave you a concept which is possible to do and you must design and test for your requirement.

xralphack, 2020-09-18:
My opinion is that synchronization between local and remote is a common usage scenario

Anyway, thank you for your contribution

mobizt, 2020-09-18:
Google implement their own socket communication in their Firebase SDK which impossible to use in unsupported platform and environment like embedded system.

We can only use Firebase through the Firebase REST (http) API which is about the http request and response and the http stream which the data transfer is one way at a time, no socket, no sync action.

Then what you are asking is not natively possible and you need to implement your own from the available features that provided by this library which you need to experiment to know how it works and design to meet your requirement.

Firebase is also different from MQTT as it is not a cloud (computing) server, it can't compute the data or user command, it's just the cloud storage.