mobizt/Firebase-ESP8266 #141: HELP

lchvsanilkumar, 2020-09-20:
Is their any example to multipath stream data from parentnode and asusal from childnode ....but I know the example of multipath stream ...but I want to steam data from parentnode and also childnode at one place .... thank you for your reply

mobizt, 2020-09-20:
The multipath stream is actually normal stream which use the class MultiPathStreamData for parsing the result data based on the predefined child node array.

The result from stream is parsed in that way you can also add "/" as child path (actually the parent path as you use /) which any child data changes will be get through stream.get("/") which is not taken any advantage from the multiple path stream concept.

> You should know that Firebase server only push the data to client that based on whether the child value changes (minor update) or whole parent data changes (replaced by some new data),

**In the first case, some of the child node value changes, the stream will push only the new value of that child node to client.
And in the later case, whole data changes (parent data replaced), the stream will push whole new data of the parent (stream) path.**

All I mentioned are the behavior that Google designed, library or other client can do nothing.

closed 2020-09-20 by mobizt

lchvsanilkumar, 2020-09-20:
Thanks sir...you help me a lot...now I am getting data