mobizt/Firebase-ESP8266 #142: Can't set JSON in path with Russian words

danmov26, 2020-09-20:
[image]

mobizt, 2020-09-20:
You should save your sketch file (.ino)  with UTF-8 encoded text and reopen it.

If you're using PlatformIO or the new version of Arduino IDE, it will save as UTF-8 encoded document by default when file contains (.ino, .h, .cpp) non-ASCII characters.

Please update Arduino IDE or open your sketch in text editor and save as UTF-8 encoding text file and rename to .ino later.

closed 2020-09-20 by mobizt

danmov26, 2020-09-22:
I couldn't solve this problem. I did everything as you said

mobizt, 2020-09-22:
If you can't do this you may have the problem with the system (PC or IDE) somewhere as I'm Thai and can store both key and value in Thai  too.

mobizt, 2020-09-22:
Firebase supports UTF-8 for [key](https://firebase.google.com/docs/database/admin/structure-data) and [value](https://firebase.google.com/docs/database/admin/structure-data).

mobizt, 2020-09-22:
This is my test for Thai text.

This is the sketch which is UTF-8 encoded.

[sketch_sep22a.zip](https://github.com/mobizt/Firebase-ESP8266/files/5261219/sketch_sep22a.zip)

The serial printing message
```

................
Connected with IP: 192.168.137.80

------------------------------------
Set JSON test...
PASSED
PATH: /Test
TYPE: json
ETag: 3mcvVydpfJifhyseqXQt0ifdB9c=
VALUE: 
Pretty printed JSON data:
{
    "ดอกไม้": [
        "ภาษาไทย",
        123
    ],
    "ตัวอักษรไทย": "ต้นไม้"
}

Iterate JSON data:

0, Type: object, Key: ดอกไม้, Value: ["ภาษาไทย",123]
1, Type: array, Value: ภาษาไทย
2, Type: array, Value: 123
3, Type: object, Key: ตัวอักษรไทย, Value: ต้นไม้
------------------------------------

```
The database result

[image]

mobizt, 2020-09-22:
Russian version

[russian.zip](https://github.com/mobizt/Firebase-ESP8266/files/5261311/russian.zip)

```
................
Connected with IP: 192.168.137.80

------------------------------------
Set JSON test...
PASSED
PATH: /Test
TYPE: json
ETag: NY1BqPhaL/p1aujIYnIFxiJkedo=
VALUE: 
Pretty printed JSON data:
{
    "Мы хотим снять квартиру в городе": "Я снял квартиру рядом с вашей.",
    "Она пригласила нас к себе на дачу": [
        "Эта квартира слишком дорогая",
        123
    ]
}

Iterate JSON data:

0, Type: object, Key: Мы хотим снять квартиру в городе, Value: Я снял квартиру рядом с вашей.
1, Type: object, Key: Она пригласила нас к себе на дачу, Value: ["Эта квартира слишком дорогая",123]
2, Type: array, Value: Эта квартира слишком дорогая
3, Type: array, Value: 123
------------------------------------

```

[image]

Note that the key should not contains dot character which is not valid for Firebase.

mobizt, 2020-09-22:
Did you hard code the Russian characters? If not, how do you making the Russian text?

danmov26, 2020-09-22:
If you write a Russian word in the firmware in advance, then everything will be displayed correctly in Firebase RTDB.

I am writing firmware for my Smart Switch.
In my firmware, the device by default creates JSON in / Devices / Without group / in the group "Without group" (without group).
[image]
Then, in my Android application, I change the value in / Devices / No group / Group to "Комната". The smart switch reads this, stores it in the EEPROM.
[image]
[image]
Reboots, deletes the old path and creates a new path using the saved string from the EEPROM.
[image]
And here's the problem.
[image]

danmov26, 2020-09-22:
Just now, while I was writing previous comment, I noticed that if you write a Russian word with spaces, then everything is displayed normally.

i.e, if I call the group "Комната", then it will turn out:
[image]
If I name the group with a space "Комната ", then everything will be displayed normally
[image]

mobizt, 2020-09-22:
The problem is not Russian characters.

The string that you created from EEPROM (actually flash) should have the null character (**\0**) at the end or null terminated string.

That is if the string contains 5 chars plus the null char or '\0' or 0 at the end the total string length is 6.

If you reserve the memory space or char array only 5, the C compiler can't see the end of your string then you will get the garbage or some memory pointer when you use it.

mobizt, 2020-09-23:
Please see the below example.

**The working code**

```C++
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>
#include <EEPROM.h>

#define FIREBASE_HOST "YOUR_FIREBASE_PROJECT.firebaseio.com" //Without http:// or https:// schemes
#define FIREBASE_AUTH "YOUR_FIREBASE_DATABASE_SECRET"
#define WIFI_SSID "YOUR_WIFI_AP"
#define WIFI_PASSWORD "YOUR_WIFI_PASSWORD"

FirebaseData firebaseData;

String path1 = "Test/hardcoded";
String path2 = "Test/eeprom";

void printResult(FirebaseData &data)
{

  if (data.dataType() == "int")
    Serial.println(data.intData());
  else if (data.dataType() == "float")
    Serial.println(data.floatData(), 5);
  else if (data.dataType() == "double")
    printf("%.9lf\n", data.doubleData());
  else if (data.dataType() == "boolean")
    Serial.println(data.boolData() == 1 ? "true" : "false");
  else if (data.dataType() == "string")
    Serial.println(data.stringData());
  else if (data.dataType() == "json")
  {
    Serial.println();
    FirebaseJson &json = data.jsonObject();
    //Print all object data
    Serial.println("Pretty printed JSON data:");
    String jsonStr;
    json.toString(jsonStr, true);
    Serial.println(jsonStr);
    Serial.println();
    Serial.println("Iterate JSON data:");
    Serial.println();
    size_t len = json.iteratorBegin();
    String key, value = "";
    int type = 0;
    for (size_t i = 0; i < len; i++)
    {
      json.iteratorGet(i, type, key, value);
      Serial.print(i);
      Serial.print(", ");
      Serial.print("Type: ");
      Serial.print(type == FirebaseJson::JSON_OBJECT ? "object" : "array");
      if (type == FirebaseJson::JSON_OBJECT)
      {
        Serial.print(", Key: ");
        Serial.print(key);
      }
      Serial.print(", Value: ");
      Serial.println(value);
    }
    json.iteratorEnd();
  }
  else if (data.dataType() == "array")
  {
    Serial.println();
    //get array data from FirebaseData using FirebaseJsonArray object
    FirebaseJsonArray &arr = data.jsonArray();
    //Print all array values
[... 212 lines cut ...]

0, Type: object, Key: квартира, Value: слишком
1, Type: object, Key: квартиру, Value: рядом

3. Save the Russian string to EEPROM
Ok

4. Store the EEPROM Russian string to Firebase
Ok -> 
Pretty printed JSON data:
{
    "квартира": "слишком",
    "квартиру": "рядом"
}

Iterate JSON data:

0, Type: object, Key: квартира, Value: слишком
1, Type: object, Key: квартиру, Value: рядом

```

**Firebase database result**

[image]

danmov26, 2020-09-23:
Thank you very much!