mobizt/Firebase-ESP8266 #143: How to use the server-side timestamp when store a data?

openedhardware, 2020-09-22:
Seems a silly question, I just push the sensor data by `Firebase.pushInt()` function, but the problem is that the ESP8266 doesn't have the RTC, so I cannot use the pushed data on my backend. 
 Any idea how to overcome this issue?

Thanks!

mobizt, 2020-09-22:
You can get the time based on the time zone from NTP as in [this examples](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/AutomaticPlantWatering/AutomaticPlantWatering.ino#L302).

Once the device time was set, you can read the device timestamp (seconds since midnight of 1st January 1970) from
`time_t timestamp = time(nullptr); //time_t is uint32_t alias`

closed 2020-09-22 by mobizt

loczi1, 2021-11-24:
Dear @mobizt, the link showing the example is not valid. Can you please share a solution?

mobizt, 2021-11-25:
@loczi1 
That example was no longer existed.

It is just about to get time from NTP server which you can search google for this.

In case you want to get server timestamp, you should set the server value to the data, then you will get the server timestamp in the return payload too.

You can see this example for how to.
https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/Timestamp/Timestamp.ino