mobizt/Firebase-ESP8266 #144: help need

lchvsanilkumar, 2020-09-25:
sir, i need to push data "\"Fan\"" like in this format..how can i push this data in this format...please tell me sir...

mobizt, 2020-09-25:
Use the back slash which followed by double quote to make a escape sequence like this.
[image]

closed 2020-09-25 by mobizt

lchvsanilkumar, 2020-09-25:
sir i can't push data like that format to firebase even i followed you suggetion...
so ,if you don't mind 
    if (Firebase.set(firebaseData2, parentPath + "/node2/new/data", "\"\"fan\"\"))
 please correct this line where it need...thank you sir

mobizt, 2020-09-25:
I don't understand what you do.
Why you need double quote in double quote like this ""Fan""? It's so strange. 

If the GitHub auto format what you are trying to post here, please use the capture image instead.

mobizt, 2020-09-25:
If you want to do so.
[image]

Please search google for the **escape sequence**.

lchvsanilkumar, 2020-09-25:
Set string...
FAILED
REASON: bad request, Invalid data; couldn't parse JSON object, array, or value..
i am getting error like this

mobizt, 2020-09-25:
Firebase is not support nested double quote.
You can test this in the web console.