mobizt/Firebase-ESP8266 #147: HELP

xralphack, 2020-09-27:
```

Panic core_esp8266_main.cpp:206 loop_task

>>>stack>>>

ctx: sys
sp: 3fffec10 end: 3fffffb0 offset: 0000
3fffec10:  00000002 00000000 40222cad 40226d30  
3fffec20:  000000fe 00000000 00000000 00000000  
3fffec30:  00000000 00000000 00000000 00000001  
3fffec40:  00000000 00000001 3fffed20 00002200  
3fffec50:  3fffc200 40102adc 04000000 4010149e  
3fffec60:  00000003 00000004 40003b55 00000000  
3fffec70:  00000000 00000000 00000000 40102abe  
3fffec80:  40102b39 ffffffff 00000000 00000000  
3fffec90:  00000001 00000031 00302063 00000004  
3fffeca0:  40003b50 2c9f0300 4000050c 3fffc278  
3fffecb0:  40102adc 3fffc200 00000022 34376237  
3fffecc0:  40003b55 00000030 00000008 ffffffff  
3fffecd0:  40001ddc 00000031 00018000 00000000  
3fffece0:  00ff0000 5ffffe00 5ffffe00 00000000  
3fffecf0:  00018000 4026d2c1 3fffecc3 3fffedea  
3fffed00:  3fffffb0 00000020 0000003c 00000030  
3fffed10:  00000020 3ffe834c 000000a0 402247cd  
3fffed20:  3fffee01 4026d2c1 0000006e ffff0208  
3fffed30:  3fffedd0 0000009f 3fffefc0 00000030  
3fffed40:  3fffffb0 3ffe834c 000000a0 402247cd  
3fffed50:  3fffee28 00000000 00000077 ffff0208  
3fffed60:  3fffee00 0000009f 40102dba 00000100  
3fffed70:  3ffe96ec 7fffffff 00002200 00000001  
3fffed80:  00000001 00000080 00000020 3ffef324  
3fffed90:  3ffe96ec 3fffffb0 3fffedd0 40224810  
3fffeda0:  3fffee90 3fffee80 00000004 3fffc278  
3fffedb0:  40102adc 3fffc200 00000022 40224810  
3fffedc0:  4020f734 3fffeee0 3fffee00 4020f728  
3fffedd0:  66666633 30313266 3120203a 30616638  
3fffede0:  20666330 35376132 62643165 36343220  
3fffedf0:  38326465 37622038 62343762 20203564  
3fffee00:  203a000a 66666633 30656565 646e6520  
3fffee10:  6633203a 66666666 6f203062 65736666  
3fffee20:  30203a74 0a303030 00006b00 40100d07  
3fffee30:  00000000 3ffef81c 00000860 00000218  
3fffee40:  4020f734 00000000 00000020 4020f728  
3fffee50:  6573550a 78652072 74706563 206e6f69  
3fffee60:  6e617028 612f6369 74726f62 7373612f  
3fffee70:  3fffee90 3fffee80 00000004 4022c326  
3fffee80:  3fff3ab4 3ffff210 18fa00cf 2a75e1db  
3fffee90:  246ed288 b7b74bd5 00000004 40100fc2  
3fffeea0:  3fffeec0 3fffeeb0 3ffff220 4020f782  
3fffeeb0:  00000020 3fffeee0 3fffffb0 00000000  
3fffeec0:  5ffffe00 5ffffe00 0000000f 3fffefc0  
3fffeed0:  3fffffb0 3fffeee0 3fffeee0 4020f97b <
3fffeee0:  00000000 3fff03ac 00000020 40100f8b  
3fffeef0:  000000fe 00000000 00000000 00000000  
3fffef00:  00000000 00000000 00000000 3ffef360  
3fffef10:  c0000000 3fffd9d0 00000000 3ffef324  
3fffef20:  3fffdad0 00000000 3ffe8538 4020f9fe  
3fffef30:  4025df12 3fffdab0 00000000 4020fa5b  
3fffef40:  00000000 3fffdad0 3ffef324 4020f486  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 00000005 000799b8 00000000  
3fffef70:  00000000 aa55aa55 00000077 40104f39  
3fffef80:  40104f3f 000799b8 00000000 00000000  
3fffef90:  4010000d 00000000 00000000 00000000  
3fffefa0:  402568fc 3fffef3c 402568ad 3ffff868  
3fffefb0:  00000005 00000000 00000020 40100594  
3fffefc0:  feefeffe feefeffe 00000005 40101b80  
3fffefd0:  3ffe8e35 40104e33 3ffec848 feefeffe  
3fffefe0:  40102717 3ffec848 feefeffe feefeffe  
3fffeff0:  ffffffdb 05f0e7ae 3ffed290 401028f8  
3ffff000:  3ffe96d4 00000000 00000000 feefeffe  
3ffff010:  ffffffdb 05f0e7ae 40102dba 00000100  
3ffff020:  3ffe96d4 7fffffff 00002200 00000001  
3ffff030:  00000001 00000080 feefeffe feefeffe  
3ffff040:  3ffe96d4 feefeffe feefeffe 05f0e7ae  
3ffff050:  3ffe96ec 2c9f0300 4000050c 3fffc278  
3ffff060:  40102adc 3fffc200 00000022 09d59475  
3ffff070:  4023504a 00000030 00000016 ffffffff  
3ffff080:  40234635 0410afef eeccb806 a42f7236  
3ffff090:  00000000 2bfbdb94 6e53b68a eda28104  
3ffff0a0:  00003b68 f6e50000 ca000000 0076d140  
3ffff0b0:  a0410afe dcac0cba 1dad761d 00000030  
3ffff0c0:  66b069fa 08fb22c2 e53f4b60 7be35244  
3ffff0d0:  8e9d3a98 4059999d b6ac57d2 8a5bbade  
3ffff0e0:  7033d702 249b9861 d5b632c8 ad9776ad  
3ffff0f0:  fc3b0f5b 905131e0 dbaad140 d752efcb  
3ffff100:  8f2d9c9d 9a9ab955 50c49de7 ef0fea68  
3ffff110:  6d13d509 b08475cc f94edb3f f476cc6b  
3ffff120:  80000000 00000000 00000000 00000000  
3ffff130:  00000000 00000000 00000000 00000300  
3ffff140:  a560a19b 803b880b cc427575 8e6eafaf  
3ffff150:  733dceb0 089adb9f d25e9fd1 e195d1ee  
3ffff160:  9c61ccce 1f353f01 6e48357e 00000080  
3ffff170:  03030000 010000e9 ef17bc30 00005c34  
3ffff180:  38f2fef2 5f03f0d8 fe2dbf5f ff1e7367  
3ffff190:  9fcf0000 3a4724f9 a8c02bc0 5acca9cc  
3ffff1a0:  30c0acc0 2fc02cc0 afc023c0 adc0aec0  
3ffff1b0:  28c009c0 27c024c0 14c02dc0 13c00ac0  
3ffff1c0:  32c025c0 31c02ec0 2ac004c0 29c026c0  
3ffff1d0:  0f009c00 0ec005c0 9dc0a0c0 9dc09cc0  
3ffff1e0:  3d002f00 a1003c00 12c003c0 35c008c0  
3ffff1f0:  31f393b5 54262f79 2d5b66c0 bdd0c80e  
3ffff200:  697c22fa fc697caf 1de5ca94 abf4e925  
3fffH!�ȩ
�@�9�

```

any suggestion?

mobizt, 2020-09-27:
Did you familiar with the esp8266 programming?
You need to debug yourself.

mobizt, 2020-09-27:
You make a stack memory overflown due to the looping in some blocks of code.

closed 2020-09-27 by mobizt

xralphack, 2020-09-27:
I am a beginner on embedded.

mobizt, 2020-09-27:
You need to post the code here for letting others to investigate or try your code.

Your code should modify to exclude the third party libraries out of your sketch before post it here to focus only the library usage error.

The error message you posted above is the stack trace which you need the tool like **[esp exception decoder](https://github.com/me-no-dev/EspExceptionDecoder)** to encode and look at the latest running code from the decoded result.

No one can decode the stack trace for you (unless you shared the full code) which required the elf file which created on the Arduino temp data folder when you compiled the sketch.