mobizt/Firebase-ESP8266 #149: Error compiling for board LOLIN(WEMOS) D1 R2 & mini

Proxcentaur, 2020-09-30:
I have been using this library since 7 months ago and everything is fine, but recently I have created a new project using this library and come up with error compiling for board  LOLIN(WEMOS) D1 R2 & mini. Any suggestion?

mobizt, 2020-09-30:
You should post the full of error message here.

It is better to check the third party libraries from some dev which don't care the system inconsistent that create the ambiguous library which its name is the same as in the core libraries.  The most case example is SD.h library which installed in libraries folder and already existed in core library and make the ambiguous name when the compiler trying to choose the right one.

closed 2020-09-30 by mobizt

doublejosh, 2020-10-21:
Often when I can't connect to a WEMOS it's because the cable only has power wires. So many of the USB minis cords are just power... don't let this gotcha bite you!