mobizt/Firebase-ESP8266 #150: pushJSON no longers work with char *

blotfi, 2020-10-07:
Hi
The new version does not accept old code that was working fine
`	char output[128];
	serializeJson(doc, output);
  Firebase.pushJSON(firebaseData, "/data", output);
`
I tried tu use
`Firebase.pushString`
put no success
 `Firebase.pushString(firebaseData, "/data", "{\"name\": \"iot4\"}");`

Whereas 
`Firebase.setString(firebaseData, datetimekey, datetime_str);`
works great
but I have no success now with push
in previous version of the library it was working

thanks for your help

mobizt, 2020-10-07:
Look like you use the very outdated library.

The library does not support third party JSON library (almost a year ago) because it has a built-in **FirebaseJso**n, the JSON parser and editor.

You should read the README or check the examples to use the **FirebaseJson** and **FirebaseJsonArray** object

mobizt, 2020-10-07:
Try this [example](https://github.com/mobizt/Firebase-ESP8266/blob/master/examples/jsonObject/jsonObject.ino).

The JSON object construction is as easy as Firebase.

blotfi, 2020-10-07:
I am with V2.9.6
but the Firebase.pushString
still exist in your library
so why  ; Firebase.pushString(firebaseData, "/data", "{\"name\": \"iot4\"}");
 doesn't work

I will try
   json.set("Data1",i + 1);
    json.set("Data2",i + 100);

    //Also can use Firebase.push instead of Firebase.pushJSON
    //JSON string does not support in v 2.6.0 and later, only FirebaseJson object is supported.
    if (Firebase.pushJSON(firebaseData, "/Test/Int", json))

blotfi, 2020-10-07:
I like the 
<ArduinoJson.h> library
is it possible to populate your json object with his DynamicJsonDocument or StaticJsonDocument
https://arduinojson.org/v6/api/staticjsondocument/

mobizt, 2020-10-07:
This **"{"name": "iot4"}"** is invalid syntax and can't compile.

You should use escape sequence by use the back space and followed by double quote as 

[image]

That library is not support for a year.

closed 2020-10-07 by mobizt

blotfi, 2020-10-07:
> This **"{"name": "iot4"}"** is invalid syntax and can't compile.
> 
> You should use escape sequence by use the back space and followed by double quote as
> 
> [image]
> 
> That library is not supported for a year.

Ya, I know,  I just copied from above and it erases the \
I suggest you to delete the function pushString from your library, since they do not work

mobizt, 2020-10-07:
The pushJson requred the FirebaseJson object or pointer.
The pushString is ok but you create invalid JSON  string.

Why to delete because of your fault?

mobizt, 2020-10-07:
Did you know the difference between string and JSON payload?

The payload that constructed from string will be inside the double quotes before sending to Firebase server.
While the JSON object will be sent to Firebase server with {} or [].

Then you have no chance to pass the JSON string to pushString or setString and expected the result as pushJson.

blotfi, 2020-10-07:
I said I wrote it well in the compiler editor and in my first post:
https://github.com/mobizt/Firebase-ESP8266/issues/150#issue-716512923
but when I copied again, github editor erased the escape char

Are you sure
`Firebase.pushString(firebaseData, "/data", "{\"name\": \"iot4\"}");`
works ?
because I tried it and it doesn't

mobizt, 2020-10-07:
Read my reply above

mobizt, 2020-10-07:
> Did you know the difference between string and JSON payload?
>
> The payload that constructed from string will be inside the double quotes before sending to Firebase server.
> While the JSON object will [...]

blotfi, 2020-10-07:
> Did you know the difference between string and JSON payload?
>
> The payload that constructed from string will be inside the double quotes before sending to Firebase server.
> While the JSON object will [...]

okay I didn't know that
thanks

mobizt, 2020-10-07:
You can try this in the Firebase web console.