mobizt/Firebase-ESP8266 #153: Unable to reconnect to firebase stream when internet connection is available again

djeceymca, 2020-10-14:
**Summary**: 
Unable to reconnect to firebase stream when internet connection is available again. But firebase Store (set) feature is working just fine.

**Simulation Steps**:

1. Turn off WiFi Connection.
2. Start Nodemcu without WiFi connected to it. Here we can safely say Firebase is no way connected to internet.
3. Turn on WiFi Connection and wait for nodemcu to automatically connect to WiFi and eventually internet.
4. At this moment, Firebase Store (set) feature is working fine. But Firebase stream is not working.

Sorry, This is the first time I am raising any bug on GitHub. Please do let me know if I need to add more information.

mobizt, 2020-10-14:
ํYou should post the code here.

mobizt, 2020-10-14:
The stream connection should keep alive as long as no other functions called i.e. set, get, push or update which will stop the stream if using the same Firebase Data object or the internet connection was interrupted. 

The stream will resume in 2 seconds after the interruption as said above.

Then using the different Firebase Data object for stream and other calls is recommended.

closed 2020-10-14 by mobizt

djeceymca, 2020-10-14:
Hello Mobizt,

Below is the code snippet

#include <Arduino.h>

// Library For Firebase
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>

// Library For WiFiManager
#include <WiFiManager.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>

#include <ArduinoJson.h>
#include <OneButton.h>

// Firebase Variables
#define FIREBASE_HOST "water-supply-a3c71.firebaseio.com"
#define FIREBASE_AUTH "7qmUrHKF3m2gJzC1krByN4fxaIVmKHqODUmtyBti"
FirebaseData firebaseData;
FirebaseData firebaseDataEvents;
String eventsParentPath = "/OneButton";
String eventsChildPath[1] = {"/oneButtonStatus"};
size_t eventsChildPathSize = 1;

// GPIO Pins
const int oneButtonInputPin = D4;
const int ledPin = D0;

// Reference Variables
const bool ON = true;
const bool OFF = false;

// State Variables
bool localOneButtonStatus = false;

// Button Initialization
OneButton oneButton = OneButton(oneButtonInputPin, true, true);

// WiFi
WiFiManager wifiManager;

void updateOneButtonStatusOnServer(bool action)
{
  if (!Firebase.setBool(firebaseData, "/OneButton/oneButtonStatus", action))
  {
    Serial.println(firebaseData.errorReason());
  }
}

void oneButtonClick()
{
  if (localOneButtonStatus)
  {
    updateOneButtonStatusOnServer(OFF);
    localOneButtonStatus = false;
  }
  else
  {
    updateOneButtonStatusOnServer(ON);
    localOneButtonStatus = true;
  }
}

void turnLED(bool action)
{
  digitalWrite(ledPin, action);
}

void oneButtonEventStreamCallback(MultiPathStreamData stream)
{
  size_t numChild = sizeof(eventsChildPath) / sizeof(eventsChildPath[0]);

  for (size_t i = 0; i < numChild; i++)
  {
    if (stream.get(eventsChildPath[i]))
    {
      if (stream.dataPath == "/oneButtonStatus")
      {
        if (stream.value == "true")
        {
          turnLED(true);
        }
        else
        {
          turnLED(false);
        }
      }
    }
  }
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
  {
    Serial.println();
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
}

void initiateFirebaseStream()
{
  Firebase.setMultiPathStreamCallback(firebaseDataEvents, oneButtonEventStreamCallback, streamTimeoutCallback);
  if (!Firebase.beginMultiPathStream(firebaseDataEvents, eventsParentPath, eventsChildPath, eventsChildPathSize))
  {
    Serial.println("REASON: " + firebaseDataEvents.errorReason());
  }
}

void setup()
{
  // Initializing Serial Communication
  Serial.begin(115200);

  // Pin Mode
  pinMode(ledPin, OUTPUT);

  wifiManager.setConfigPortalBlocking(false);

  // Button Event Listener
  oneButton.attachClick(oneButtonClick);
  oneButton.attachLongPressStart(oneButtonClick);
  oneButton.setPressTicks(200);

  // Wifi Connect
  Serial.println("Connecting...");
  if (wifiManager.autoConnect("AutoConnect"))
  {
    Serial.println("connected.");
  }
  else
  {
    Serial.println("Configportal running");
  }

  // Firebase Connect
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  delay(1000);
  initiateFirebaseStream();
}

void loop()
{
  wifiManager.process();
  oneButton.tick();
  yield();
}

djeceymca, 2020-10-14:
> The stream connection should keep alive as long as no other functions called i.e. set, get, push or update which will stop the stream if using the same Firebase Data object or the internet [...]

I am using two different Firebase objects. One for Store (set) and other for Stream

djeceymca, 2020-10-14:
Hello Mobizt,

Below is the code snippet

#include <Arduino.h>

// Library For Firebase
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>

// Library For WiFiManager
#include <WiFiManager.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>

#include <ArduinoJson.h>
#include <OneButton.h>

// Firebase Variables
#define FIREBASE_HOST "XXXXXXXX"
#define FIREBASE_AUTH "XXXXXXXX"
FirebaseData firebaseData;
FirebaseData firebaseDataEvents;
String eventsParentPath = "/OneButton";
String eventsChildPath[1] = {"/oneButtonStatus"};
size_t eventsChildPathSize = 1;

// GPIO Pins
const int oneButtonInputPin = D4;
const int ledPin = D0;

// Reference Variables
const bool ON = true;
const bool OFF = false;

// State Variables
bool localOneButtonStatus = false;

// Button Initialization
OneButton oneButton = OneButton(oneButtonInputPin, true, true);

// WiFi
WiFiManager wifiManager;

void updateOneButtonStatusOnServer(bool action)
{
if (!Firebase.setBool(firebaseData, "/OneButton/oneButtonStatus", action))
{
Serial.println(firebaseData.errorReason());
}
}

void oneButtonClick()
{
if (localOneButtonStatus)
{
updateOneButtonStatusOnServer(OFF);
localOneButtonStatus = false;
}
else
{
updateOneButtonStatusOnServer(ON);
localOneButtonStatus = true;
}
}

void turnLED(bool action)
{
digitalWrite(ledPin, action);
}

void oneButtonEventStreamCallback(MultiPathStreamData stream)
{
size_t numChild = sizeof(eventsChildPath) / sizeof(eventsChildPath[0]);

for (size_t i = 0; i < numChild; i++)
{
if (stream.get(eventsChildPath[i]))
{
if (stream.dataPath == "/oneButtonStatus")
{
if (stream.value == "true")
{
turnLED(true);
}
else
{
turnLED(false);
}
}
}
}
}

void streamTimeoutCallback(bool timeout)
{
if (timeout)
{
Serial.println();
Serial.println("Stream timeout, resume streaming...");
Serial.println();
}
}

void initiateFirebaseStream()
{
Firebase.setMultiPathStreamCallback(firebaseDataEvents, oneButtonEventStreamCallback, streamTimeoutCallback);
if (!Firebase.beginMultiPathStream(firebaseDataEvents, eventsParentPath, eventsChildPath, eventsChildPathSize))
{
Serial.println("REASON: " + firebaseDataEvents.errorReason());
}
}

void setup()
{
// Initializing Serial Communication
Serial.begin(115200);

// Pin Mode
pinMode(ledPin, OUTPUT);

wifiManager.setConfigPortalBlocking(false);

// Button Event Listener
oneButton.attachClick(oneButtonClick);
oneButton.attachLongPressStart(oneButtonClick);
oneButton.setPressTicks(200);

// Wifi Connect
Serial.println("Connecting...");
if (wifiManager.autoConnect("AutoConnect"))
{
Serial.println("connected.");
}
else
{
Serial.println("Configportal running");
}

// Firebase Connect
Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
Firebase.reconnectWiFi(true);
delay(1000);
initiateFirebaseStream();
}

void loop()
{
wifiManager.process();
oneButton.tick();
yield();
}