mobizt/Firebase-ESP8266 #154: Unable to reconnect to firebase stream when internet connection is available again

djeceymca, 2020-10-14:
Creating new issue again as previous one got closed. Below are details asked

**Summary**:
Unable to reconnect to firebase stream when internet connection is available again. But firebase Store (set) feature is working just fine. Using two different firebase objects, one for Store (set) feature and other for Stream.

**Simulation Steps**:

1. Turn off WiFi Connection.
2. Start Nodemcu without WiFi connected to it. Here we can safely say Firebase is no way connected to internet.
3. Turn on WiFi Connection and wait for nodemcu to automatically connect to WiFi and eventually internet.
4. At this moment, Firebase Store (set) feature is working fine. But Firebase stream is not working.

// Library For Firebase
#include <FirebaseESP8266.h>
#include <ESP8266WiFi.h>

// Library For WiFiManager
#include <WiFiManager.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>

#include <ArduinoJson.h>
#include <OneButton.h>

// Firebase Variables
#define FIREBASE_HOST "XXXXXXXX"
#define FIREBASE_AUTH "XXXXXXXX"
FirebaseData firebaseData;
FirebaseData firebaseDataEvents;
String eventsParentPath = "/OneButton";
String eventsChildPath[1] = {"/oneButtonStatus"};
size_t eventsChildPathSize = 1;

// GPIO Pins
const int oneButtonInputPin = D4;
const int ledPin = D0;

// Reference Variables
const bool ON = true;
const bool OFF = false;

// State Variables
bool localOneButtonStatus = false;

// Button Initialization
OneButton oneButton = OneButton(oneButtonInputPin, true, true);

// WiFi
WiFiManager wifiManager;

void updateOneButtonStatusOnServer(bool action)
{
if (!Firebase.setBool(firebaseData, "/OneButton/oneButtonStatus", action))
{
Serial.println(firebaseData.errorReason());
}
}

void oneButtonClick()
{
if (localOneButtonStatus)
{
updateOneButtonStatusOnServer(OFF);
localOneButtonStatus = false;
}
else
{
updateOneButtonStatusOnServer(ON);
localOneButtonStatus = true;
}
}

void turnLED(bool action)
{
digitalWrite(ledPin, action);
}

void oneButtonEventStreamCallback(MultiPathStreamData stream)
{
size_t numChild = sizeof(eventsChildPath) / sizeof(eventsChildPath[0]);

for (size_t i = 0; i < numChild; i++)
{
if (stream.get(eventsChildPath[i]))
{
if (stream.dataPath == "/oneButtonStatus")
{
if (stream.value == "true")
{
turnLED(true);
}
else
{
turnLED(false);
}
}
}
}
}

void streamTimeoutCallback(bool timeout)
{
if (timeout)
{
Serial.println();
Serial.println("Stream timeout, resume streaming...");
Serial.println();
}
}

void initiateFirebaseStream()
{
Firebase.setMultiPathStreamCallback(firebaseDataEvents, oneButtonEventStreamCallback, streamTimeoutCallback);
if (!Firebase.beginMultiPathStream(firebaseDataEvents, eventsParentPath, eventsChildPath, eventsChildPathSize))
{
Serial.println("REASON: " + firebaseDataEvents.errorReason());
}
}

void setup()
{
// Initializing Serial Communication
Serial.begin(115200);

// Pin Mode
pinMode(ledPin, OUTPUT);

wifiManager.setConfigPortalBlocking(false);

// Button Event Listener
oneButton.attachClick(oneButtonClick);
oneButton.attachLongPressStart(oneButtonClick);
oneButton.setPressTicks(200);

// Wifi Connect
Serial.println("Connecting...");
if (wifiManager.autoConnect("AutoConnect"))
{
Serial.println("connected.");
}
else
{
Serial.println("Configportal running");
}

// Firebase Connect
Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
Firebase.reconnectWiFi(true);
delay(1000);
initiateFirebaseStream();
}

void loop()
{
wifiManager.process();
oneButton.tick();
yield();
}

closed 2020-10-15 by mobizt

mobizt, 2020-10-15:
You're using the outdated WiFIManager library as the **process** function is not available in the latest version.

There are no problem when I test with this code.

```C++

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

#define FIREBASE_HOST "XXXXXXXX"
#define FIREBASE_AUTH "XXXXXXXX"

#define WIFI_SSID "ZZZZZZZZZZ"
#define WIFI_PASSWORD "ZZZZZZZZZZZ"

FirebaseData firebaseData;
FirebaseData firebaseDataEvents;
String eventsParentPath = "/OneButton";
String eventsChildPath[1] = {"/oneButtonStatus"};
size_t eventsChildPathSize = 1;

WiFiManager wifiManager;

void updateOneButtonStatusOnServer(bool action)
{
  if (!Firebase.setBool(firebaseData, "/OneButton/oneButtonStatus", action))
  {
    Serial.println(firebaseData.errorReason());
  }
}

void oneButtonEventStreamCallback(MultiPathStreamData stream)
{
  size_t numChild = sizeof(eventsChildPath) / sizeof(eventsChildPath[0]);
  for (size_t i = 0; i < numChild; i++)
  {
    if (stream.get(eventsChildPath[i]))
    {
      if (stream.dataPath == "/oneButtonStatus")
      {
        Serial.println("Current Btn Status: " + stream.value);
      }
    }
  }
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
  {
    Serial.println();
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
}

void initiateFirebaseStream()
{
  Firebase.setMultiPathStreamCallback(firebaseDataEvents, oneButtonEventStreamCallback, streamTimeoutCallback);
  if (!Firebase.beginMultiPathStream(firebaseDataEvents, eventsParentPath, eventsChildPath, eventsChildPathSize))
  {
    Serial.println("REASON: " + firebaseDataEvents.errorReason());
  }
}

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  initiateFirebaseStream();
}

void loop()
{
 
}

```

Your problem can be the following
1. The long time blocking code in **wifiManager.process();** or **oneButton.tick();** which prevent the stream task (schedule task) to run. I can't track what's inside the **wifiManager.process()** function for you because it's outdated.

2. The WiFIManager library trying to reconnect the Wi-Fi with its own implementation and the Firebase library trying to reconnect the WiFi using the core function i.e. **WiFi.reconnect()** every 10 seconds when the WiFi status is notconnected which will make a conflict due to WiFI will disconnect and reconnect many times in different librariess.  

I recommend that don't use the WiFI manager to control the connection after the setup.The Firebase library will handle the WiFi reconnection in the smarter and safer ways for you when you use **Firebase.reconnectWiFi(true);** in the setup.

If you want the WiFIManager to take control the rWiFi econnection, you can use **Firebase.reconnectWiFi(false);**.

Please don't open the same new issue just because I close it. I close the issue incase where the issue get my reply and I verify that there is no bugs related and that open issue is likely to be conflicts from the usage or I don't get the feedback intime.

djeceymca, 2020-10-15:
> You're using the outdated WiFIManager library as the **process** function is not available in the latest version.
> 
> There are no problem when I test with this code.
> 
> ```c++
> #include <Arduino.h>
> #include <ESP8266WiFi.h>
> #include <FirebaseESP8266.h>
> 
> #define FIREBASE_HOST "XXXXXXXX"
> #define FIREBASE_AUTH "XXXXXXXX"
> 
> #define WIFI_SSID "ZZZZZZZZZZ"
> #define WIFI_PASSWORD "ZZZZZZZZZZZ"
> 
> 
> FirebaseData firebaseData;
> FirebaseData firebaseDataEvents;
> String eventsParentPath = "/OneButton";
> String eventsChildPath[1] = {"/oneButtonStatus"};
> size_t eventsChildPathSize = 1;
> 
> WiFiManager wifiManager;
> 
> void updateOneButtonStatusOnServer(bool action)
> {
>   if (!Firebase.setBool(firebaseData, "/OneButton/oneButtonStatus", action))
>   {
>     Serial.println(firebaseData.errorReason());
>   }
> }
> 
> void oneButtonEventStreamCallback(MultiPathStreamData stream)
> {
>   size_t numChild = sizeof(eventsChildPath) / sizeof(eventsChildPath[0]);
>   for (size_t i = 0; i < numChild; i++)
>   {
>     if (stream.get(eventsChildPath[i]))
>     {
>       if (stream.dataPath == "/oneButtonStatus")
>       {
>         Serial.println("Current Btn Status: " + stream.value);
>       }
>     }
>   }
> }
> 
> void streamTimeoutCallback(bool timeout)
> {
>   if (timeout)
>   {
>     Serial.println();
>     Serial.println("Stream timeout, resume streaming...");
>     Serial.println();
>   }
> }
> 
> void initiateFirebaseStream()
> {
>   Firebase.setMultiPathStreamCallback(firebaseDataEvents, oneButtonEventStreamCallback, streamTimeoutCallback);
>   if (!Firebase.beginMultiPathStream(firebaseDataEvents, eventsParentPath, eventsChildPath, eventsChildPathSize))
>   {
>     Serial.println("REASON: " + firebaseDataEvents.errorReason());
>   }
> }
> 
> void setup()
> {
> 
>   Serial.begin(115200);
> 
>   WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
>   Serial.print("Connecting to Wi-Fi");
>   while (WiFi.status() != WL_CONNECTED)
>   {
>     Serial.print(".");
>     delay(300);
>   }
>   Serial.println();
>   Serial.print("Connected with IP: ");
>   Serial.println(WiFi.localIP());
>   Serial.println();
> 
>   Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
>   Firebase.reconnectWiFi(true);
> 
>   initiateFirebaseStream();
> }
> 
> void loop()
> {
>  
> }
> ```
> 
> Your problem can be the following
> 
> 1. The long time blocking code in **wifiManager.process();** or **oneButton.tick();** which prevent the stream task (schedule task) to run. I can't track what's inside the **wifiManager.process()** function for you because it's outdated.
> 2. The WiFIManager library trying to reconnect the Wi-Fi with its own implementation and the Firebase library trying to reconnect the WiFi using the core function i.e. **WiFi.reconnect()** every 10 seconds when the WiFi status is notconnected which will make a conflict due to WiFI will disconnect and reconnect many times in different librariess.
> 
> I recommend that don't use the WiFI manager to control the connection after the setup.The Firebase library will handle the WiFi reconnection in the smarter and safer ways for you when you use **Firebase.reconnectWiFi(true);** in the setup.
> 
> If you want the WiFIManager to take control the rWiFi econnection, you can use **Firebase.reconnectWiFi(false);**.
> 
> Please don't open the same new issue just because I close it. I close the issue in the case where the issue gets my reply and I verify that there is no bugs related and that opened issue is likely to be conflicts from the usage or I don't get the feedback intime.

Hello Mobizt,

Some observatons:
1. I would like to again mention that Firebase Store (set) is working fine, If wifiManager.process or oneButton.tick() would have been blocking code then Firebase Store (set) should not work as well.
2. I am using WiFiManager's development branch, wifiManager.process() is available in development branch. You are right release branch does not have this function. If possible can you have a look in development branch.
3. If I have follow-up queries, is there any other channel I can talk to you as this thread is closed. Or if possible can you keep issue open for a day, so that follow-up queries can be raised.

I will try to use Firebase.reconnectWiFi(false) to see if that solves the stream issue and will update you on results.

mobizt, 2020-10-15:
No need to re-quote the post that waste the memory used in GitHub.

No library issue in your case unless the conflicts of your implementation if you still intend to use something that I not recommend and no need to let me know because it's not relevant to my works.

The library passes the long time intensive testing that will not affected the user experiences when uses in collaboration with other library.