mobizt/Firebase-ESP8266 #155: "connection refused" for everything, new Admin SDK?

doublejosh, 2020-10-23:
I'm seeing "connection refused for all reads or writes that previously worked. Is this a result of database secrets being deprecated?

https://firebase.google.com/docs/admin/setup?authuser=0
https://firebase.google.com/docs/auth/cpp/custom-auth?authuser=0

```
Database secrets are currently deprecated and use a legacy Firebase token generator. Update your source code with the Firebase Admin SDK.
```

**ESP8266 Arduino Core SDK version**
 - Version 2.9.8

mobizt, 2020-10-23:
No issue from the test now.

Don't mention about the deprecated database secret cause it's not revoked now and still be activated unless obsoleted or the better auth tokens are recommended by Google for more secure.

mobizt, 2020-10-23:
Check your provided FIREBASE_HOST, it should not end with slash "/".
Check your FIREBASE_AUTH or database secret, are you ever renewed it?
Check your free heap, it may fail to perform ssl handshaking when the memory is not enough (approx. < 10K)

closed 2020-10-23 by mobizt

mobizt, 2020-10-23:
Nothing is related to Firebase Admin SDK here which it relies on tcp/ip socket when working with REST APIs as used in this library which relies on http.

The authentication methods supported by Firebase are OAuth2.0 and custom tokens which database secret is the legacy key that used only in RTDB product and service and still be supported.

doublejosh, 2020-10-23:
Perhaps it's a rules issue then.

FYI: This is used on a Wemos D1 mini.

mobizt, 2020-10-23:
The read/write security rules are only for access token while we are using the database secret which is the key which can bypass the security rules that's why Google convinced us to use the new way of authentication.

https://github.com/mobizt/Firebase-ESP8266/issues/128#issuecomment-673436234

mobizt, 2020-10-23:
You should check what I recommend above and I currently test it and  have no problem.

mobizt, 2020-10-23:
Update the library and test with some examples and if it ok then check your working code what is the conflicts or post your clean code version (third party libs removed) here.

doublejosh, 2020-10-23:
https://github.com/doublejosh/FortuneTeller/blob/master/FortuneTeller.ino

I tried setting the rules to allow all, that wasn't the cause.

I've been through all the basic steps, and was able to connected before.

mobizt, 2020-10-23:
Did you update the library?

The error connection refused is now reserved only for the file/backup/restore errors. 

Where is the error return (from which function)?

doublejosh, 2020-10-23:
I updated the FirebaseESP8266 library to 2.9.8

Error comes from any attempt to read or write.

mobizt, 2020-10-23:
You need to debug and tell what the function this error was returned.

I can't debug your code and you need to prove where it comes. Said that it from any read or write can't help because all function you used passed the test prior to library published.

doublejosh, 2020-10-23:
In my code here's where I see the failure first...
https://github.com/doublejosh/FortuneTeller/blob/master/FortuneTeller.ino#L161

I just pushed an update to the code with some added debugging.

doublejosh, 2020-10-23:
I just lowered all three of these to 8192 from 16384...
FIREBASE_BUFFER_SIZE_RECEIVE
FIREBASE_RESPONSE_SIZE
JSON_DOC_BYTES

And the error reason has now become `data type mismatch` ...so perhaps there's a memory issue on the Wemos.

mobizt, 2020-10-23:
Ok I'm checking and inform you soon.

mobizt, 2020-10-23:
Ok the bug is now fixed, please update the library by download (may be available in library manager on the next few hours).

The memory reserved by the BearSSL rx buffer will reduced your free heap which reserve as much as 8k or 16k is not recommended. 

When you get the data from push collection, the log push name which auto created costs the size of JSON payload and you may need to reserve more rx buffer to read all data.

I recommend to reduce the data to avoid BearSSL memory reservation by using the function **updateNode** with shorter predefined key instead of **pushJson**.

Use query to filter the return payload data inside the scope or range, you also need to set some key as index in database rules.

doublejosh, 2020-10-23:
For reference...
https://github.com/mobizt/Firebase-ESP8266/commit/d1392d8ce8ca14e0ff19aef3bbaa0d53169f72cf

doublejosh, 2020-10-24:
My second use of `Firebase.updateNode` causes a `Soft WDT reset`, within (`saveInteractionEnd`). I already used `delay(0)` inside my loops. Why would `updateNode` cause a reset? I've watched it update FB data once, but the second time it crashes.

mobizt, 2020-10-24:
You're running out of memory.
With empty sketch, the free heap is about 52k.
With one Firebase Data object (1k for BearSSL rx/tx reserved memory), your free heap is reduced to approx. 32k.
With two Firebase Data objects (1k for BearSSL rx/tx reserved memory), your free heap is reduced to approx. 25k.

Reserve the large memory for BearSSL rx/tx which you currently doing, you will lost your free heap as much as you reserved memory for it.

You need to manage the memory usage and debug every step in your code because you're programming the embedded device with the the WiFi and it need the time and resource to operate WiFi too.